esp8266/Arduino #4260: Problem on using SPI Extended Mode

procodr, 2018-01-30:
#### Hardware
Hardware:			NodeMcu Lolin V3
Core Version:      	2.4.0

### Description

According to [this documentation](https://arduino-esp8266.readthedocs.io/en/latest/libraries.html#spi) the pins are used for SPI Extended Mode is:
    `MOSI` = SD1
    `MISO` = SD0
    `SCLK` = CLK
    `HWCS` = GPIO0

but I cant use SD0 pin because the ESP HANG or CRASH when i use this pin (ex: Connecting it to a SDCard Reader).
it seems there is some data on it, but i can not see because i don't have logic analyzer..

is there any way to use this pin for SPI Extended Mode?

### Sketch

```cpp

#include <SPI.h>

void setup() {
  SPI.pins(6, 7, 8, 0); //with or without it there is problem with SD0/GPIO7 pin
  pinMode(15, OUTPUT);
}

void loop() {
  //Just for test as ESP heartbeat, this will freeze once the connecting anything to SD0 pin
  digitalWrite(15, HIGH);
  delay(100);
  digitalWrite(15, LOW);
  delay(100);
}
```

MarkyAD, 2018-01-30:
You call SPI.pins(); but you left out SPI.begin();

MarkyAD, 2018-01-30:
Also note that if you use GPIO0 for chip/slave select, you cant actually control it yourself, as its controlled by the hardware to give flash a higher priority, so you have to let the bus arbiter control CS/SS.

procodr, 2018-01-31:
Thanks for your quick reply @MarkyAD
I tested it by adding `SPI.begin();` (sorry for forgetting this), now the sketch is:
```cpp
#include <SPI.h>

void setup() {
  SPI.pins(6, 7, 8, 0); //with or without it there is problem with SD0/GPIO7 pin
  SPI.begin();
  pinMode(15, OUTPUT);
}

void loop() {
  //Just for test as ESP heartbeat, this will freeze once the connecting anything to SD0 pin
  digitalWrite(15, HIGH);
  delay(100);
  digitalWrite(15, LOW);
  delay(100);
}
```
but it does not change anything.
and the SS/CS pin is not the problem...

Here is the schematic:
[image]

The problem is that if I start ESP with this schematic it does not do any thing (Heartbeat LED does not works)
but if I disconnect S0 pin (and restart) the ESP continues to work.
now if I connect that S0 pin to ESP again, it will FREEZE (and Heartbeat stops).

procodr, 2018-01-31:
A rare problem 🤔...
It seems the problem is because of the SD Card Reader Module.
According to [this post](https://forum.arduino.cc/index.php?topic=360718.msg2942160#msg2942160) my SD Card Module does not have control over it's `MISO` pin and does not care of `CS` to activate `MISO` just when it's needed!

Admins can close this issue they agree with me.

MarkyAD, 2018-01-31:
@procodr you could do that MISO/CS mod it shows... Id've tried it 20 years ago but my eyes can't deal with soldering smd stuff even with magnifiers now. I have only been able to use those HSPI0 pins successfully with a tft display... even if you get it to boot there may still be issues with the SD library versus the hardware-controlled CS. You can close this if its done.

closed 2018-01-31 by procodr

alexmod588, 2021-11-01:
Hello! Did you manage to add the SD card module to NodeMCU using the extended mode? Could you send me the code if you have? I am trying the same thing because I need some extra pins in my project. Did you use the standard SD.h library?

procodr, 2021-11-06:
Hello @alexmod588!
No unfortunately, AFAIK.
That was the SD card module problem, You maybe find workarounds [here](https://forum.arduino.cc/index.php?topic=360718.msg2942160#msg2942160).
Share please if you found any solutions 👍