esp8266/Arduino #4267: lost wifi cause a reboot

chongding, 2018-02-01:
### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description
I have the below code to detect lost wifi connection then re-connect (with the existing ssid and pw). and I trigger the lost-wifi by shutting down my router, but when the wifi is lost, it cause 8266 to reboot. 

### Settings in IDE

Module:  			?ESP9266MOD-12E/F
Flash Size: 		?4MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?80Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <ESP8266WiFi.h> 

void setup() {
  WiFi.mode(WIFI_AP_STA);
  WiFi.begin(ssid, pass);
}

void loop() {
   if (WiFi.status() == WL_CONNECTED) {
       ....
   }else{
     WiFi.begin(ssid, pass);
   }

}
```

### Debug Messages

```
messages here
```

tonton81, 2018-02-01:
you do realize your flooding WiFi.begin? im not sure if theres a check inside until its finished, but, your still running it millions of times, it just aint right :P

lrmoreno007, 2018-02-01:
There is a function that will do that job better:
`WiFi.setAutoReconnect (true);`

tonton81, 2018-02-01:
and a handler :)

chongding, 2018-02-02:
thanks for the autoreconnect  tip, it seems reset on lost connection while exchanging data with a website...

closed 2018-02-02 by chongding