esp8266/Arduino #4269: Simple write to RTC memory causing ODD code operation.

LZHenry, 2018-02-01:
for Example Code... RTCUserMemory.ino 
https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/RTCUserMemory/RTCUserMemory.ino

Using esp8266 12E
Compiling with MS Visual Studio/ vMicro

The Example operates OK when in stand-alone testing.  Nice, clean comments.. thank you .  

Adding this rtc functionality to my existing application causes strange operation, but not CRASH.

Solution found, but not understood.  Off doing RTC research... any advice is appreciated.
…......................................................................................................................................

This line caused ODD operation.
	for (int i = 0; i < (sizeof(rtcData)); i++)  //this line caused the failure of DHT read 

Fixed with this line subtracting 2 from the sizeof rtcData:

	for (int i = 0; i < (sizeof(rtcData)-2); i++)  // had to subtract 2 or it overwrites something I needed... 

**Bad Operation:**

- NO compile errors.
- Code operates but “skips” some in line code but executes other.
- 
- Call to getAndSendThingsBoardData_DHT(); is SKIPPED when the sizeof(rtcData) = 512
- but my code operates OK when it = 510
- I do not understand.  

However... thank you igrr very much for the excellent example code... this and others I follow from you.  LZHenry

**In SETUP calls....**
	for (int i = 0; i < (sizeof(rtcData)-2); i++)  // had to subtract 2 or it overwrites something I needed... 
	{
 		rtcData.data[i] = 4;   // 
	}
	// Update CRC32 of data
	rtcData.crc32 = calculateCRC32(((uint8_t*)&rtcData) + 4, sizeof(rtcData) - 4);
	// Write struct to RTC memory
	if (ESP.rtcUserMemoryWrite(0, (uint32_t*)&rtcData, sizeof(rtcData)))
	{
		Serial.println("Write: ");
		printMemory();
		Serial.println();
	}

**In LOOP Calls...** 

void loop() 
{
			//Connected to ThingsBoard??? 
			if (!client.connected())
			{
			reconnect();
			}

		client.loop();
	 
		if (millis() - lastSend > SENSOREADINTERVAL)
		{ // Update and send only after 1 or 5  seconds

#ifdef  ENABLE_DHTSENSOR
				#ifdef   PRINTDIAG
				Serial.println("at get send DHT ");
				#endif
			getAndSendThingsBoardData_DHT();
#endif

#ifdef  ENABLE_DS1820SENSOR
					#ifdef   PRINTDIAG
								Serial.println("at get send1820 ");
					#endif
			//	getAndSendThingsBoardData_1820();
#endif
			lastSend = millis();
		}

		FlashBST();   // MILLIS Controls when it is time to cycle the BST

#ifdef   ENABLE_SLEEPMODE
	  //// DO LOOP things until the LARGE timer expires ... then sleep
	   unsigned long TBCONNECT_currentMillis = millis();

	   if (TBCONNECT_currentMillis - TBCONNECT_lastMillis >= TBCONNECT_interval)
	   {
		   TBCONNECT_lastMillis = TBCONNECT_currentMillis;  // Set up for next time 
		   {
			   ////Serial.println("TIME TO SLEEP ???????????????????????????????");
			   // IF timer is up... SLEEP 

			   Serial.print("FINALLY going to deep sleep, num_seconds="); Serial.println(TIMETOSLEEP);

			 //  ESP.deepSleep(TIMETOSLEEP * 1000000, RF_DISABLED);
			   ///ESP.deepSleep(TIMETOSLEEP * 1000000);
			   PutMicroToSleep();
		   }  // END TIME TO SLEEP
	   }

#endif	
}  // END LOOP
   //////   //+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
   //////   //+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
   //////   //+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

devyte, 2018-02-04:
@LZHenry when you opened this issue you were presented with a template requesting specific info. While your explanation is ok, info such as core version is missing. Please edit your post and add the missing info. Also, please fix the markup to make the code parts readable. There's a preview tab, it's there to check how your post will look like.

devyte, 2018-02-04:
The example seems to have a bug, and you're apparently making the same mistake.
From the example:
```
struct {
  uint32_t crc32;
  byte data[508];
} rtcData;
...
  for (int i = 0; i < sizeof(rtcData); i++) {
    rtcData.data[i] = random(0, 128);
  }
...
  for (int i = 0; i < sizeof(rtcData); i++) {
    sprintf(buf, "%02X", rtcData.data[i]);
...

```

In your case:
```
for (int i = 0; i < (sizeof(rtcData)); i++)
...

```
The iteration is being done over sizeof(rtcData), which is of size 512, but the index is used for rtcData.data[], which is of size 508. That means that the last 4 runs through the loop are indexing beyond the end of data[]. 
If you're reading, then it's garbage. If you're writing, then you're corrupting who knows what.

The correct loop for accessing only the data should be something like this:
```
  for (int i = 0; i < sizeof(rtcData)-4; i++) {
    rtcData.data[i] = random(0, 128);
  }
...
  for (int i = 0; i < sizeof(rtcData)-4; i++) {
    sprintf(buf, "%02X", rtcData.data[i]);
...
```

In your case:
```
for (int i = 0; i < (sizeof(rtcData)-4); i++)
  //do something with rtcData.data[i]

```

I'll submit a PR with a fix for the example.

devyte, 2018-02-04:
Better yet:
```
for (int i = 0; i < (sizeof(rtcData.data)); i++)
  //do something with rtcData.data[i]
```

closed 2018-02-06 by devyte

LZHenry, 2018-02-07:
Thanks devyte... LZH

PR #4288 (2018-02-06): Fix RTCmem example buffer access past end
Fixes #4269