esp8266/Arduino #4272: Using an object in my own lib resets the esp

GregorHse, 2018-02-02:
### Basic Infos

#### Hardware
Hardware:			 NodeMCU
Core Version:      	?2.1.0-rc2?

### Description

Hey, im new to the object oriented part of Arduino, so it could very well be that i made a beginner mistake. Im trying to use a ledstrip object from the Adafruit_NeoPixel library in my own library. Whenever i run the sketch on my esp, it crashes. It seems that using the strip object causes the crash, but im not sure what exactly triggers it since i am able to set single leds without the esp crashing. If im using the for loop to set more leds, as shown in my .cpp. the esp crashes everytime.

Reset:
```
 for(int i = 0;i<strip.numPixels();i++){
 strip.setPixelColor(i,255,0,0,0);
 }
```
Does not reset:
` strip.setPixelColor(77,255,0,0,0);
`

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp
#include "LedStrip.h"

LedStrip ledStrip(13, 266);

void setup() {
  Serial.begin(115200); //start the serial line
  ledStrip.begin();
  Serial.println("");
  Serial.println("setup successfull");
  Serial.println("");
}

void loop() {
  yield();
}
```
### LedStrip.h
```cpp
#ifndef ledStrr
#define ledStrr

#include "Arduino.h"
#include <Adafruit_NeoPixel.h>

class LedStrip  {
  public:
    // Constructor
    LedStrip(int PIN, int NUM_LEDS);
    Adafruit_NeoPixel strip;
    void begin();
};
#endif
```
### LedStrip.cpp
```cpp
#include "LedStrip.h"
LedStrip::LedStrip(int PIN, int NUM_LEDS) {
  strip = Adafruit_NeoPixel(NUM_LEDS, PIN, NEO_GRBW + NEO_KHZ800);
}

void LedStrip::begin() {
  delay(100);
  delay(100);
  strip.begin();
  delay(100);
  strip.show();
  delay(100);
 for(int i = 0;i<strip.numPixels();i++){
 strip.setPixelColor(i,255,0,0,0);
 }
  delay(1000);
  strip.show();
}
```

### Debug Messages

```
Exception (3):
epc1=0x401003e9 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4006f931 depc=0x00000000

ctx: sys 
sp: 3ffffd10 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffeb0:  40000f68 00000030 00000011 ffffffff  
3ffffec0:  40000f58 00000000 00000020 000001f4  
3ffffed0:  40209b74 3ffe8e84 3ffe8e84 401004d8  
3ffffee0:  ffffffff 3fffc6fc 3ffe9834 4020f8a8  
3ffffef0:  00000000 3fffdcb0 3ffe8f68 40208f30  
3fffff00:  3ffeeb88 00000007 3ffe8e84 40208f6c  
3fffff10:  40209b74 3ffefc90 3fff016c 401064a5  
3fffff20:  3ffe8f50 3ffe9834 0000003a 3ffefc8c  
3fffff30:  40209b74 3ffefc90 3ffe8e84 40209b90  
3fffff40:  40100e66 00029172 00000000 40209c40  
3fffff50:  3ffe8e84 3ffe9834 00029172 00000000  
3fffff60:  4022053c 3ffe8f34 3ffe8f50 60000600  
3fffff70:  00087800 3ffe8f50 3ffe8f34 40220549  
3fffff80:  4022058e 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffe8f70 3fffdab0 00000000 4020269f  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
```

tonton81, 2018-02-02:
for(int i = 0;i<strip.numPixels();i++){
 strip.setPixelColor(i,255,0,0,0);
 }

are you referring to a function or variable return?
this is probably why.

tonton81, 2018-02-02:
ahh adafruit actually has that function, have you tried defining the number statically in the for loop instead?

Hiddenvision, 2018-02-02:
@GregorHse 
is it just a WD timeout looping without giving up some time.
Also perhaps try shifting from pin 13 to 15 or 4
I only say that cos I have two sets of 60 leds running on those pins without issues.

If you drop that Exception output into the exception decoder it may help you see a cause.

Hv.

tonton81, 2018-02-02:
try adding delay(0) or yield() in your for loop
the watchdog gets angry if you dont feed him

GregorHse, 2018-02-02:
@tonton81 
> ahh adafruit actually has that function, have you tried defining the number statically in the for loop instead?
yes, still crashes. the strip.numPixels(); function works without a problem tho.

@Hiddenvision 
im running at 80mhz. if i put a yield() or delay() in the loop, nothing changes.

And if i create the object in my sketch and use the same functions and the for loop with it, it works just fine. with and without a delay/yield in int.
exception decoder gave me this:
```
Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
Decoding 14 results
0x401003e9: _umm_malloc at C:\Users\Gregor\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1386
0x401004d8: malloc at C:\Users\Gregor\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1668
0x40106f1c: pvPortMalloc at C:\Users\Gregor\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/heap.c line 40
0x402133ed: ieee80211_decap at ?? line ?
0x402336ac: pbuf_alloc at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c line 634
0x40213304: ieee80211_deliver_data at ?? line ?
0x40217abc: sta_input at ?? line ?
0x40230e8b: pp_tx_idle_timeout at ?? line ?
0x4023076b: ppPeocessRxPktHdr at ?? line ?
0x402026a3: loop_task at C:\Users\Gregor\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/core_esp8266_main.cpp line 57
```

tonton81, 2018-02-02:
ahhhhh sketch vs library, gotcha your out of scope perhaps but try to #include neopixels in the cpp

GregorHse, 2018-02-02:
@tonton81 

> ahhhhh sketch vs library, gotcha your out of scope perhaps but try to #include neopixels in the cpp
Doesnt help.
Is there a way to keep objects in a lib in scope? As i said, im new to arduinos oop so i might
now know something fundamental.

Hiddenvision, 2018-02-02:
> im running at 80mhz

Unless your running on battery then crank it up to 160, and feel the wind in your hair.
Happy to be corrected on that, although ALL my devices are clocked at 160.

All I can recommend is take a step back, load up the default LIBs and go from there again.
That is the great thing with these devices, so easy to "start over".
But as I have previously mentioned I am no coder so this approach may seem odd.

I started with [https://raw.githubusercontent.com/kitesurfer1404/WS2812FX](url)
the only one that worked 1st time for me.

devyte, 2018-02-02:
I think your constructor is wrong. You're initing strip with copy construction from a temp object, and the adafruit object doesn't have a specific copy constructor defined. Therefore, all it does is assign everything, including e.g. pixels. Then, the temp object is destroyed, which deallocates the same pointer, hence your strip ends up with an uninited class.
This is a typical mistake with adafruit classes: they don't implement copy constructor or assignment operators, and honestly I don't understand why so many programmers try do copy construction instead of direct construction.
Instead, do this:
```
LedStrip::LedStrip(int PIN, int NUM_LEDS) :
  strip(NUM_LEDS, PIN, NEO_GRBW + NEO_KHZ800) {
}
```
Closing as user error.

closed 2018-02-02 by devyte