esp8266/Arduino #4279: Error compiling for board NodeMCU 0.9 (ESP-12 Module. Being used by another process

PETEIOT, 2018-02-02:
Error compiling for board NodeMCU 0.9 (ESP-12 Module  
open sketch\Blink.ino.cpp: The process cannot access the file because it is being used by another process.

### Basic Infos

#### Hardware
Hardware:			Any ESP Board. ESP-01, NodeMCU 0.9 (ESP-12 module) 

### Description

Every time I try to verify or upload a sketch I get this error it does matter the sketch I use even a simple one. I have tried to just verify a sketch without the board plugged in and same thing.  I have tried uninstalling Arduino removed the esp8266 from the board manager tried a different url link for the boards manager. I can get to work on other computers

**THIS IS THE ERROR ON THE BOTTOM OF THE SCREEN.**
open sketch\Blink.ino.cpp: The process cannot access the file because it is being used by another process.
Error compiling for board NodeMCU 0.9 (ESP-12 Module).

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?
[image]

### Sketch
/*
 ESP8266 Blink by Simon Peter
 Blink the blue LED on the ESP-01 module
 This example code is in the public domain
 
 The blue LED on the ESP-01 module is connected to GPIO1 
 (which is also the TXD pin; so we cannot use Serial.print() at the same time)
 
 Note that this sketch uses LED_BUILTIN to find the pin with the internal LED
*/

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, LOW);   // Turn the LED on (Note that LOW is the voltage level
                                    // but actually the LED is on; this is because 
                                    // it is active low on the ESP-01)
  delay(1000);                      // Wait for a second
  digitalWrite(LED_BUILTIN, HIGH);  // Turn the LED off by making the voltage HIGH
  delay(2000);                      // Wait for two seconds (to demonstrate the active low LED)

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

***THIS IS THE ERROR THAT I'M GETTING ON THE BOTTOM OF THE SCREEN***

open sketch\Blink.ino.cpp: The process cannot access the file because it is being used by another process.
Error compiling for board NodeMCU 0.9 (ESP-12 Module)

lrmoreno007, 2018-02-02:
What is your operating system?
Have you rebooted the PC?
Do you have any antivirus that may be causing this?

Regards

PETEIOT, 2018-02-02:
Windows 10 and yes I have restarted several time and yes I have Mcafee livesafe that came with the computer.

tonton81, 2018-02-02:
this is a file permission issue

tonton81, 2018-02-02:
run with admin rights and fix permissions in ur IDE folder

larryk240, 2018-02-03:
I have the same problem using a NodeMCU 1.0 (ESP-12E Module)
Arduino IDE 1.8.5 on Windows 10

_Error compiling.....................
open sketch\Blink.ino.cpp: The process cannot access the file because it is being used by another process.
Error compiling for board NodeMCU 1.0 (ESP-12E Module)._
This must be a Win 10 update issue, because I wrote to this module several weeks ago. Now it doesn't work.
I can write to the Blink ESP8266 to the module with my Win7 machine using Arduino 1.6.12
Any suggestions ??

PETEIOT, 2018-02-03:
So it looks like the virus software that came on my computer Macfee was blocking it so I removed it and it seems to be working fine. Thanks Irmoreno007 for the suggestion.

larryk240, 2018-02-04:
Yep, McAfee was the issue with my system also. Temporarily disabled the scan and the firewall, and all is well.
However, McAfee came with the computer, and I know for sure that it didn't use to cause problems.
Thanks go out to Irmoreno007 and PETEIOT

tonton81, 2018-02-04:
if you add that folder to the exception list of mcaffee, it wont touch them ever again...

PETEIOT, 2018-02-04:
Yah I been using it for a few weeks with out any problems and it just stopped working one day it must have been an update from Macfee.

closed 2018-02-04 by PETEIOT

caiobertacco, 2018-02-17:
I had the same problem, thank you all!!

Kalai1227, 2018-02-21:
I am having the same problem in nodemcu 12E.I am using arduino ide 1.8.5 and getting this error again and again.i have tried all possible things but it failed.The error is
Archiving built core (caching) in: C:\Users\KALAID~1\AppData\Local\Temp\arduino_cache_744989\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,FlashSize_4M1M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,UploadSpeed_9600_0be5937afd45902e722a34277baa72a9.a
Sketch uses 247063 bytes (23%) of program storage space. Maximum is 1044464 bytes.
Global variables use 32868 bytes (40%) of dynamic memory, leaving 49052 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

Kalai1227, 2018-02-21:
please help to sort the above issue

lrmoreno007, 2018-02-21:
This is the important message: error: **espcomm_open failed**
Check that you have drivers for your board and your OS is detecting it.

Read this:
https://github.com/esp8266/Arduino/blob/master/doc/faq/a01-espcomm_sync-failed.rst#initial-checks
*1.- Arduino IDE is unable to open your serial port. Check if you have selected port where your module is connected to.*

And you have selected 9600bauds, select 115200.

Kalai1227, 2018-02-22:
I have tried this one but getting same error

lrmoreno007, 2018-02-22:
https://arduino-esp8266.readthedocs.io/en/latest/faq/a01-espcomm_sync-failed.html#i-m-stuck

karunanayaka, 2018-03-13:
i cant compile the code when select the NodeMCU 1.0 (ESP-12E Module) 
error code is
Arduino: 1.8.4 (Windows 8.1), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Imalka Karunanayaka\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Imalka Karunanayaka\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Imalka Karunanayaka\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=115200 -ide-version=10804 -build-path C:\Users\IMALKA~1\AppData\Local\Temp\arduino_build_705770 -warnings=all -build-cache C:\Users\IMALKA~1\AppData\Local\Temp\arduino_cache_851299 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Imalka Karunanayaka\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\Imalka Karunanayaka\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.13 -prefs=runtime.tools.mkspiffs.path=C:\Users\Imalka Karunanayaka\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.2.0 -verbose C:\Users\Imalka Karunanayaka\Documents\Arduino\libraries\Arduino-master\libraries\esp8266\examples\Blink\Blink.ino

Board nodemcuv2 (platform esp8266, package esp8266) is unknown

Error compiling for board NodeMCU 1.0 (ESP-12E Module).

please help me!!!

pieman64, 2018-03-13:
@karunanayaka this is not a site to ask for assistance regarding how to install boards and fix compiler errors. The purpose is to reports bugs in the code. Check out the ESP8266 forum.

karunanayaka, 2018-03-13:
i though that this can be a bug, because its not even compileing basic
examples provided. i have tried with changing IDE versions, installing
drivers, installing board managers with different versions, uninstalling
virus guards etc.
but nothing happens
anyway thank you

devyte, 2018-03-13:
Not a bug, it's a windows thing. You have to delete the tmp dir where the built binaries are put, or better yet, do a full clean install from scratch. 
I don't know where the tmp dir is, I develop on Linux within a VM, so you'll have to figure that out yourself.

pieman64, 2018-03-13:
@karunanayaka the ESP8266 forum guys will tell you to try a portable install of the IDE as you then have full control over the system rather than leaving it to the whims of M$.

lrmoreno007, 2018-03-13:
@karunanayaka https://arduino-esp8266.readthedocs.io/en/latest/faq/a04-board-generic-is-unknown.html

karunanayaka, 2018-03-13:
> @karunanayaka <https://github.com/karunanayaka> https://arduino-esp8266.
> readthedocs.io/en/latest/faq/a04-board-generic-is-unknown.html

oh god
this is the solution
thanks  ZaPpInG and  theCHARioT

KhamkarSagar, 2018-06-02:
yes, I had same problem:error **compiling for board nodemcu 1.0 (esp-12e module)** .It solved after deleting Mcaffee Anti virus .

r4adi0GaX2, 2019-05-11:
Hi, I'm having this error message
[image]
:

Arduino: 1.8.8 (Mac OS X), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 9600"

fork/exec /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++: no such file or directory
Error compiling for board NodeMCU 0.9 (ESP-12 Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Armilsyam, 2020-09-10:
Arduino: 1.8.5 (Windows 10), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Board nodemcu (platform esp8266, package esp8266) is unknown

Error compiling for board NodeMCU 0.9 (ESP-12 Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

benjudePremier, 2020-11-24:
Hi everyone, who can help me fix this issue, i tried multiple time  to connect with thingsboard  and it does not work...
exit status 1
Error compiling for board NodeMCU 0.9 (ESP-12 Module).