esp8266/Arduino #4281: 2.4.0 readString() Timeout

bstsoftorg, 2018-02-03:
### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	2.3.0-rc2, 2.4.0

### Description

In 2.4.0 a function "readString" for too long runs.
tell me what to fix in the core for re-test?

### Installed 2.3.0-rc2
First the ESP module sends
the answer HTTP_CODE_OK
```cpp
  int httpCode = Client.sendRequest("GET", s);
```

The second module performs ESP
```cpp
  uint32_t xxx = millis();
  s = HTTP.client().readString();
  Serial.println(String() + F("ms_port_out1=")+String(millis() - xxx));
  "code..."
  Serial.println(String() + F("ms_port_out2=")+String(millis() - xxx));
  HTTP.send(200, text_html, s);
  Serial.println(String() + F("ms_port_out3=")+String(millis() - xxx));
```
### Debug Messages "second module"

```
ms_port_out1=1093
ms_port_out2=1093
ms_port_out3=1117
```

### Installed 2.4.0
First the ESP module sends
the answer [HTTP] GET_port_out failed, error: (-11) read Timeout
```cpp
  int httpCode = Client.sendRequest("GET", s);
```

The second module performs ESP
```cpp
  uint32_t xxx = millis();
  s = HTTP.client().readString();
  Serial.println(String() + F("ms_port_out1=")+String(millis() - xxx));
  "code..."
  Serial.println(String() + F("ms_port_out2=")+String(millis() - xxx));
  HTTP.send(200, text_html, s);
  Serial.println(String() + F("ms_port_out3=")+String(millis() - xxx));
```
### Debug Messages "second module"

```
ms_port_out1=5001
ms_port_out2=5002
ms_port_out3=5006
```

### Settings in IDE

Module:  			NodeMCU 0.9
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

devyte, 2018-02-04:
@bstsoftorg 2.3.0-rc2 was a release candidate for 2.3.0, so it is beyond ancient. Did you mean 2.4.0-rc2?

I believe that some fixes that may be relevant to this were merged after 2.4.0 release. Please retest with latest git core. Instructions for installing are in readthedocs. Don't forget to uninstall all release versions you have via package manager, and to restart the IDE after installing git.

bstsoftorg, 2018-02-04:
I did a fully clean machine.
Were removed folder "Arduino15" and is made of 100% new install. In 2.4.0 the problem, had to roll back to 2.3.0-rc2.

PS Also identified the problem with the FiWi reconnect with ESP to ESP. After the break the client the ESP cannot connect to the ESP point.

devyte, 2018-02-05:
@bstsoftorg  a "fully clean machine" is not what I meant. I meant install latest git instead of 2.4.0 release.
Also, please add full sketches for both ESPs. After a second look, I suspect you're doing something wrong.

bstsoftorg, 2018-04-26:
Too bad your advice didn't help. The problem is that in newer versions you cannot read the buffer that was processed in Parsing.cpp. 
Decision, you need to use
    Client.sendRequest ("POST", s);          
and
    s = HTTP.arg(F ("plain"));

closed 2018-04-26 by bstsoftorg