esp8266/Arduino #4282: getVcc() different in 2.3.0 and 2.4.0

Nigel9455426, 2018-02-03:
Can anyone explain why the getVcc() is different in Firmware 2.4.0 from 2.3.0.
Also with the resistor divider removed and nothing connected to ADC it still reports the wrong voltage.
Nigel

```
// Test for reading internal supply voltage via analogue input
// Board: NodeMCU Amica, http://amzn.eu/05ramaW
// Firmware esp8266 by ESP8266 Community version as below

// Note R13 (220K) & R14 (100K) make a potential divider on the analogue input (ADC). 
// Instructions state for ESP.getVcc to work then these R13 & R14 should be removed.
// Previously Firmware 2.3.0 gave the correct reading when the resistor divider 
// was fitted and the 3.3 supply was connected to ADC_EX.
// Firmware 2.4.0 appears to break both the resistor divider fitted and not fitted results. 

ADC_MODE(ADC_VCC); // Analogue read battery voltage

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  delay(200);
  Serial.println("Hello");
  float supplyVolts = ((float)ESP.getVcc() / 1024.0f) ;
  Serial.print(supplyVolts);
}

void loop() {
  // put your main code here, to run repeatedly:

}

// Results

// Firmware 2.3.0
// 220K R13 & 100K R14 fitted & external link fitted from ADC_EX to 3V3
// supplyVolts read 3.29, actual supply voltage 3.281 - Correct, about the same voltage.
//
// Firmware 2.4.0
// 220K R13 & 100K R14 fitted & external link fitted from ADC_EX to 3V3
// supplyVolts read 3.59, actual supply voltage 3.279 - Different / Wrong!
//
// Firmware 2.4.0-rc2
// 220K R13 & 100K R14 fitted & external link fitted from ADC_EX to 3V3
// supplyVolts read 3.58, actual supply voltage 3.278 - Different / Wrong!
//
// Firmware 2.4.0-rc2
// Both 220K R13 & 100K R14 REMOVED, NO external link fitted to ADC_EX, ADC_EX & ADC unconnected.
// supplyVolts read 3.43, actual supply voltage 3.275 - Different / Wrong!
//
// Firmware 2.3.0
// Both 220K R13 & 100K R14 REMOVED, NO external link fitted to ADC_EX, ADC_EX & ADC unconnected.
// supplyVolts read 3.15, actual supply voltage 3.275 - Different / Wrong!
```

5chufti, 2018-02-03:
Hi,
1) the adc in the esp8266 is .... simple.
2) actually info about adc from espressif is .... changing:
depending on datasheet revison and/or faq
it is better to read with/without wifi active...
resolution may be 1/1023 or 1/1024 
not clear if system_get_vdd33 returns real Volts or ADC counts
3) 2.3.0 is based on NONOS_SDK 1.5.3 and release_note to NONOS_SDK_V1.5.4_16_05_20 states:
"8. Revised the issue that API system_adc_read and system_get_vdd33 may return wrong value."

but one thing is definitely clear in all datasheets and faqs: ADC pin has to stay unconnected if using system_get_vdd33

devyte, 2018-02-03:
Previous comment is correct: in release 2.4 we migrated to a new sdk, and the sdk release notes state that there was a change. The difference is not due to the code in this repo (see src code for ESP.getVcc() )
Closing.

closed 2018-02-03 by devyte

Nigel9455426, 2018-02-04:
Hi.
Thanks for the replies.
So in summary, for anyone else tripping over this conversation.
The different voltage problem between firmware versions is real but note the following:
1. The 'ESP8266 Community' code inside Firmware 2.4.0 is correct. - Hence this issue has been 'Closed'.
2. The SDK NONOS v1.5.4 that the Firmware 2.4.0 is built upon, now gives a different result for reading system_get_vdd33. This is _possibly_ known by the makers of the SDK NONOS v1.5.4.

Therefore the issue is with the makers of the SDK, not the 'ESP8266 Community'.

Thank you to all the contributors who have made this repository. Using your firmware has given me hours of enjoyment. Keep up the good work, it is greatly appreciated.

Thanks
Nigel