esp8266/Arduino #4283: [2.4.0] Communication issue between ArduinoOTA and espota.py

eXtatic-GitHub, 2018-02-03:
### Basic Infos
I'm not a python developer and also new to all this here so please have mercy.
While migrating my project from 2.3.0 to 2.4.0 I experienced the OTA to behave different.
espota.py is not closed like it was in 2.3.0

#### Hardware
Hardware:			ESP-07
Core Version:      	2.4.0

### Description

**ArduinoOTA**
*espota.py* is waiting for an "OK" from *ArduinoOTA library* after upload was successful but only receives "K", thus not closing connection.
The "O" seems to get lost somewhere.

With 2.3.0 version of *espota.py* this was not an issue since *espota.py* closed anyway.
#4053 introduced this new handling

**Hack**
By trial and error I just added a leading character to the char array ArduinoOTA returns and it was working:
[image]

**Fix?**
Next try was then to provide a String instead of char*
[image]

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1 MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		OTA
Reset Method:		ck

### Sketch

basicOTA.ino

### Debug
*espota.py 2.3.0* - prints error "K" but flashed successful
[image]

*espota 2.4.0* - hangs there forever but sketch is updated and ESP restarts fine (I canclled execution some minutes after)
[image]

*espota 2.4.0 + ArduioOTA hack* - nice and smooth
[image]

*espota 2.4.0 + ArduioOTA fix?* - nice and smooth
[image]

devyte, 2018-02-03:
There have been fixes merged since 2.4 release that might be relevant. Please test with latest git.

eXtatic-GitHub, 2018-02-04:
Hey, latest git works.
Eventhough I did spend some time on it prior to open a new issue I wasn't able to find a related issue or pr.
Maybe someone can reference it.

Thank you @devyte

devyte, 2018-02-04:
Closing as already fixed.

closed 2018-02-04 by devyte

washcroft, 2018-12-08:
@devyte I don't believe this issue has been fixed... all the recent issues with OTA seems to be caused by the changes @d-a-v made to the **espota.py** OTA script in the IDE, not the ESP8266 OTA code itself.

https://github.com/esp8266/Arduino/blob/master/tools/espota.py#L163 assumes it will always be more than 1000 bytes of data received/written (reads 4 bytes of numerical characters) when the device responds with the length after each packet, however in practice this is not always the case.

For instance, even if all the chunks of data of an upload are > 1000 bytes (usually 1460 depending on MTU), there is no guarantee the last chunk would be, that depends on the total size of the binary being uploaded.

If this last chunk is say 900 bytes, L163 will still read 4 bytes, first 3 would be "900" and the fourth would be "O" from the "OK" that https://github.com/esp8266/Arduino/blob/master/tools/espota.py#L180 is waiting for.

washcroft, 2018-12-08:
@devyte I have no PR, I’m not a python developer - but I can understand the logic and I am merely pointing out why so many people are having issues with OTA uploads. Take it or leave it.

d-a-v, 2018-12-08:
@washcroft did you actually run into the issue ?

washcroft, 2018-12-09:
@d-a-v yes, that’s how I ended up investigating the cause

d-a-v, 2018-12-09:
Thanks for that!