esp8266/Arduino #4284: NodeMcu never passing WiFi.status

Ethan17155, 2018-02-03:
### Basic Info

#### Hardware
Hardware:		NodeMcu esp8266 Amica
Core Version:      	?

### Description

I just got my first esp8266 board and tried to upload the WebServer library. And I upload the sketch and on the serial monitor it tells me it never connects. I have tried every single upload speed thinking I may be going to fast for the board but none of them work.

### Settings in IDE

Module:  			esp 8266 built in
Flash Size: 		?
CPU Frequency:       80Mhz
Flash Mode:		?
Flash Frequency:	?
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp

#include <ESP8266WiFi.h>

const char ssid[] = "************";
const char password[] = "*************";

WiFiServer server(80);

void setup() {
  Serial.begin(74880);
  while(!(Serial.available()));
  delay(10);

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println("ssid");
  delay(10);
  
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid,  password);

  //This is where it get stuck and just prints never ending dots <----------------------------------------
  while (WiFi.status() != WL_CONNECTED) {  //<-------------------
    delay(500);                                                 //<-------------------
    Serial.print(".");                                           //<-------------------
  }
  Serial.println("");
  Serial.println("WiFi connected");
  
  server.begin();
  Serial.println("Server started");

  Serial.println(WiFi.localIP());
}

void loop() {
  WiFiClient client = server.available();
  if (!client) {
    return;
  }
  
  Serial.println("new client");
  while(!client.available()){
    delay(1);
  }
  String req = client.readStringUntil('\r');
  Serial.println(req);
  client.flush();
  
  if (req.indexOf("/hi") != -1)
  {client.print("hi");}
}

```
Serial COM13
```

Connecting to *************
........................................................................And so on

Pablo2048, 2018-02-03:
Try to call wifi_station_get_connect_status() and check the reason why connection request failed.

tonton81, 2018-02-03:
sounds like bad ssid/pass

Ethan17155, 2018-02-03:
I have tried all 4 of my Wifi connections, none of them work.

Pablo2048, 2018-02-03:
No info, no help :-( ... Do as i wrote and give us what wifi_station_get_connect_status() returns.

Ethan17155, 2018-02-03:
How do I use the wifi_station_get_connect_status()?

Pablo2048, 2018-02-03:
Did you try at least to search the web, or are you too lazy?

extern "C" {
  #include "user_interface.h"
}
....
  _status = wifi_station_get_connect_status();
....

Ethan17155, 2018-02-03:
I tried using a Serial.print(WiFi.status()); inside the: while (WiFi.status() != WL_CONNECTED) {...} and it comes back as 6 for a while but then goes to 1.

Pablo2048, 2018-02-03:
wifi.status gives only arduino shield compatible results (why i'm wasting my time to suggest something when nobody listen :-( ) but in this case it means that you have written bad ssid to ssid[] - what exactly did you put into ssid[]?

Ethan17155, 2018-02-03:
my wifi name

Ethan17155, 2018-02-03:
my ssid and wifi name is the same

Ethan17155, 2018-02-03:
I did a Serial.print(wifi_station_get_connect_status()); and it printed 1 for a while then went to 3.

Pablo2048, 2018-02-03:
So you can interpret by yourself - just take a look into user_interface.h for station_status_t. It seems like status is STATION_NO_AP_FOUND. What is the type of your AP?

tonton81, 2018-02-03:
wrongly typed, maybe hidden ssid, or 5ghz AP, theres reasons why its not seen

Ethan17155, 2018-02-03:
my AP is 2.4ghz

Pablo2048, 2018-02-03:
You are really not giving much info, so briefly:
write sketch which scan wifi for available AP's (STFW for examples) and try to figure out why your ESP did not see your AP.

Ethan17155, 2018-02-03:
WiFiScan is not picking up my wifi or mobile hotspot.

Pablo2048, 2018-02-03:
Standard, channel, SSID?

Ethan17155, 2018-02-03:
Here is all the info on the WiFi:

SSID:                               **************
Protocol:	                        802.11n
Security type:	                WPA2-Personal
Network band:	        2.4 GHz
Network channel:	        6
IPv4 address:	                192.168.1.57
IPv4 DNS servers:	        192.168.1.1
Manufacturer:   	        Intel Corporation
Description:	                Intel(R) Dual Band Wireless-AC 8260
Driver version:	                19.50.1.6
Physical address (MAC):	44-85-00-FB-8E-D9

Pablo2048, 2018-02-03:
http://www.esp8266.com/viewtopic.php?f=6&t=1612 hope that you really don't have SSID as ************ (really don't know what is so secret on broadcasted SSID)

Pablo2048, 2018-02-03:
probably more relevant https://github.com/esp8266/Arduino/issues/3941 ...

tonton81, 2018-02-03:
with the latest git im able to connect to my iphone7+’s AP without issue, and it’s a dual band phone. it also has no issues connecting to a crappy Bell router either :)

Ethan17155, 2018-02-03:
The Board itself does not read newer WiFi connections, all my connections are less than a year old so I am thinking that is the reason.

Ethan17155, 2018-02-03:
more specifically: 802.11n AP.

tonton81, 2018-02-03:
no, its not, did wifiscan see any APs at app?

tonton81, 2018-02-03:
*all

Ethan17155, 2018-02-03:
no

Ethan17155, 2018-02-03:
I might have a bad board?

tonton81, 2018-02-03:
it sounds like it, can you try another?

Ethan17155, 2018-02-03:
It should support it because the wifi block on the board says: 802.11b/g/n

tonton81, 2018-02-03:
yeah but if you dont see any APs, something is definately wrong with the chip i would presume

tonton81, 2018-02-03:
i get 20 APs listed from my livingroom
are you in the middle of nowhere with only ur two APs? :D

Ethan17155, 2018-02-03:
This is my first NodeMcu Esp8266 board, but I have used an Adafruit Feather Huzzah and it worked perfectly fine. So I will buy a Huzzah and see if that works.

Ethan17155, 2018-02-03:
nowere-ish, My computer picks up our Ethernet, our 4 Wifi connections, our printer, and two others wifies near us that are very weak.

closed 2018-02-03 by Ethan17155