esp8266/Arduino #4285: spiffs seek(..., SeekEnd) doesn't work

brianjmurrell, 2018-02-03:
### Basic Infos

#### Hardware
Hardware:			NodeMCU
Core Version:      	2.4.0 I believe

### Description
`f.seek(80, SeekEnd);` doesn't actually seek from the end

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		4MB/3MB
CPU Frequency:		80Mhz
Flash Mode:			not sure
Flash Frequency:	not sure
Upload Using:		SERIAL
Reset Method:		not sure

### Sketch

```cpp

#include <FS.h>

void setup() {
  SPIFFS.begin();
  File f = SPIFFS.open("/stats.txt", "r");
  f.seek(80, SeekEnd);
  Serial.println(f.size());
  Serial.println(f.position());
  f.seek(f.size() - 80);
  Serial.println(f.position());
}

void loop() {
// NA
}
```

### Debug Messages

```
2041660
2041660
2041580
```

As you can see the ` seek(80, SeekEnd)` didn't actually work and seeked to the end, not 80 bytes from the end.  The `seek(f.size() - 80)` did what the ` seek(80, SeekEnd)` should have done.

igrr, 2018-02-03:
Probably duplicate of https://github.com/esp8266/Arduino/issues/2597.

devyte, 2018-02-04:
It is. I'll pull the other one into 2.4.1 and make a PR.
Closing this as duplicate.

closed 2018-02-04 by devyte

devyte, 2018-02-04:
@brianjmurrell please try #4287 .