esp8266/Arduino #4289: AWS IoT integration issue - constantly resetting

alexstatie, 2018-02-04:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-8266Mod https://ae01.alicdn.com/kf/HTB1xcvXRXXXXXbvapXXq6xXFXXXs/10pcs-lot-ESP8266MOD-Wireless-Module-NodeMcu-Lua-WIFI-Development-Board-Based-ESP8266-CP2102-with-Pcb-Antenna.jpg
Core Version:      	?2.1.0-rc2?

### Description

Problem description

Resets after a few mqtt calls using AWS IoT Services when calling update_shadow

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	40Mhz
Upload Using:		 SERIAL
Reset Method:		 nodemcu

### Sketch

```cpp

//#include <EmonLib.h>
#include <ESP8266WiFi.h>

#include <AmazonIOTClient.h>
#include "Esp8266AWSImplementations.h"

//char* ssid = "xxxxxxxx";
//char* password = "xxxxxxx"; 

//EnergyMonitor emon1; 

Esp8266HttpClient httpClient;
Esp8266DateTimeProvider dateTimeProvider;

AmazonIOTClient iotClient;
ActionError actionError;

void initWLAN()
{
  WiFi.begin("Tenda_58BCB0", "Parola123");
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
  }
}

void initAWS()
{
  iotClient.setAWSRegion("eu-west-2");
  iotClient.setAWSEndpoint("amazonaws.com");
  iotClient.setAWSDomain("a2s172jeixrz3.iot.eu-west-2.amazonaws.com");
  iotClient.setAWSPath("/things/PowerMeter/shadow");
  iotClient.setAWSKeyID("AKIAJBF4DW5FUGVCMITA");
  iotClient.setAWSSecretKey("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");
  iotClient.setHttpClient(&httpClient);
  iotClient.setDateTimeProvider(&dateTimeProvider);
}

void setup() {
  

Serial.begin(115200);

//emon1.current(A0, 111.1);

  delay(10);
  Serial.println("begin");
  initWLAN();
  Serial.println("wlan initialized");
  initAWS();
  Serial.println("iot initialized");

}

void loop() {

  
  char shadow[100];
  strcpy(shadow, "{\"state\":{\"reported\":{\"test_value1\":1234, \"test_value2\":2345}}}");
  
  //Serial.println("Trying to send data");
  //Serial.print(shadow);
  iotClient.update_shadow(shadow, actionError);
  //char* result = iotClient.update_shadow(shadow, actionError);
  //Serial.print(result);
  Serial.print("ok");
  
  delay(10000);

}
```

### Debug Messages

```
0x401004d8: malloc at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1668
0x4010020c: _umm_free at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1291
0x40100690: free at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1737
0x4020646c: operator delete[](void*) at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/abi.cpp line 84
0x40204071: MinimalString::innerDelete() at C:\Program Files (x86)\Arduino\libraries\AmazonIOTClient/AWSFoundationalTypes.cpp line 49
0x402040d5: MinimalString::operator=(MinimalString const&) at C:\Program Files (x86)\Arduino\libraries\AmazonIOTClient/AWSFoundationalTypes.cpp line 49
0x40206458: operator delete(void*) at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/abi.cpp line 84
0x40203f2c: AWSClient4::createRequest(MinimalString&) at C:\Program Files (x86)\Arduino\libraries\AmazonIOTClient/AWSClient4.cpp line 197
0x4021fde9: SHA256_Final at crypto/sha256.c line 273 (discriminator 3)
0x4021c248: hmac_sha256_v at crypto/hmac.c line 165
0x40100ece: pp_post at ?? line ?
0x401042b0: lmacTxFrame at ?? line ?
0x401034cf: lmacRecycleMPDU at ?? line ?
0x40103932: lmacRecycleMPDU at ?? line ?
0x40103416: lmacProcessTxSuccess at ?? line ?
0x40102297: wDev_ProcessFiq at ?? line ?
0x40101f94: wDev_ProcessFiq at ?? line ?
0x4021e594: SHA256_Process at crypto/sha256.c line 163
0x4021fd40: SHA256_Update at crypto/sha256.c line 231
0x40232dbf: pp_attach at ?? line ?
0x40232e0e: pp_attach at ?? line ?
0x4021fde9: SHA256_Final at crypto/sha256.c line 273 (discriminator 3)
0x4021c248: hmac_sha256_v at crypto/hmac.c line 165
0x4021ae36: bi_free at crypto/bigint.c line 672
0x4021add4: bi_depermanent at crypto/bigint.c line 672
0x4010020c: _umm_free at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1291
0x4010020c: _umm_free at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1291
0x40100690: free at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1737
0x4021aef4: bi_terminate at crypto/bigint.c line 672
0x4010020c: _umm_free at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1291
0x4010020c: _umm_free at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1291
0x4010020c: _umm_free at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1291
0x40100690: free at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1737
0x40206458: operator delete(void*) at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/abi.cpp line 84
0x40203644: SSLContext::unref() at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 110
0x4020366e: WiFiClientSecure::~WiFiClientSecure() at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 415
0x40204485: Esp8266HttpClient::send(char const*, char const*, int) at C:\Program Files (x86)\Arduino\libraries\AmazonIOTClient/ESP8266AWSImplentations.cpp line 54
0x401004d8: malloc at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1668
0x40206444: operator new[](unsigned int) at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/abi.cpp line 84
0x40206520: esp_yield at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/core_esp8266_main.cpp line 57
0x40204175: AmazonIOTClient::update_shadow(MinimalString, ActionError&) at C:\Program Files (x86)\Arduino\libraries\AmazonIOTClient/AmazonIOTClient.cpp line 19
0x40202250: loop at C:\Users\Alexandru Statie\Desktop\energymon\energymon/energymon.ino line 80
0x402025d4: ESP8266WiFiSTAClass::status() at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 483
0x40205df8: Print::println() at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/Print.h line 93
0x40205e1c: Print::println(char const*) at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/Print.h line 93
0x40206541: esp_schedule at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/core_esp8266_main.cpp line 57
0x4020656c: loop_wrapper at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/core_esp8266_main.cpp line 57
0x40100718: cont_norm at C:\Users\Alexandru Statie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/cont.S line 109

```

tarzan115, 2018-02-05:
I read the log. I guess it seems to be like this issues https://github.com/esp8266/Arduino/issues/4207 and discussion in https://github.com/esp8266/Arduino/issues/4134
try with latest git version maybe can solve your problem.

devyte, 2018-07-13:
This looks like oom or mem corruption.
Closing due to merge of bearssl (axtls will be deprecated in favor of bearssl).

closed 2018-07-13 by devyte

silva05flaco27-sudo, 2025-07-19:
- open