esp8266/Arduino #4292: Adafruit Huzzah breakout removed from list, replaced by Feather

tonton81, 2018-02-05:
Adafruit has 2 Huzzah's. One is a /Feather/. The other is a Breakout. It seems when you added the feather you also removed the HUZZAH breakout.

d-a-v, 2018-02-05:
The boards generator is originally based on previous boards.txt.
There was only one huzzah in 2.4.0rc2 ([link](https://raw.githubusercontent.com/esp8266/Arduino/2.4.0-rc2/boards.txt)).
I can't find the other in boards.txt's history. Do you have more details ?

tonton81, 2018-02-05:
i upgraded from ide 1.8.3 to 1.8.5 and noticed the feather listed, it wasnt before, im using a huzzah breakout, not that it matters they both function the same and are almost named the same :D

tonton81, 2018-02-05:
ahh even though i reinstalled git i forgot to reinstall the staging boards manager! doh! i see it listen there:

name | "Adafruit HUZZAH ESP8266 (ESP-12)"
-- | --

:)

closed 2018-02-05 by tonton81

ladyada, 2018-04-27:
hiya just to clarify, they're identical board definitions. you can use either for the breakout and feather