esp8266/Arduino #4295: ESP12 dont start

kamilbystryk, 2018-02-05:
#### Hardware
Hardware:			ESP-12
Core Version:      	2.4.0

### Description

I have problem with program starts on ESP12 device (ESP8266MOD, producer - DOIT). After start i see don't readable characters on serial. Program starts only, when I choose "generic ESP8255 module". I check - inside module is ESP8266 and SPI flash. No one sketch starts on ESP8266 generic stettings (e.g. blink).
On AI thinker ESP12 program run without problems.
I read settings by  Esspressiff download tool:
Problem module:
flash vendor:
C8h : GD
flash devID:
4016h
QUAD;32Mbit
crystal:
26 Mhz

"good" module:
flash vendor:
EFh : WB
flash devID:
4016h
QUAD;32Mbit
crystal:
26 Mhz

How can I solve problem?

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 	4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

RudyFiero, 2018-02-06:
This is not the place for your problem. This is for issues regarding the ESP8266 Arduino software. Issues are bugs or feature requests. 

Have a look at this. 
https://arduino-esp8266.readthedocs.io/en/latest/boards.html

http://www.esp8266.com/ is a more appropriate place for asking about new user problems. But try and be specific in what you have done. The information you provided here is inadequate. Explaining how you have the module connected and powered would be a good start.

devyte, 2018-02-06:
This gives a hint:
flash vendor:
C8h : GD

Closing as duplicate of #3616 .

closed 2018-02-06 by devyte

Karpesh, 2018-02-18:
I solved the problem for my case.
This flash has DIO mode, not QIO.  I switched the mode in my makefile and all works fine now.