esp8266/Arduino #4296: 2.4.0 and stepper motor crash and strange things

Tarthesos, 2018-02-05:
Hi to all.

I really hope you can help me find a solution.

Thanks in advance :)

### Basic Infos

#### Hardware
Hardware:			ESP-12E Module
Core Version:      	2.4.0

### Description

I've already read a bunch of threads:
https://github.com/esp8266/Arduino/issues/2558
https://github.com/esp8266/Arduino/pull/2533/files
https://github.com/esp8266/Arduino/issues/2083
... and more, but none solved my problem.

Using the stepper motor in general cause a crash. Even the example "stepper_oneRevolution." crashes while performing the counterclock rotation. Using delay(...) or yield() or system_soft_wdt_feed() is useless the problem persists ... mostly during the counterclock rotation a crash happens.

The strange thing is that after the crash if I "restart" (unplug and plug back in the usb port) the NoideMCU begins to work and stepper continuously turns without crashing, but the delay after the counterclock rotation i totally wrong.

### Settings in IDE

Producer:  		AZ-Delivery
Module:  			NodeMCU ESP8266 V2.0 Amica
Flash Size: 		4MB
CPU Frequency:	80Mhz
Upload Using:		SERIAL

Here my code:

### Sketch

```cpp

#include <Stepper.h>

const int stepsPerRevolution = 1500;  // change this to fit the number of steps per revolution

// initialize the stepper library on pins 8 through 11:
// Stepper myStepper(stepsPerRevolution, 8, 10, 9, 11); // keep for arduino
Stepper myStepper(stepsPerRevolution, D4, D2, D3, D1); // Initialize pins on the NodeMCU

void setup() {
  // set the speed at 20 rpm:
  myStepper.setSpeed(20);
  // initialize the serial port:
  Serial.begin(115200);
}

void loop() {
  // step one revolution  in one direction:
  yield();
  Serial.println("clockwise 01");
  myStepper.step(stepsPerRevolution);
  delay(500);

  // step one revolution in the other direction:
  yield();
  Serial.println("counterclockwise 01");
  myStepper.step(-stepsPerRevolution);
  delay(500);
  
  // step one revolution  in one direction:
  yield();
  Serial.println("clockwise 02");
  myStepper.step(stepsPerRevolution);
  delay(500);

  // step one revolution in the other direction:
  yield();
  Serial.println("counterclockwise 02");
  myStepper.step(-stepsPerRevolution);
  delay(1000);
}
```
And that's qhat the Serial monitor reports:

### Debug Messages

```
Soft WDT reset

ctx: cont 
sp: 3ffefb00 end: 3ffefd00 offset: 01b0

>>>stack>>>
3ffefcb0:  3ffe8870 00000001 3ffeea88 402022cc  
3ffefcc0:  402010b2 00000095 3ffeea88 40202447  
3ffefcd0:  3fffdad0 3ffeeca8 3ffeea88 402020b0  
3ffefce0:  feefeffe 00000000 3ffeecc4 4020284c  
3ffefcf0:  feefeffe feefeffe 3ffeece0 40100108  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

I used "Exception Decoder" and it reported this;
### Debug Messages

```
Decoding 6 results
0x402022cc: Stepper::stepMotor(int) at D:\Programmi\Arduino\libraries\Stepper\src/Stepper.cpp line 283
0x402010b2: delay at C:\Users\Roberto\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/core_esp8266_wiring.c line 54
0x40202447: Stepper::step(int) at D:\Programmi\Arduino\libraries\Stepper\src/Stepper.cpp line 198
0x402020b0: loop at D:\GitArea\Arduino\Stepper_OneRevolution_modded\stepper_oneRevolution/stepper_oneRevolution.ino line 41
0x4020284c: loop_wrapper at C:\Users\Roberto\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/core_esp8266_main.cpp line 57
0x40100108: cont_norm at C:\Users\Roberto\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/cont.S line 109

```

Have you some clue about it?

Lastly ... sorry for may bad english ^^"

lrmoreno007, 2018-02-05:
The stepper.h library is made for Arduino and is not adapted for ESP8266. There is a PR that you should try, but in any case I think this is not the best place.

Take a look at Microstepping example
https://github.com/Attila-FIN/Stepper/tree/de220f1e5010250a4ac376fee9502dc8170bfa27

Tarthesos, 2018-02-05:
First of all, thanks for the reply .

I'll look at the exemple as soon as possible.

About the number of steps, i just used the default in the example for this thread. Thuth be told I've used other values too, but it came out weirdly, or the stepper was turning slowly or faster or it didn't turn at all (but it was for too high values for sure). My point in this case is, if i set the speed (20 rpm) why the number of steps is influencing how fast the stepper spins?

Another thing is i've found sites stating that the stepper i am using needs 4096 steps for a rotation, i was skeptical but even tryed that value tand it resulted in the stepper not being able to move and ticking.

lrmoreno007, 2018-02-05:
La velocidad y el número de pasos están relacionados, en stepper.cpp con la variable llamada step_delay que se calcula con esto:
`` `
void Stepper :: setSpeed ​​(long whatSpeed) {
  this-> step_delay = 60L * 1000L * 1000L / this-> number_of_steps / whatSpeed;
}
`` `
El movimiento de pasos solo se produce si ha transcurrido el retraso adecuado y la biblioteca utiliza un comando "while", Y ESTE ES EL PROBLEMA, porque WDT se produce con mucho tiempo. ESP8266 es un poco especial y necesita tener control cada cierto tiempo, la biblioteca debe hacerse con banderas en lugar de un "while" (o método similar), para permitir el control del ESP8266.

Un remedio es usar el rendimiento dentro de ese "while", para esto, edite el archivo stepper.cpp search para la línea 196 e inserte yield (); o use el PR previo que le dije y elimine las líneas 413 y 474, allí puede ver el rendimiento también y una explicación.

---

Volviendo al tema de las revoluciones por minuto, 1500 pasos / rev son demasiados ... es 360/1500 = 0.24 grados por paso. NEMA17 tiene 200 pasos o 1.8 grados. Cuéntanos qué es el paso a paso y trato de buscar sus características.

devyte, 2018-02-05:
Closing as off topic.

closed 2018-02-05 by devyte