esp8266/Arduino #4298: SD Card on SPI0 pins

usonix, 2018-02-05:
Hi, I'm new here and also not very experienced in programming, I'm working with esp 12E, through the Arduino IDE. I read in a snippet of the documentation and saw that it is possible to use SPI0. What I need is to use the SD card on the SPI0 pins, leaving the SPI1 pins free for other applications, if not ask too much, could someone publish a simple example code how to do this?

Hiddenvision, 2018-02-06:
SPI0 is the Flash memory .?
You want to Boot from SD.?

usonix, 2018-02-08:
No, I'm referring to the following excerpt from the documentation:

SPI
SPI library supports the entire Arduino SPI API including transactions, including setting phase (CPHA). Setting the Clock polarity (CPOL) is not supported, yet (SPI_MODE2 and SPI_MODE3 not working).

The usual SPI pins are:

MOSI = GPIO13
MISO = GPIO12
SCLK = GPIO14

There’s an extended mode where you can swap the normal pins to the SPI0 hardware pins. This is enabled by calling SPI.pins(6, 7, 8, 0) before the call to SPI.begin(). The pins would change to:

MOSI = SD1
MISO = SD0
SCLK = CLK
HWCS = GPIO0

This mode shares the SPI pins with the controller that reads the program code from flash and is controlled by a hardware arbiter (the flash has always higher priority). For this mode the CS will be controlled by hardware as you can’t handle the CS line with a GPIO, you never actually know when the arbiter is going to grant you access to the bus so you must let it handle CS automatically.

Available in: https://arduino-esp8266.readthedocs.io/en/2.4.0/libraries.html#spi

According to this documentation it is possible to use SPI0, apparently the problem is in the line CS that in this case is automatically controlled by hardware, something that maybe the SD library is not adapted to do.

RudyFiero, 2018-02-10:
Take a look at https://github.com/esp8266/Arduino/issues/1429 near the bottom just above the closed notification. It looks like you need to specify hardware chip select.

igrr, 2018-02-10:
I don't think it is easily possible (without external circuitry, at least) to have an SD card working on SPI0. SPI0 can only be used with hardware CS, meaning that CS will go low before a si

closed 2018-02-10 by igrr