esp8266/Arduino #4301: IS THERE A RELIABLE ESP8266 STACK?

tomgregg, 2018-02-05:
```cpp

#include <Arduino.h>
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:		ESP8266
Core Version:      	multiple

### Description
All my ESP8266s hang when used as simple server.  IS THERE A RELIABLE FIRMWARE STACK?

Problem description
I've tried 5 different ESP8266 modules (4 are ESP12e and 1 is a WeMOS D1 mini).  All kinds of power supplies, well regulated and high current capable.   With each, I've successfully uploaded a simple Arduino script to turn ON and OFF the blue LED.  Really simple.  After a while, the modules stop responding to requests from my Windows laptop and/or Android tablet.  They often work for a few days.  Using wireshark, it appears that the ESP8266 is most likely to hang when it gets too many requests.  I attached an ATTiny13 programmed to toggle the ESP8266 reset pin every 10 minutes, but once the ESP8266 hung in a way that I had to power it down and up to make it work again.

Settings in IDE

Module:  			Generic ESP8266 Module & WeMos D1 mini
Flash Size: 		4Mb and 32Mb
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		?ck / nodemcu?

### Sketch

#include <ESP8266WiFi.h>
 
const char* ssid = "...";//type your ssid
const char* password = "...";//type your password
 
int ledPin = 2; // GPIO2 of ESP8266
WiFiServer server(6969);//Service Port
 
void setup() {
  Serial.begin(115200);
  delay(10);
 
  pinMode(ledPin, OUTPUT);
  digitalWrite(ledPin, LOW);
   
  // Connect to WiFi network
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
   
  WiFi.begin(ssid, password);
   
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
   
  // Start the server
  server.begin();
  Serial.println("Server started");
 
  // Print the IP address
  Serial.print("Use this URL to connect: ");
  Serial.print("http://");
  Serial.print(WiFi.localIP());
  Serial.println("/");
}
 
void loop() {
  // Check if a client has connected
  WiFiClient client = server.available();
  if (!client) {
    return;
  }
   
  // Wait until the client sends some data
  Serial.println("new client");
  while(!client.available()){
    delay(1);
  }
   
  // Read the first line of the request
  String request = client.readStringUntil('\r');
  Serial.println(request);
  client.flush();
   
  // Match the request
 
  int value = LOW;
  if (request.indexOf("/LED=ON") != -1) {
    digitalWrite(ledPin, HIGH);
    value = HIGH;
  } 
  if (request.indexOf("/LED=OFF") != -1){
    digitalWrite(ledPin, LOW);
    value = LOW;
  }
 
  //Set ledPin according to the request
  //digitalWrite(ledPin, value);
   
  // Return the response
  client.println("HTTP/1.1 200 OK");
  client.println("Content-Type: text/html");
  client.println(""); //  do not forget this one
  client.println("<!DOCTYPE HTML>");
  client.println("<html>");
   
  client.print("Led pin is now: ");
   
  if(value == HIGH) {
    client.print("On");  
  } else {
    client.print("Off");
  }
  client.println("<br><br>");
  client.println("Click <a href=\"/LED=ON\">here</a> turn the LED on pin 2 ON<br>");
  client.println("Click <a href=\"/LED=OFF\">here turn the LED on pin 2 OFF<br>");
  client.println("</html>");
 
  delay(1);
  Serial.println("Client disconnected");
  Serial.println("");
}

RudyFiero, 2018-02-06:
<h1>**A Bad Workman Always Blames His Tools** </h1>

Try some code that works.  What you have there is crap.
https://gist.github.com/bbx10/667e3d4f5f2c0831d00b

const int LEDPIN = 0;     <-change this to 2

devyte, 2018-02-06:
@tomgregg which core version? did you test with latest git? There have been seveeal fixes merged after 2.4 release that could be relevant.

tonton81, 2018-02-06:
most complaints are from people who dont use git :P

RudyFiero, 2018-02-06:
I ran the code he posted and he is right, it is flaky. There is a problem but I didn't want to do a lot of work trying to find it.  What I wanted was for him to run a known working sketch.

devyte, 2018-02-06:
@RudyFiero it matters. Like I said, several fixes were merged after 2.4 release. If the issue doesn't happen with latest git, this should be closed, hence my previous comment.
@tomgregg please test your code with latest git.

d-a-v, 2018-02-06:
```
  WiFiClient client = server.available();
  if (!client) {
    return;
  }
```
should be
```
  if (!server.hasClient())
    return;
  WiFiClient client = server.available();
```

RudyFiero, 2018-02-06:
@devyte  Sure there have been fixes since 2.4.0 but before blaming the stack it helps to start with a program that works.

tomgregg, 2018-02-07:
Hi.  I really appreciate all the excellent responses I've received!  With the additional header files in the script, I'm running into a new set of problems.

I'm using Arduino 1.8.5.  I added "http://arduino.esp8266.com/stable/package_esp8266com_index.json" and INSTALLED it.  It seems like a 'vanilla' setup.  At this point, I'm just trying the "verify/compile" step.

I'm trying the code Rudy pointed me to, but there was one header file I had to add:  WebSocketsServer.h.  After I got by the "fatal error: WebSocketsServer.h: No such file or directory                 #include "WebSocketsServer.h""
I found the header file in GitHub, included it in the script directory, and tried again.  This time it couldn't find "WebSockets.h".  So I found that file on GitHub,  included it in the script directory, and tried again.
Now, I get passed the "header files not found" problem, but I get many messages like:

"C:\aaa Tom Stuff\Atmel\Arduino\sketch_feb06a/sketch_feb06a.ino:154: undefined reference to `WebSocketsServer::broadcastTXT(unsigned char*, unsigned int, bool)'"

and "C:\Users\TOM1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266mDNS/ESP8266mDNS.h:77: undefined reference to `WebSocketsServer::~WebSocketsServer()'
sketch\sketch_feb06a.ino.cpp.o: In function `__static_initialization_and_destruction_0':"

Anyway, I probably have bad header files.  Could you please point me to the right ones?  Thanks.

Pablo2048, 2018-02-07:
Why you did not install whole websocketserver library? ( https://github.com/Links2004/arduinoWebSockets ). It seems like you even don't know what are you doing...

RudyFiero, 2018-02-07:
Download the zip file for the library then go to  Sketch  -  Include Library  -  Add Zip Library

tomgregg, 2018-02-07:
Thanks to RUDY, it works fine (Pablo's pointer was important too).
I'm new to GitHub and just learning how to access it.  I hope I can contribute to it in the future.  Thanks again!!

closed 2018-02-07 by tomgregg