esp8266/Arduino #4307: wifi phone app controlled rgb strip

Aggeloz, 2018-02-06:
### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	2.1.0-rc2

### Description

Problem description

### Settings in IDE

Module:  			Nodemcu Amic
Flash Size: 		4MB/
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	40Mhz
Upload Using:		 SERIAL
Reset Method:		 nodemcu

### Sketch

```cpp

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

int ledB = D5;
int ledG = D6;
int ledR = D7;

const char* ssid = "Aggelos ssid";
const char* password = "12345678900";

ESP8266WebServer server(80);

void handleRoot() {
  String message = "<html><head></head><body>Hello</body></html>";

  server.send(200, "text/html", message);
}

void handleNotFound(){
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
}

void setup(void){
  Serial.begin(115200);
  delay(10);
    pinMode(D5, OUTPUT);
  pinMode(D6, OUTPUT);
  pinMode(D7, OUTPUT);
  analogWriteRange(255);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  server.on("/", handleRoot);

  server.on("/rgb", [](){
    server.send(200, "text/plain", "rgb");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");

}

void loop() {
  server.handleClient();

}

```

### Debug Messages

```
So hello i want to make a code so i can control a rgb led strip over wifi with the phone app that i made i want some help with the code on how to receive date on my nodemcu (http request) from my app for now my phone app has 3 sliders which have a min value of 0 and a max of 255 every time i use a slider it sends an http request like this http://192.168.1.4/rgb/000000000/     the zeros are the values for RED GREEN and BLUE 
```

devyte, 2018-02-06:
This is not the right place to ask for help with your particular project. This is an issue tracker. Please refer to a community forum for such questions.
Closing per issue POLICY doc.

closed 2018-02-06 by devyte

Aggeloz, 2018-02-06:
ok thanks