esp8266/Arduino #4309: WiFi.waitForConnectResult() takes up to 1 hour to return

Raphyyy, 2018-02-06:
Hardware: ESP12E
Core Version: Latest master

Hi,

I noticed since the 2.4.0 (even the latest) I get random network disconnection. Not that hard to handle. But in few case, I get WiFi.waitForConnectResult() taking up to 1 hour to return the result : 

Code : 
```
Serial.printf(F("Connecting to %s with key %s...\n"), conf.SSID, conf.WKEY);
WiFi.begin(conf.SSID, conf.WKEY);
delay(50);

if (WiFi.waitForConnectResult() == WL_CONNECTED)
{
	Serial.println(F("WiFi connected"));
	return true;
}
```

Serial monitor : 
```
[06/02/18 - 01:34:15:776] Connecting to *** with key ***...

[06/02/18 - 02:20:21:509] WiFi connected
```

It could takes 1 minute or 5 minutes or in that case 45 minutes to return status code.
I have others ESP8266 with same code that are able to connect and send data during those disconnection.
This problem doesn't appear at the beggining. In that case, the problem appears after 10 hours of running with no problems.

Thank you

tablatronix, 2018-02-13:
I have noticed this as well, not sure what changed, but waitForConnectResult() just sits in the loop after begin() and nothing happens, not sure whats going on.

tablatronix, 2018-02-13:
I just noticed if autoreconnect is false waitForConnectResult jsut sits there, because there is nothing happening, wifi is doing nothing just stays disconnected.

minida28, 2018-06-02:
I observed the same behavior (setting `autoreconnect ` to false will cause `waitForConnectResult` just sits forever and nothing happened).

So what is the real use of `WiFi.waitForConnectResult()` ? Can I safely omit it?
I usually put it after `begin()`, I am just following example sketchs I found on the net.

Also there are already callbacks available to monitor wifi status such as `WiFiEventStationModeConnected`, `WiFiEventStationModeGotIP` or `WiFiEventStationModeDisconnected`.

devyte, 2018-07-13:
A timeout is needed in here:
https://github.com/esp8266/Arduino/blob/fcf2ac5d3de6c17fe177e16ccd1a2789826c2aa0/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L377

PR #5371 (2019-07-04): Add timeout to waitForConnectResult
Aims to fix #4309
closed 2019-07-04 by d-a-v