esp8266/Arduino #4310: SPIFFS failing to read data and end of file

brianjmurrell, 2018-02-06:
### Basic Infos

Same as in #4291

### Sketch

```cpp
#include <FS.h>

void setup() {

  Serial.begin(115200);
  SPIFFS.begin();

  FSInfo fs_info;
  SPIFFS.info(fs_info);
  printf("SPIFFS: %lu of %lu bytes used.\n",
         fs_info.usedBytes, fs_info.totalBytes);

}

void loop() {

  char line[80];
  int num_read = 0;
    File f = SPIFFS.open("/stats-22.txt", "r");

  Serial.print("1 pos: ");
  Serial.println(f.position());
  num_read = f.readBytes(line, 80);
  Serial.print("1 read ");
  Serial.print(num_read);
  Serial.print(" bytes: ");
  Serial.println(line);

  f.seek(f.size() - 81);  // assume a line is max of 80 chars long
  Serial.print("2 pos: ");
  Serial.println(f.position());
  num_read = f.readBytes(line, 80);
  Serial.print("2 read ");
  Serial.print(num_read);
  if (num_read) {
    Serial.print(" bytes: ");
    Serial.println(line);
  } else
    Serial.println();
  while(1);
}
```

### Debug Messages

```
SPIFFS: 2205788 of 2949250 bytes used.
1 pos: 0
1 read 80 bytes: 1517888260  44  22  19 999 225 1019 0 1
1517888265  45  22  19 999 225 1023 0 10⸮⸮?
2 pos: 63097
2 read 0
```

The actual contents of that file at those _positions_ is:

```sh
$ dd if=stats-22.txt bs=1 skip=0 count=80
1517888260  44  22  19 999 225 1019 0 1
1517888265  45  22  19 999 225 1023 0 1

$ dd if=stats-22.txt bs=1 skip=63097 count=80
7891993  44  21  13 999 225 1023 0 1
1517891995  44  21  14 999 225 1023 0 1
```

I cannot see why it should fail to read at postition 63097.

brianjmurrell, 2018-02-06:
For this sketch:

```cpp
#include <FS.h>

void setup() {

  Serial.begin(115200);
  SPIFFS.begin();

  FSInfo fs_info;
  SPIFFS.info(fs_info);
  printf("SPIFFS: %lu of %lu bytes used.\n",
         fs_info.usedBytes, fs_info.totalBytes);

}

void loop() {

  char line[80];
  int num_read = 1;
  File f = SPIFFS.open("/stats-22.txt", "r");

  for (int pos=0; num_read > 0; pos++) {
    f.seek(pos);
    Serial.print("pos: ");
    Serial.print(f.position());
    num_read = f.readBytes(line, 80);
    Serial.print(", read ");
    Serial.println(num_read);
  }

  while(1);
}
```

I got:

```
...
pos: 57900, read 80
pos: 57901, read 80
pos: 57902, read 0
```

The file is clearly bigger than 57902 so why is it failing to read at that point?

apicquot, 2018-12-02:
what board configuration are you using ?

brianjmurrell, 2018-12-03:
> what board configuration are you using ?

Board description is in the ticket referenced in the original comment:

Hardware:	NodeMCU 1.0 ESP-12E
Core Version: 2.4.0

> Also, how is the file put on the device?

The device is creating and updating the file itself.  It's a "log".

apicquot, 2018-12-03:
@devyte, for the Arduino IDE, mkspiff takes the block size from the boards.txt file. The code compiles with information from the ld files, that is why spiffs uploaded in not compatible with the binary. I checked with all other boards configuration - and just obviously noticed that the 2M boards with spiffs of 128, 256, did not have compatible data in the 2 files.

@brianjmurrell, I use the espressif board WROOM-02 that comes with a 2M flash. I eventually debug sometime on a Huzzah Adafruit board that is 4M but it works fine with a 2M configuration.

earlephilhower, 2020-01-01:
SPIFFS is no longer maintained by the author.  There are some weird corner cases when the filesystem is under heavy use involving GC/etc.  We have added the `::gc()` method which can be called as desired (because it may have a long runtime).  The suggestion for issues like this is to mode to LittleFS which is presently under active support by the ARMbed folks.

closed 2020-01-01 by earlephilhower