esp8266/Arduino #4313: Help sketch

folny, 2018-02-06:
Hello friends

Can anyone advise me how to edit this sketch to work with ESP32 in advance thanks for your reply.

https://github.com/dc42/arduino/blob/master/MetalDetector/MetalDetector.ino

Problem is with this frequent code

ISR(TIMER1_OVF_vect)
{
++ticks;
uint8_t ctr = TCNT0;
int16_t val = (int16_t)(uint16_t)ADCH; // only need to read most significant 8 bits
if (ctr != ((lastctr + 1) & 7))
{
++misses;
}
lastctr = ctr;
int16_t *p = &bins[ctr & 3];
if (ctr < 4)
{
*p += (val);
if (*p > 15000) *p = 15000;
}
else
{
*p -= val;
if (*p < -15000) p = -15000;
}
if (ctr == 7)
{
++numSamples;
if (numSamples == numSamplesToAverage)
{
numSamples = 0;
if (!sampleReady) // if previous sample has been consumed
{
memcpy((void)averages, bins, sizeof(averages));
sampleReady = true;
}
memset(bins, 0, sizeof(bins));
}
}
}

tonton81, 2018-02-07:
you need to ask on the forums, not here

folny, 2018-02-07:
Ok thanks.