esp8266/Arduino #4315: Software Serial  Inestable msg

pacocuervo, 2018-02-07:
Randomly, I have experiencing communication probles between ESP12N and ATM902E device using softwareSerial  to comunicate.

Finally I have isolated the problem. I'll try to show what si happening...

I have create a loop sending 0x55 to serial. 

In the scope you can see the data trace...

[image]

Sometimes the scope show  somenthing like that.

[image]

The arrows show that this bit has  a 40% width les than normall bit.

Any suggestion about?

Regards.

devyte, 2018-02-07:
@igrr should this be handled in the SoftwareSerial repo?

devyte, 2018-02-08:
@pacocuervo BTW, when you opened this issue, you were presented with an issue template requesting certain info. You pretty much ignored it in its entirety. Please edit your post and add the requested info.

igrr, 2018-02-08:
/cc @plerup

earlephilhower, 2019-01-16:
Closing here as it's being followed in the SWSerial repo (plus it's a really old core).

closed 2019-01-16 by earlephilhower