esp8266/Arduino #4316: during erase flash, send data to Serial, module will wdt reset。

chenkaiyao, 2018-02-07:
### Basic Infos

#### Hardware
Hardware:	        ESP-12
Core Version:      	2.1.0-rc2

### Description

during erase flash, send data to Serial。module will wdt reset。

send data 
```
122222222222221222222222222212222222222222122222222222221222222222222212222222222222122222222222221222222222222212222222222222122222222222221222222222222212222222222222122222222222221222222222222212222222222222
```

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		 SERIAL
Reset Method:		nodemcu

### Sketch

```cpp
extern "C" {
#include "c_types.h"
#include "ets_sys.h"
#include "os_type.h"
#include "osapi.h"
#include "spi_flash.h"
}

extern "C" uint32_t _SPIFFS_end;

void setup()
{
  Serial.begin(115200);
}

uint32_t _sector = ((((uint32_t)&_SPIFFS_end - 0x40200000) / SPI_FLASH_SEC_SIZE));

void loop()
{
  while(Serial.available()>0)
  {
    Serial.print(Serial.read());
  }
  
  Serial.print("erase\r\n");
  noInterrupts();
  spi_flash_erase_sector(_sector);
  interrupts();
}
```

### Debug Messages

```
erase
erase

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v074402e2
~ld
erase
```

WereCatf, 2018-02-08:
Try adding Serial.flush() before noInterrupts() or a delay?

devyte, 2018-02-08:
@chenkaiyao I can't reproduce this on a Wemos D1 mini r2 or on a WittyCloud. I get continous "erase" lines in the Serial monitor non stop. Left it running for a couple of minutes.

Writing to flash draws more current, so it's possible this is a bad power supply.

Closing as can't reproduce.

chenkaiyao, 2018-02-08:
@devyte you should send data to serail

chenkaiyao, 2018-02-08:
@devyte you can send data to module through Serial tool。 eg. 

send data 
```
122222222222221222222222222212222222222222122222222222221222222222222212222222222222122222222222221222222222222212222222222222122222222222221222222222222212222222222222122222222222221222222222222212222222222222
```

devyte, 2018-02-08:
I managed to reproduce on a wemos d1 mini. 
Doing a pure erase loop works.
Doing a pure serial loop works, even with the long string.
Doing both together with a short string works.
Doing both together with the long string doesn't triggers wdt.
Adding yield in the while orn right after it doesn't help.
Adding Serial.flush doesn't help. 
A delay(5) before the erase helps, but I suspect because it slows things down.

chenkaiyao, 2018-02-08:
i find module all enter uart interrupts, end trigger wdt

d-a-v, 2018-02-08:
Do you think this sketch should be converted into pure esp-nonos-sdk C program ?

earlephilhower, 2019-01-24:
I believe this was related to the Serial FIFO overflow crash (since IRQs disabled while in flash op, you wouldn't get the IRQ and the old Serial code would crash).  That's been fixed and a) overflow now is noted and doesn't crash, and b) IRQs aren't disabled by flash ops.

closed 2019-01-24 by earlephilhower