esp8266/Arduino #4319: ESP8266-01 trobleshooting

Andy2682, 2018-02-07:
So I bought an ESP8266-01 on a black PCB and I have some problems whit it  like I can't communicate whit it whit the AT commands . For programming the ESP266 I use Arduino IDE and in the serial monitor when I fire up the board it says ,,Failed to mount file system " and I have some other problems like when I upload my code it have a 10% chance of the board getting very hot and the indicator led will turn off. I have verified my connections a hundred of times and there are good. To upload the code to the ESP8266 I use an FTDI ft232rl breakout board. I tried everything to fix it even a firmware update but when I press the install button it says ,,Programmer not responding" some maybe the FTDI breakout can be the problem or I don't now. If you what to now more informations about my setup I will happily give them to you.

Andy2682, 2018-02-07:
Oh and the connections are like in the image below.
https://iot-playground.com/blog/2-uncategorised/35-esp8266-firmware-update

Andy2682, 2018-02-07:
Oh and my wireing is like in the image below.
[image]

devyte, 2018-02-07:
Not the right place to get help with your programming setup. This is an issue tracker, meant for tracking issues in the core libs, and not a general support forum. I suggest seeking help at esp8266.com or stackoverflow.
Closing per issue POLICY doc.

closed 2018-02-07 by devyte