esp8266/Arduino #4320: Using git core in PORTABLE IDE

s0170071, 2018-02-07:
I found the instructions for the Arduino IDE, but not for the portable version (where you have the portable folder inside the IDE main folder). 
Please add. Thanks.

tonton81, 2018-02-07:
its the same way, just add it to the portable’s hardware folder following same instructions

WereCatf, 2018-02-08:
I always use portable IDE myself and there is literally no change in how you set it up, just proceed as usual.

devyte, 2018-02-08:
Closing per last two comments.

closed 2018-02-08 by devyte

s0170071, 2018-02-08:
Alright. I'll try that. 
Still, a note in the readme would be nice, especially since for Arduino portable installations usually all the esp stuff ends up in the "portable" folder, just not the git core...

tonton81, 2018-02-08:
the “git core” never ends up in the arduino installation, core or not..

tonton81, 2018-02-08:
*portable or not i mean

s0170071, 2018-02-08:
We all sometimes mean sth. different than we write, don't we ;-)