esp8266/Arduino #4324: DHCP Server: ESP serves option 43 (vendor specific option) with no context

mjcross, 2018-02-08:
Examining the DHCPOFFER packet served by the ESP, after the DHCP_OPTION_PERFORM_ROUTER_DISCOVERY option and its data, it adds a final option: 43, 6, 0x01, 4, 0x00, 0x00, 0x00, 0x02.

From RFC2132 this is a Vendor Specific Information option that contains an encapsulated option code '0x01' with data '00 00 00 02'.

AFAIK, vendor specific options only make sense if there's a preceding Option 60 (VCI) to identify the type of vendor device - so what's this option doing?

It's clear where this option is generated in the source code [dhcpserver.c](https://github.com/espressif/esp-idf/blob/master/components/lwip/apps/dhcpserver.c#L387) but there are no hints as to the intent.

d-a-v, 2018-02-08:
When I ported lwip2, I saw all this but did not take time to verify everything.
This could be a trigger for some zero-day-exploit in windows network stack :grin:
So we should remove it, should we ?

mjcross, 2018-02-08:
Crikey - hopefully nothing so dramatic (!) But I'm struggling to understand what it's for. I've raised the quesiton on the Espressif support BB so if I get a response then I'll update here.

The only thing I've found so far that seems vaguely possible is a PXE Boot option... PXE suboption 1 (PXE_MTFTP_IP) specifies the multicast IP address for the boot file - but just that on its own that wouldn't really make much sense. Also an IP address of 0.0.0.2 doesn't figure.

dragondaud, 2018-02-10:
I do not see option 43 in my DHCP requests using latest lwip2.

mjcross, 2018-02-10:
It’s in the responses not the requests (DHCPOFFER)

jasonharper, 2018-03-26:
+1 on removing this apparently useless option.  Its presence triggers a known bug that prevents Windows XP systems from acquiring an IP address, unless they have a hotfix installed (KB953761) that is no longer available from Microsoft.  (That's not a large number of potential users these days, but it's certainly nonzero.)

michaelradhuber, 2019-02-03:
Android seems to have problems with it too....

d-a-v, 2019-08-29:
This dhcp answer option has been removed very recently (lwip2 only).

closed 2019-08-29 by d-a-v