esp8266/Arduino #4329: ESP Exception 3 @ strcopy

s0170071, 2018-02-08:
Using ESPEasy with the latest core the ESP crashes when I try to access the web server. Until then its fine. 
Tried all lwip versions.

Exception decoder shows: 

Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
Decoding 38 results
0x4024460c: strcpy at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strcpy.S line 60
0x4028bd32: lwip_init at ?? line ?
0x4028bc12: lwip_init at ?? line ?
0x4023fa85: String::copy(char const*, unsigned int) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 714
0x4023fad2: String::String(char const*) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 714
0x4028bd32: lwip_init at ?? line ?
0x4023fdcc: String::String(__FlashStringHelper const*) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 714
0x40239484: ESP8266WebServer::_prepareHeader(String&, int, char const*, unsigned int) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/libraries/ESP8266WebServer/src/detail/RequestHandlersImpl.h line 119
0x4023fa85: String::copy(char const*, unsigned int) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 714
0x4023fad2: String::String(char const*) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 714
0x402395a7: ESP8266WebServer::send(int, char const*, String const&) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/libraries/ESP8266WebServer/src/detail/RequestHandlersImpl.h line 119
0x4023fad2: String::String(char const*) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 714
0x4020db8f: sendWebPageChunkedBegin(String&) at /home/john/Arduino/scetchbooks/ESPEasy/TimeZoneESPeasy.ino line 251
0x402251bc: processAndSendWebPageTemplate(String&, String&) at /home/john/Arduino/scetchbooks/ESPEasy/TimeZoneESPeasy.ino line 251
0x4023fda4: String::operator=(__FlashStringHelper const*) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 714
0x402093eb: checkRAM(__FlashStringHelper const*) at /home/john/Arduino/scetchbooks/ESPEasy/TimeZoneESPeasy.ino line 251
0x40283ec3: lwip_init at ?? line ?
0x402253f8: sendWebPage(String const&, String&) at /home/john/Arduino/scetchbooks/ESPEasy/TimeZoneESPeasy.ino line 251
0x401077c8: xPortWantedSizeAlign at ?? line ?
0x4023fb10: String::copy(__FlashStringHelper const*, unsigned int) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 714
0x40229dee: handle_setup() at /home/john/Arduino/scetchbooks/ESPEasy/TimeZoneESPeasy.ino line 251
0x4023fa07: String::changeBuffer(unsigned int) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 714
0x4023fa53: String::reserve(unsigned int) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 714
0x402399e4: FunctionRequestHandler::handle(ESP8266WebServer&, HTTPMethod, String) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/libraries/ESP8266WebServer/src/detail/RequestHandlersImpl.h line 119
0x40241b7e: std::_Function_handler ::_M_invoke(std::_Any_data const&) at /home/john/Arduino/scetchbooks/ESPEasy/TimeZoneESPeasy.ino line 251
0x402399da: std::function ::operator()() const at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/libraries/ESP8266WebServer/src/detail/RequestHandlersImpl.h line 119
0x40239a16: FunctionRequestHandler::handle(ESP8266WebServer&, HTTPMethod, String) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/libraries/ESP8266WebServer/src/detail/RequestHandlersImpl.h line 119
0x4023fbb0: String::String(String const&) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 714
0x40239a9a: ESP8266WebServer::_handleRequest() at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/libraries/ESP8266WebServer/src/detail/RequestHandlersImpl.h line 119
0x402373fc: WiFiClient::~WiFiClient() at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 372
0x4023823c: FunctionRequestHandler::canHandle(HTTPMethod, String) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/libraries/ESP8266WebServer/src/detail/RequestHandlersImpl.h line 119
0x40239ce0: ESP8266WebServer::handleClient() at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/libraries/ESP8266WebServer/src/detail/RequestHandlersImpl.h line 119
0x4023b3b5: DNSServer::processNextRequest() at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/libraries/DNSServer/src/DNSServer.cpp line 51
0x4023e9c5: HardwareSerial::available() at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/HardwareSerial.cpp line 180
0x40222333: backgroundtasks() at /home/john/Arduino/scetchbooks/ESPEasy/TimeZoneESPeasy.ino line 251
0x4022d2a7: loop at /home/john/Arduino/scetchbooks/ESPEasy/TimeZoneESPeasy.ino line 251
0x402409d4: loop_wrapper at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 57
0x40100710: cont_norm at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/cont.S line 109

mudkip908, 2018-02-08:
I was about to report the same thing when I saw this issue so here's all the info I have as a comment.
#### Hardware
Hardware:			NodeMCU 1.0 (ESP-12E Module)
Core Version:      	bcbd5961c0a19882dbf05117f5a287087dc0909b

### Description

After requesting a static file from SPIFFS, the ESP8266 reboots after printing a stack dump.
Requests that are handled by handler functions (like /test in the sketch below) work fine.
With commit 0339bbb serving files from SPIFFS works.

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Upload Using:		OTA
Reset Method:		OTA upload
lwIP Variant:		v2 Higher Bandwidth

### Sketch

```cpp
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <FS.h>
#include <ArduinoOTA.h>

#include "wifi_creds.h"

ESP8266WebServer server(80);

void setup()
{
	Serial.begin(115200);
	WiFi.mode(WIFI_STA);
	WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
	while(WiFi.status() != WL_CONNECTED)
	{
		delay(50);
		digitalWrite(2, !digitalRead(2));
	}
	digitalWrite(2, 1);

	SPIFFS.begin();

	server.on("/", [](void) {
		server.sendHeader("Location", "/index.html");
		server.send(301);
	});
	
	server.on("/test", [](void) {
		server.send(200, "text/plain", "test");
	});

	server.serveStatic("/", SPIFFS, "/");
	server.begin();

	ArduinoOTA.setHostname("esptest");
	ArduinoOTA.onStart([](void) {
		SPIFFS.end();
	});
	ArduinoOTA.begin();
}

void loop()
{
	server.handleClient();
	ArduinoOTA.handle();
}
```

index.html:
```html
<html><body>Hello!</body></html> 
```
### Debug Messages
[Raw serial output with Debug Level: HTTP_SERVER](https://gist.github.com/mudkip908/4200111c55ee69a1c150480e6198e91e)
[Decoded stack trace (please disregard the /2.4.0/ folder name, it's actually bcbd596)](https://gist.github.com/mudkip908/d59f6081a09dc8aecf47b8a7edc86096)

igrr, 2018-02-09:
Looks like the mime type table has been moved to progmem, but its members are accessed without `F(...)`: https://github.com/esp8266/Arduino/commit/bcbd5961c0a19882dbf05117f5a287087dc0909b#diff-89232ba90a9c9da52cc615de4acd59ebR104

earlephilhower, 2018-02-09:
@s0170071 Can you try the PR in #4338?  I've done the most basic of tests and it seems to be working, but you're obviously a heavy user and could test and find any other hidden problems faster.

Thx
-EFP3

s0170071, 2018-02-10:
@earlephilhower  👍  ... yep, that fixes it. Thank you. Good Job!

PR #4338 (2018-02-10): Wrap mimetype strings in FSPTR()s
Mimetype is now in progmem, so any accesses to it need to be using FPSTR() wrapped Strings or you'll get Exceptions.
    
Fixes #4329
commit 291bc6bc (2018-02-10): Wrap mimetype strings in FSPTR()s (#4338)

Mimetype is now in progmem, so any accesses to it need to be using FPSTR()
wrapped Strings.

Fixes #4329
closed 2018-02-10 by earlephilhower