esp8266/Arduino #4330: Weird crash while casting 4 byte data array  to uint32_t

alieslam, 2018-02-08:
### Basic Infos

#### Hardware
Hardware: Wemos D1 mini
Core Version: 2.4.0
### Description

Attached bellow a sketch were i am trying to cast a 4 byte data array to a 32 bit integer, but the system crashes.
PS: Sketch contains .cpp and .h files

### Settings in IDE

Module:  ESP8266 Module
Flash Size: 4MB
CPU Frequency: 80Mhz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:	 SERIAL
Reset Method:		?ck / nodemcu?

### Sketch
Sketch has multiple files.
Main
```cpp
#include  "test.h"

My_Class  test;

void setup() {
  Serial.begin(256000);
  Serial.println("Start");
  printf("%p \n",test.data);
  printf("%p \n",test.getDataPointer());
  uint32_t  temp = *((uint32_t *) test.getDataPointer());   
  Serial.println(temp);
}

void loop() {
}
```
test.cpp file
```cpp
#include  "test.h"

uint8_t * My_Class::getDataPointer()
{
  return data;
}
```
test.h file
```cpp
#ifndef test_h
#define test_h
#include <Arduino.h>
#include <string.h> 
class My_Class{
public:
uint8_t * getDataPointer();
bool isValid;  
uint8_t data[4]={0x00,0x00,0x00,0x20};
};

#endif 
```
### Debug Messages

```
2/8/2018 19:05:20.829 [RX] - ÀüÀ<<<<NUL>À<<<NUL>À<<NUL>ü<ü<<NUL><À<À<NUL>À<NUL><üÀ<NUL>À<À<NUL>ÀÀÀ<À<NUL>À<NUL>À<NUL><<NUL>À<NUL>ü<<<ü<À<ÀüÀÀ<NUL>À<NUL>üÀ<NUL><<NUL>ÀÀÀ<ü<ü<üü<À<ÀÀüÀ<NUL>À<<NUL>À<<NUL>ü<<NUL><Àü<ü<À<NUL>À<<NUL>À<ü<NUL><ü<<<À<<<üü<À<NUL>À<NUL><ÀÀ<NUL>À<ü<NUL>À<ü<NUL><À<<<üü<À<NUL>À<NUL><ÀÀ<NUL>À<ü<NUL>À<Àü<üÀü<NUL><<<<NUL><<NUL><<<<NUL><ÀÀ<<NUL>À<À<ü<À<<Start<CR><LF>
0x3ffee999 <LF>
0x3ffee999 <LF><CR>
<LF>
Exception (9):<CR><LF>
epc1=0x4020208f epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffee999 depc=0x00000000<CR><LF>
<LF>
ctx: cont <LF>
sp: 3ffef840 end: 3ffefa10 offset: 01a0<CR><LF>
<LF>
>>>stack>>><LF>
3ffef9e0:  feefeffe 3ffee998 3ffee9bc 40202088  <CR><LF>
3ffef9f0:  3fffdad0 00000000 3ffee9d8 40202514  <CR><LF>
3ffefa00:  feefeffe feefeffe 3ffee9f0 40100108  <CR><LF>
<<<stack<<<<LF>
```

alieslam, 2018-02-08:
Including them all in single main sketch doesn't crash. Any insights? 
```cpp
class My_Class{
public:
uint8_t * getDataPointer();
bool isValid;  
uint8_t data[4]={0x00,0x00,0x00,0x20};
};

uint8_t * My_Class::getDataPointer()
{
  return data;
}

My_Class  test;

void setup() {
  Serial.begin(256000);
  Serial.println("Start");
  printf("%p \n",test.data);
  printf("%p \n",test.getDataPointer());
  uint32_t  temp = *((uint32_t *) test.getDataPointer());   
  Serial.println(temp);

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

earlephilhower, 2018-02-08:
This is an unaligned access problem.  You can't safely do that cast from char[] to int w/o guarantees about the array starting on a 32-bit boundary, of which there are none for stack char variables.

Please use www.esp8266.com or your search engine of choice for further information.

Closing per POLICY document.

closed 2018-02-08 by earlephilhower

igrr, 2018-02-08:
(A solution for such situations is to use `memcpy(&temp, test.data, sizeof(temp))` — it is aware of alignment requirements. Unless when one of the operands is a peripheral register.)

alieslam, 2018-02-09:
Thanks for the information, apologies for surpassing the forums.