esp8266/Arduino #4331: NodeMCU wdt reset

queenbey91, 2018-02-08:
Hi!
I am working in a website on the Nodemcu
but i have some troubles, frequently It appears this error:
rst cause: 4, boot mode:(1,6)
Anybody knows about that?
I ve read that is a problem with de watchdog but i don't know how to fix it and i don't know
what is the wdt hardware also.

I was thinking that is because of the amount of html lines in my code... but i am not sure, because
takes a long time to load all the website sometimes.

Any help? :) THANKS!

bhumindesai, 2018-02-09:
same here,
using "Adafruit_MPR121" & "Wire" library.

bhumindesai, 2018-02-09:
adding delay(1); solved my problem 
credit goes to @battosai30.

[https://github.com/esp8266/Arduino/issues/3333](url)

queenbey91, 2018-02-09:
Thank you for replaying me but the link it doesn't work !

battosai30, 2018-02-09:
https://github.com/esp8266/Arduino/issues/3333

queenbey91, 2018-02-11:
In case I can help someone, finally i found the explanation to this:

:'D

https://portal.vidadesilicio.com.br/watchdog-esp8266/

devyte, 2018-02-15:
Closing per issue POLICY doc.

closed 2018-02-15 by devyte

lijosmedia, 2020-08-20:
I made ESP8266 relay  control project ..But ESP8266 disconnecting frequently.
Below i given message from serial monitor
Please  guide me to resolve this issue.
Here i am attaching my code 
lijosmedia@gmail.com

#include <FirebaseArduino.h>
#include <ESP8266WiFi.h>

#define FIREBASE_HOST "ddcroom1.firebaseio.com" //Your Firebase Project URL without "http:", "\" and "/"
#define FIREBASE_AUTH "gGZ85jHyRC8pI7A0q2uaWv2xdIGxXx720hW66Q7Z" //Your Firebase Database Secret goes here
#define WIFI_SSID "LIJO"         //WiFi SSID name for which NodeMCU connects
#define WIFI_PASSWORD "ethancyril!"      //SSID Password of your wifi network
#define DHTPIN 5    // Data Pin of DHT 11

#define Relay1 14  //D5
#define Relay2 12  //D6
int rel1, rel2;

void setup()
{
  Serial.begin(115200);// Set the baud rate 115200 in Serial Monitor to display debug messages
  pinMode(Relay1, OUTPUT);
  pinMode(Relay2, OUTPUT);
  
 
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
    delay(100);
  }

void loop()
{
   rel1 = Firebase.getString("FB1").toInt(); //Reading the value of the varialble Status from the firebase
   rel2 = Firebase.getString("FB2").toInt(); //Reading the value of the varialble Status from the firebase

  if (rel1 == 1)                     // If, the Status is 1, turn on the Relay1
  {
    digitalWrite(Relay1, LOW);
    Serial.println("Relay 1 ON");
  }
  if (rel1 == 0)                      // If, the Status is 0, turn Off the Relay1
  {
    digitalWrite(Relay1, HIGH);
    Serial.println("Relay 1 OFF");
  }
  if (rel2 == 1)                       // If, the Status is 1, turn on the Relay2
  {
    digitalWrite(Relay2, LOW);
    Serial.println("Relay 2 ON");
  }
  if (rel2 == 0)                      // If, the Status is 0, turn Off the Relay2
  {
    digitalWrite(Relay2, HIGH);
    Serial.println("Relay 2 OFF");
  }
   
  }

messsages from serial monitor
--------------------------------

ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld

battosai30, 2020-08-20:
Be careful : you put all your credentials in clear ....

battosai30, 2020-08-20:
And in order to solve your problem : you don't connect to Wifi, so it will never connect Firebase ...

lijosmedia, 2020-08-20:
> Be careful : you put all your credentials in clear ....

Yes credentials put properly and its connecting with fibase database.But after 30 to 40 its fiving the error shown below and again reconnecting in 2 seconds.But between this 2 second period its dropping the output.
-------------------------------------------
ets Jan 8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3456, room 16
tail 0
chksum 0x84
csum 0x84
va5432625
~ld
----------------------------------------

lijosmedia, 2020-08-20:
> And in order to solve your problem : you don't connect to Wifi, so it will never connect Firebase ...

without wifi how esp8266 will work

devyte, 2020-08-20:
@lijosmedia This is a closed issue, and it is in an issue tracker meant for tracking issues in our Arduino core. This is not the right place to ask for general help, especially if that help relates to 3rd party libs like firebase. Please ask such questions at a community forum like esp8266.com or stackoverflow, not here.