esp8266/Arduino #4332: Response lag if ESP was idle

s0170071, 2018-02-08:
I am using the sketch from https://github.com/esp8266/Arduino/issues/3871 with a Wemos D1 mini and core 2.4.0 The webpage from the example usually loads swiftly within 74 ms. Repeatedly.
If I let the ESP alone for about 2 minutes, it does not respond at first. Webpage load then takes 7-15s. After that its back to 74ms.

d-a-v, 2018-02-08:
If you are running git master version of the core, a dump of the traffic is welcome (wireshark, tcpdump).

s0170071, 2018-02-09:
It has got something to do with my router and the repeater. If I switch off the repeater, the issue is gone. 
Still, as this only affects the ESP I'll keep digging.

s0170071, 2018-02-09:
Wireshark capture.
Note: the last (red)  packet always appears, even if the page loads right away. Is this normal ?
Here is the full capture, one for normal operation (with my repeater disabled) the other one with the repeater enabled and the ESP idle time.
[Wireshark.zip](https://github.com/esp8266/Arduino/files/1711785/Wireshark.zip)

[image]

s0170071, 2018-02-09:
About that last packet. I just noticed it has the numbers for ACK and Seq swapped. Looks like a bug to me...

s0170071, 2018-03-17:
Does no one know about Seq and ACK numbers ?

d-a-v, 2018-03-18:
@s0170071 There seems to be nothing wrong with seq and ack.
What is seq for one peer is at some point acked by the other.
On the last two lines, IP are swapped, and so are seq/ack.

Regarding your initial issue, can you check the bottom of [#2330](https://github.com/esp8266/Arduino/issues/2330#issuecomment-373359854) and retry with disabling the power management mode ?

ximon, 2020-05-11:
I'm getting a very similar intermittent issue, some requests come back in <500ms, the odd one will take >5000ms.
I'm using PlatformIO and the arduino framework, I have this issue using the arduino IDE.
I have a MVP project which allows to to replicate this.
I've tried disabling the power management mode, makes no difference.
I've got wireshark logs and debug output with the following specified: DEBUG_ESP_HTTP_SERVER, DEBUG_ESP_CORE, DEBUG_ESP_WIFI

Should i attach them here or create another issue? I added the info here as it sounds very similar to my issue.

Bascht74, 2020-12-30:
@ximon I am using Tasmota and have the same problems as stated from you and @s0170071 . 
I am using UNIFI AP and Tasmota with Core-Version 2.7.4  / SDK-Version 2.2.2-dev(38a443e)

can you please add the MVP or did you created another issue?

Sebastian

s0170071, 2020-12-31:
i am not entirely sure after two years but if you run two routers, in my cases they were FritzBoxes, check if you completed the  mesh setup by pressing the wps button on each box.