esp8266/Arduino #4337: NodeMCU 1.0 (ESP-12E Module) OTA problem

SergioMorello, 2018-02-09:
### Basic Infos

#### Hardware
Hardware:              NodeMCU 1.0 (ESP-12E Module)
Core Version:         ??

### Description

Hi!
Can't establish OTA update using BasicOTA example code.  Upload sketch using USB connection, pushing RST button on module, unplug module from laptop, connect it to other power source.
After power-on ESP get IP and Arduino IDE seen it on IP ports, but when i try to upload this sketch again, but now in OTA-mode, i get Error[4]. All FireWalls are disabled, ping on IP in cmd get no issues. 
Thanks in advance for any help with this one!

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4MB/1MB
CPU Frequency:       80Mhz
Flash Mode:		
Flash Frequency:	
Upload Using:		OTA 
Reset Method:		

### Sketch

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

const char* ssid = "XXXX";
const char* password = "XXXX";

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  // Port defaults to 8266
  // ArduinoOTA.setPort(8266);

  // Hostname defaults to esp8266-[ChipID]
  // ArduinoOTA.setHostname("myesp8266");

  // No authentication by default
  // ArduinoOTA.setPassword("admin");

  // Password can be set with it's md5 value as well
  // MD5(admin) = 21232f297a57a5a743894a0e4a801fc3
  // ArduinoOTA.setPasswordHash("21232f297a57a5a743894a0e4a801fc3");

  ArduinoOTA.onStart([]() {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH)
      type = "sketch";
    else // U_SPIFFS
      type = "filesystem";

    // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
    Serial.println("Start updating " + type);
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });
  ArduinoOTA.begin();
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  ArduinoOTA.handle();
}

### Debug Messages

Ready
IP address: 192.168.0.101
Start updating sketch
Progress: 0%
Error[4]: End Failed

dragondaud, 2018-02-16:
Your "onStart" and "onEnd" are missing the \[\]\(\) part all the others have, like:

```
  ArduinoOTA.onStart([]() {
  ArduinoOTA.onEnd([]() {

```
That might cause a problem (they're present in current example). Also, you did not specify which core version you are using; 2.3.0 is very old, and 2.4.0 is fairly new, and has a lot of bug fixes available in the git version: https://github.com/esp8266/Arduino#using-git-version

mrWheel, 2019-03-26:
I also posted this on #959 Sorry if I doing this wrong..

I'm on core 2.5.0 with ArduinoIDE 1.8.8 (macOS)
I have uploaded numerous firmware to a lot of ESP boards. All without problems .. until today :-(
I can upload the basicOTA sketch wired and then upload the basicOTA or my own sketch using OTA. But the firmware I'm working on will not upload OTA (configuring OTA is exactly the same as the basicOTA) a second time after first uploading the basicOTA. Keep getting the popup to enter a Board Password which I have not set (and even if I set one, it will not work). Occasionally I get a same popup if I try to open the Serial Monitor..

I can though upload OTA using the command line espota.py program. That works all the time.

What am I missing???

mrWheel, 2019-04-17:
Hi,
I'm now on Arduino IDE 1.8.9 and still have the same problem. 
Can it be that it's due to a combination of libraries?
I have a sketch that has exactly the same ArduinoOTA setup that does upload OTA by the Arduino IDE. But the sketch with the problem does not even show the:

`python /User/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/espota.py -i 192.168.12.162 -p 8266 --auth= -f /Arduino/build/Sketch_Name.ino.bin `

line, but it does show a pop-up window asking for the password (which I have NOT set!)!!!

After uploading a Sketch that does not have this problem I can **_one time_** and only **_one_** time OTA upload the sketch that does have this problem with the Arduino IDE..
```
Build options changed, rebuilding all
Sketch uses 538180 bytes (51%) of program storage space. Maximum is 1044464 bytes.
Global variables use 57308 bytes (69%) of dynamic memory, leaving 24612 bytes for local variables. Maximum is 81920 bytes.
python /User/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/espota.py -i 192.168.12.162 -p 8266 --auth= -f /Arduino/build/TheProblemSketch.ino.bin 
Uploading.................................................. (lots of dots) .............
```
After this successful OTA upload I can not OTA upload with the Arduino IDE again.

I can however OTA upload the sketch by entering the espota.py command in a terminal window. That works like a charm all the time.

This is very annoying!

mrWheel, 2019-04-21:
I asked some other people to upload my Sketch to a ESP8266 wired and after that to do it again OTA. All with the same result: a popup asking for a password.

You can find this sketch here: “https://github.com/mrWheel/DSMRloggerWS“

I’m now beginning to suspect it has to do with the size of the binary. Could it be that in the step before calling espota.py something goes wrong? 
I don’t fully understand the chain of commands involved with compiling and linking the sketch to a binary file but I see something like ‘signing’. Whats that for?

The fact that OTA does work if I just enter the espota.py command in a terminal window puzzles me.

Can anyone (@igrr) elaborate? Please?

mrWheel, 2019-04-23:
This is the verbose output of the ArduinoIDE
```
Linking everything together...
/Users/WillemA/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-gcc -CC -E -P -DVTABLES_IN_FLASH /Users/WillemA/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/ld/eagle.app.v6.common.ld.h -o /Users/WillemA/tmp/Arduino/build/local.eagle.app.v6.common.ld
/Users/WillemA/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-gcc -fno-exceptions -Wl,-Map -Wl,/Users/WillemA/tmp/Arduino/build/DSMRloggerWS_v42.ino.map -g -w -Os -nostdlib -Wl,--no-check-sections -u app_entry -u _printf_float -u _scanf_float -Wl,-static -L/Users/WillemA/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/lib -L/Users/WillemA/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/ld -L/Users/WillemA/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/libc/xtensa-lx106-elf/lib -Teagle.flash.4m1m.ld -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read -o /Users/WillemA/tmp/Arduino/build/DSMRloggerWS_v42.ino.elf -Wl,--start-group /Users/WillemA/tmp/Arduino/build/sketch/DSMRloggerWS_v42.ino.cpp.o /Users/WillemA/tmp/Arduino/build/libraries/Time-master/DateStrings.cpp.o /Users/WillemA/tmp/Arduino/build/libraries/Time-master/Time.cpp.o /Users/WillemA/tmp/Arduino/build/libraries/arduino-dsmr-master/dsmr/fields.cpp.o /Users/WillemA/tmp/Arduino/build/libraries/ArduinoOTA/ArduinoOTA.cpp.o /Users/WillemA/tmp/Arduino/build/libraries/ESP8266WiFi/ESP8266WiFi.a /Users/WillemA/tmp/Arduino/build/libraries/ESP8266WebServer/ESP8266WebServer.a /Users/WillemA/tmp/Arduino/build/libraries/ESP8266mDNS/ESP8266mDNS.a /Users/WillemA/tmp/Arduino/build/libraries/WiFiManager/WiFiManager.cpp.o /Users/WillemA/tmp/Arduino/build/libraries/DNSServer/DNSServer.a /Users/WillemA/tmp/Arduino/build/libraries/TelnetStream-master/TelnetStream.cpp.o /Users/WillemA/tmp/Arduino/build/libraries/arduinoWebSockets-master/WebSockets.cpp.o /Users/WillemA/tmp/Arduino/build/libraries/arduinoWebSockets-master/WebSocketsClient.cpp.o /Users/WillemA/tmp/Arduino/build/libraries/arduinoWebSockets-master/WebSocketsServer.cpp.o /Users/WillemA/tmp/Arduino/build/libraries/arduinoWebSockets-master/libb64/cdecode.c.o /Users/WillemA/tmp/Arduino/build/libraries/arduinoWebSockets-master/libb64/cencode.c.o /Users/WillemA/tmp/Arduino/build/libraries/arduinoWebSockets-master/libsha1/libsha1.c.o /Users/WillemA/tmp/Arduino/build/libraries/SSD1306Ascii-master/SSD1306Ascii.cpp.o /Users/WillemA/tmp/Arduino/build/libraries/Wire/Wire.cpp.o /Users/WillemA/tmp/Arduino/build/libraries/Hash/Hash.a /Users/WillemA/tmp/Arduino/build/core/core.a -lhal -lphy -lpp -lnet80211 -llwip2-536-feat -lwpa -lcrypto -lmain -lwps -lbearssl -laxtls -lespnow -lsmartconfig -lairkiss -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group -L/Users/WillemA/tmp/Arduino/build
/Users/WillemA/Library/Arduino15/packages/esp8266/tools/esptool/2.5.0-3-20ed2b9/esptool -eo /Users/WillemA/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/bootloaders/eboot/eboot.elf -bo /Users/WillemA/tmp/Arduino/build/DSMRloggerWS_v42.ino.bin -bm dout -bf 40 -bz 4M -bs .text -bp 4096 -ec -eo /Users/WillemA/tmp/Arduino/build/DSMRloggerWS_v42.ino.elf -bs .irom0.text -bs .text -bs .data -bs .rodata -bc -ec
python /Users/WillemA/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/signing.py --mode sign --privatekey "/Users/WillemA/Library/Mobile Documents/com~apple~CloudDocs/ArduinoProjects/SlimmeMeter/DSMRloggerWS_v42/private.key" --bin /Users/WillemA/tmp/Arduino/build/DSMRloggerWS_v42.ino.bin --out /Users/WillemA/tmp/Arduino/build/DSMRloggerWS_v42.ino.bin.signed
Using library Time-master at version 1.5 in folder: /Users/WillemA/Library/Mobile Documents/com~apple~CloudDocs/ArduinoProjects/libraries/Time-master 
Using library arduino-dsmr-master at version 0.1 in folder: /Users/WillemA/Library/Mobile Documents/com~apple~CloudDocs/ArduinoProjects/libraries/arduino-dsmr-master 
Using library ArduinoOTA at version 1.0 in folder: /Users/WillemA/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ArduinoOTA 
Using library ESP8266WiFi at version 1.0 in folder: /Users/WillemA/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi 
Using library ESP8266WebServer at version 1.0 in folder: /Users/WillemA/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WebServer 
Using library ESP8266mDNS at version 1.2 in folder: /Users/WillemA/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266mDNS 
Using library WiFiManager at version 0.14 in folder: /Users/WillemA/Library/Mobile Documents/com~apple~CloudDocs/ArduinoProjects/libraries/WiFiManager 
Using library DNSServer at version 1.1.1 in folder: /Users/WillemA/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/DNSServer 
Using library TelnetStream-master at version 0.0.1 in folder: /Users/WillemA/Library/Mobile Documents/com~apple~CloudDocs/ArduinoProjects/libraries/TelnetStream-master 
Using library arduinoWebSockets-master at version 2.1.2 in folder: /Users/WillemA/Library/Mobile Documents/com~apple~CloudDocs/ArduinoProjects/libraries/arduinoWebSockets-master 
Using library SSD1306Ascii-master at version 1.2.4 in folder: /Users/WillemA/Library/Mobile Documents/com~apple~CloudDocs/ArduinoProjects/libraries/SSD1306Ascii-master 
[... cut ...]
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2070)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at cc.arduino.packages.ssh.SSHConfigFileSetup.setup(SSHConfigFileSetup.java:49)
	at cc.arduino.packages.uploaders.SSHUploader.uploadUsingPreferences(SSHUploader.java:112)
	... 5 more

```