esp8266/Arduino #4339: error: espcomm_upload_mem failed

Zayyir, 2018-02-09:
### Basic Infos

#### Hardware
Hardware:			ESP-12F Devkit V3 from [Banggood.com](https://www.banggood.com/Geekcreit-Doit-NodeMcu-Lua-ESP8266-ESP-12E-WIFI-Development-Board-p-985891.html?rmmds=myorder&cur_warehouse=CN)
Core Version:      	Not sure

### Description

So I just got my ESP-12F today and I can't seem to get past the flashing of the basic blink script.
I installed the libraries needed but I just can't seem to make it work.
Getting this over and over again:
Arduino: 1.8.5 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (1M SPIFFS), v2 Prebuilt (MSS=536), Disabled, None, 115200"

Build options changed, rebuilding all
Archiving built core (caching) in: C:\Users\...\AppData\Local\Temp\arduino_cache_884149\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,FlashSize_4M1M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,UploadSpeed_115200_3e6f9da155b7b9a751288a5db62730df.a
Sketch uses 247063 bytes (23%) of program storage space. Maximum is 1044464 bytes.
Global variables use 32868 bytes (40%) of dynamic memory, leaving 49052 bytes for local variables. Maximum is 81920 bytes.
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Frequency:	40Mhz
Upload Using:		COM4

### Sketch

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, LOW);   // Turn the LED on (Note that LOW is the voltage level
                                    // but actually the LED is on; this is because 
                                    // it is active low on the ESP-01)
  delay(1000);                      // Wait for a second
  digitalWrite(LED_BUILTIN, HIGH);  // Turn the LED off by making the voltage HIGH
  delay(2000);                      // Wait for two seconds (to demonstrate the active low LED)

### Debug Messages

none

Pablo2048, 2018-02-09:
Reset method?

tonton81, 2018-02-09:
press GPIO0 button, then RESET/FLASH button,  then let go RESET/FLASH button then let go of GPIO0 button.
Upload your sketch from the IDE.
Close this “issue” ?

Zayyir, 2018-02-10:
Sorry if it seems like not an issue, but I'm new (sorry)
May I ask where is the GPIO0 button? Only buttons here are Reset/Flash

Pablo2048, 2018-02-10:
Try to search with google - GPIO0 == FLASH ( for example https://bennthomsen.files.wordpress.com/2015/12/nodemcu_pinout_700-2.png?w=584 )

Zayyir, 2018-02-10:
I have connected GPIO0 to GPIO16, built-in LED now lights up when I press Flash

Pablo2048, 2018-02-10:
Why did you do this?

Zayyir, 2018-02-10:
I assumed that was what's needed, forgive my lack of knowledge

Pablo2048, 2018-02-10:
GPIO16 is used for deep sleep wakeup and has to be connected with reset.
Ok, so once again: Press Flash (and hold it), Press Reset and release it, wait 1 second and release Flash button. Then try to upload sketch again.
This all is complete off-topic here and this kind of questions has to be post in forum...

Zayyir, 2018-02-10:
Alright, I deeply appreciate the help, thanks, even though I know it may be annoying

closed 2018-02-10 by Zayyir

Zayyir, 2018-02-10:
(Update on this, solved by switching to PlatformIO

SithiVaheeth, 2018-09-21:
Build options changed, rebuilding all
Archiving built core (caching) in: C:\Users\Hemanth\AppData\Local\Temp\arduino_cache_797693\core\core_esp8266_esp8266_generic_CpuFrequency_80,VTable_flash,ResetMethod_ck,CrystalFreq_26,FlashFreq_40,FlashMode_qio,FlashSize_512K0,led_2,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_115200_2d7735b7171c68d5a228e47eff30259f.a
Sketch uses 354016 bytes (70%) of program storage space. Maximum is 499696 bytes.
Global variables use 31904 bytes (38%) of dynamic memory, leaving 50016 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

How to rectify this error??Try to help

lrmoreno007, 2019-01-21:
It's solved in the documentation:

https://arduino-esp8266.readthedocs.io/en/latest/faq/a01-espcomm_sync-failed.html

Slayer4477, 2020-02-18:
change tx pin to gpio ,,, i solve this problem with ota upload

void setup() {
  Serial.begin(9600);
  while (!Serial);
WiFi.hostname("xxxxx");      // DHCP Hostname (useful for finding device for static lease)
  ArduinoOTA.setHostname("xxxx");
  ArduinoOTA.setPort(8266);
  ArduinoOTA.setPassword("");
  ArduinoOTA.onStart([]() {

    /////////////////change pin to gpio////////
    //GPIO 1 (TX) swap the pin to a GPIO.
    pinMode(1, FUNCTION_3);
    //GPIO 3 (RX) swap the pin to a GPIO.
    pinMode(3, FUNCTION_3);
    ///////////////////////
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH) {
      type = "sketch";
    } else { // U_SPIFFS
      type = "filesystem";
    }

    // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
    Serial.println("Start updating " + type);
  });
  ArduinoOTA.onEnd([]() {
    ////////////////change pin to rx tx again//////////
    //GPIO 1 (TX) swap the pin to a TX.
    pinMode(1, FUNCTION_0);
    //GPIO 3 (RX) swap the pin to a RX.
    pinMode(3, FUNCTION_0);
    ///////////////////////
    Serial.println("\nEnd");
  });
  ArduinoOTA.begin();
  delay(1000);
}