esp8266/Arduino #4340: Wemos D1 cannot upload

henryquu, 2018-02-09:
Arduino: 1.8.5 (Windows 7), Board: "WeMos D1 R2 & mini, 80 MHz, 4M (3M SPIFFS), v2 Prebuilt (MSS=536), Serial, None, 9600"

Archiving built core (caching) in: C:\Users\SEBAST~1\AppData\Local\Temp\arduino_cache_531703\core\core_esp8266_esp8266_d1_mini_CpuFrequency_80,FlashSize_4M3M,LwIPVariant_v2mss536,Debug_Serial,DebugLevel_None____,UploadSpeed_9600_9521dc13fab40e2c5e19047dea14bb74.a
Sketch uses 256827 bytes (24%) of program storage space. Maximum is 1044464 bytes.
Global variables use 33876 bytes (41%) of dynamic memory, leaving 48044 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Pablo2048, 2018-02-09:
It seems like we have some sort of "users" with same problems here. Are you a troll, or just can't do some search?

tonton81, 2018-02-10:
they simply cant read the last “issue”.

 searching is irrelevant at that point.

devyte, 2018-02-15:
This is documented. You either have a bad board, bad settings in IDE or OS, or bad connection.
Closing.

closed 2018-02-15 by devyte