esp8266/Arduino #4341: wtd reset after WiFi.forceSleepBegin();

p99will, 2018-02-10:
### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	2.4.0

### Description

Getting watchdog reset when trying to turn off WiFi (to save power).

```
ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

I think there is a loop that is in the WiFi library that watchdog doesn't like.

I've tried replacing the delay with a loop calling yheld(); with a smaller delay, I still get error.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			idk?
Flash Frequency:	also dont know?
Upload Using:		USB Serial.
Reset Method:		idk, MCU?

### Sketch

```cpp

void loop() { 
    beginWIFI(); // reconnects WiFi if disconnect.
    beginMQTT(); // reconnects MQTT if it was disconnected.
    printValues();
    Serial.println("Turning off wifi");

    delay(100);
    client.disconnect();   //MQTT disconnect
    WiFi.disconnect();

    WiFi.forceSleepBegin(); // Error only occurs when using this.
    delay(1);

    delay(20 * 1000) //Delay 10 seconds -- Problem occurs while waiting. 

    WiFi.forceSleepWake();

}
```

### Debug Messages

```
 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```

p99will, 2018-02-10:
Sorry, correction: (I'll edit it 2morrow) but,
The problem doesn't occur at or during the delay. It occurs on anything after the delay. Including any Serial.write, additional delays and the force sleep wake.

alexbelck, 2018-02-22:
I had same issue and figured out that I was usinig the wrong library.

For the ESP12e you need to use the "ESP8266WiFi" library, not the standar1 "WiFi"

Check if your code has the right #include <ESP8266WiFi.h>

earlephilhower, 2019-07-25:
Closing, this is being tracked currently in #6172.

closed 2019-07-25 by earlephilhower