esp8266/Arduino #4342: Client.read() Client.available problems with v2.4.0

madesroc, 2018-02-10:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

With the version 2.3.0, my code works fine but with the version 2.4.0 the command client.read() while client.available() returns absolutely nothing.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		?4MB/1MB?
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck 

#include <ESP8266WiFi.h>

const char* ssid = "XXXXX";    
const char* password = "**********";
const char* host = "192.168.1.106"; // ISY IP
int ReadValue = 0; //Valeur de l'output
int v;
unsigned long storedTime=0;
String Response(5000);
String NumString(100);
            

//============================================================== setup
//==============================================================
//==============================================================
void setup() {

  Serial.begin(9600);                          // Enable this line only if DEBUG=true
  delay(10);
  
  //----  1. Settings as Station - Connect to a WiFi network
  Serial.println("Connecting to "+String(ssid));
  
  WiFi.begin(ssid, password);
   while (WiFi.status() != WL_CONNECTED) {
     delay(500);
     Serial.print(".");
    }
   Serial.println("");
   Serial.println("WiFi connected");
   Serial.println(WiFi.localIP());
}

//---------------------------------------------------------------
//---------------------------------------------------------------
//Sous-routine
//---------------------------------------------------------------
//---------------------------------------------------------------

void ReadISY(int VarNb) {                //Sous routine qui lit la variable dans le ISY 

       int NbSearch = 1;
       int NbFois = 1;
       int Temp = 1;
       NumString = "";
       Response = "";
       WiFiClient client;
       const int httpPort = 80;
        if (!client.connect(host, httpPort)) {
            Serial.println("connection failed");
        return;
        } 
        client.print("GET /rest/vars/get/2"); // Type de variable 1=integer, 2=state
        client.println (" HTTP/1.1");       // Lecture des variables au ISY994.
        client.println("Host: 192.168.001.106:80"); // Adresse IP du ISP994 avec son Port
        client.println("User-Agent: Mozilla/4.0");
        client.println("Connection: Close");
        client.println("Content-Type: application/x-www-form-urlencoded");
        client.println("Authorization: Basic xxxxxxxxxxxxxxxxxxxxx="); // Conversion Nom et mot de passse du ISY994 encodé Base64
                                                                         // Nom:Mot de passe important : entre les deux et coller
                                                                         // https://www.base64encode.net/ 

        client.println();
        while (client.available()) {        
               char c = client.read();
              Response = Response + c;
              NbFois = NbFois + 1;  
        }
        while (NbSearch <= VarNb) {       //Recherche la valeur "init><val>" jusqu'à le numéro de variable
              Temp = Response.indexOf("init><val>",Temp + 1);
              NbSearch = (NbSearch + 1);
        }
       // Serial.println(Response);
        
        int NbChar = 10;                //Nb de caractère après la recherche de "init><val>" pour trouver la valeur numérique
        NbSearch = 0;
        while (NbSearch == 0) {         //On vient validé s'il y a plusieurs chiffre dans la valeur du ISY
            char d = Response.charAt(Temp+NbChar);
            if ((isDigit(d))||(d == '-')){    //Verifie si le charactère est un numéro ou un négatif.
              NumString = NumString + d;
              NbChar = NbChar + 1;
            }
            else { 
               NbSearch = 1;
            }
        }
        ReadValue = NumString.toInt();
        Serial.print("Lecture de la variable ");
        Serial.print(VarNb);
        Serial.print(" = ");
        Serial.println(ReadValue);
        client.stop();
    
}
//---------------------------------------------------------------
//---------------------------------------------------------------
//Programme principale
//---------------------------------------------------------------
//---------------------------------------------------------------

void loop() {
         unsigned long t= millis();       // read the time
         if (t>storedTime+10000){
            ReadISY(4);             //Lecture de la variable 4 du ISY dans ReadValue
            storedTime = t;
        }
}

d-a-v, 2018-02-13:
Perhaps you may give the client enough time to answer.
Add a timeout in your ```while (client.available())``` or use ```client.readStringUntil('>')```.

madesroc, 2018-02-13:
Ok but why is the code working with the version 2.3.0 and not with the 2.4.0?

d-a-v, 2018-02-13:
short answer, and if the proposed fix is working: because you were lucky.

madesroc, 2018-02-13:
I added a 400 milliseconds delay before the while (client.available()) and it worked.
Thank you

closed 2018-02-13 by madesroc