esp8266/Arduino #4345: Issues #353/#3100 Continue. WiFi only works once after erasing flash

mlbranchus, 2018-02-10:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware 
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description WiFi only works once after erasing flash

### Settings in IDE

Module:  		Huzzah ESP8266 Module
Flash Size: 		?4MB/1MB?
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		/ SERIAL?
Reset Method:		?ck 

### Sketch

```cpp

#include <Arduino.h>

// Adafruit IO Temperature & Humidity Example
// Tutorial Link: https://learn.adafruit.com/adafruit-io-basics-temperature-and-humidity
//
// Adafruit invests time and resources providing this open source code.
// Please support Adafruit and open source hardware by purchasing
// products from Adafruit!
//
// Written by Todd Treece for Adafruit Industries
// Copyright (c) 2016-2017 Adafruit Industries
// Licensed under the MIT license.
//
// All text above must be included in any redistribution.

/************************** Configuration ***********************************/

// edit the config.h tab and enter your Adafruit IO credentials
// and any additional configuration needed for WiFi, cellular,
// or ethernet clients.
#include "config.h"

/************************ Example Starts Here *******************************/
#include <Adafruit_Sensor.h>
#include <DHT.h>
#include <DHT_U.h>
#include <Adafruit_SSD1306.h>

// oled display
Adafruit_SSD1306 oled = Adafruit_SSD1306();

// pin connected to DH22 data line
#define DATA_PIN 2

// create DHT22 instance
DHT_Unified dht(DATA_PIN, DHT22);

// set up the 'temperature' and 'humidity' feeds
AdafruitIO_Feed *temperature = io.feed("temperature");
AdafruitIO_Feed *humidity = io.feed("humidity");

void setup() {
  oled.begin(SSD1306_SWITCHCAPVCC, 0x3C);  // initialize with the I2C addr 0x3C (for the 128x32)
  oled.display();

  // start the serial connection
  Serial.begin(115200);

  // wait for serial monitor to open
  while(! Serial);

  // initialize dht22
  dht.begin();

  // connect to io.adafruit.com
  Serial.print("Connecting to Adafruit IO");
  io.connect();

  // wait for a connection
  while(io.status() < AIO_CONNECTED) {
    Serial.print(".");
    delay(500);
  }

  // we are connected
  Serial.println();
  Serial.println(io.statusText());

  // text display tests
  oled.setTextSize(1);
  oled.setTextColor(WHITE);
}

void loop() {

  // io.run(); is required for all sketches.
  // it should always be present at the top of your loop
  // function. it keeps the client connected to
  // io.adafruit.com, and processes any incoming data.
  io.run();

  sensors_event_t event;
  dht.temperature().getEvent(&event);

  float celsius = event.temperature;
  float fahrenheit = (celsius * 1.8) + 32;

  Serial.print("celsius: ");
  Serial.print(celsius);
  Serial.println("C");

  Serial.print("fahrenheit: ");
  Serial.print(fahrenheit);
  Serial.println("F");

  // save fahrenheit (or celsius) to Adafruit IO
  temperature->save(fahrenheit);

  dht.humidity().getEvent(&event);
   
  Serial.print("humidity: ");
  Serial.print(event.relative_humidity);
  Serial.println("%");

  // save humidity to Adafruit IO
  humidity->save(event.relative_humidity);

  // print it to the OLED
  oled.clearDisplay();
  oled.setCursor(0,0);
  oled.print("SSID: "); oled.println(WIFI_SSID);
  oled.print("IP: "); oled.println(WiFi.localIP());
  oled.print("Temp: "); oled.print(fahrenheit,0); oled.print(" *F ");
  oled.print("Hum: "); oled.print(event.relative_humidity,0); oled.println(" %");
  oled.print("IO Status: ");
  aio_status_t aio_status = io.status();
  Serial.print("Status: "); Serial.println(aio_status);
  switch (aio_status) {
     case AIO_IDLE:  oled.println("IDLE"); break;
     case AIO_DISCONNECTED:
     case AIO_NET_DISCONNECTED:  oled.println("DISCONNECT"); break;
     case AIO_NET_CONNECTED:
     case AIO_CONNECTED_INSECURE:
     case AIO_CONNECTED: oled.println("CONNECTED"); break;
  }
  oled.display();

  // wait 5 seconds (5000 milliseconds == 5 seconds)
  delay(2000);
}
void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

tonton81, 2018-02-11:
i dont see you running WiFi.begin()?

mlbranchus, 2018-02-11:
TonTon,
Thanks for the swift response.
Unfortunately I'm fairly new to programming, basically I follow examples to learn.

Ok. I inserted  wifi.begin() in void.loop() but got a scope error. I tried brackets without success.
So, I'm going to need help to do anything to fix this mess.

 

 Thanks

tonton81, 2018-02-11:
it has to be in your setup()
WiFi.begin(“SSID”,”PASSWORD”);

tonton81, 2018-02-11:
note the upper and lower case “WiFi”, must be exact

mlbranchus, 2018-02-11:
TonTon,

I installed the fix, restarted IDE and still got status code 21. Shortly thereafter I got a halt. Is there more to this?

Here is the fault info.

Thanks,

Mike

mlbranchus, 2018-02-11:
TonTon,

BTW...I entered my router SSID and PASSWORD.

 

 Mike

beicnet, 2018-02-11:
You also need to #include wifi libraries too!

mlbranchus, 2018-02-11:
Which wifi libraries need to be included?

Mike

beicnet, 2018-02-11:
As I wrote before above at the top of your code:
`#include "ESP8266WiFi.h"`

mlbranchus, 2018-02-11:
TonTon and beic,

I've added #include ESP8266WiFi.h 
but I'm still getting fault code 21.

Mike

beicnet, 2018-02-11:
Did you include ESP8266 Arduino core to your references?

mlbranchus, 2018-02-11:
beic,

I don't know. I don't think so. What is the specific name and where is it located?

Thanks,

Mike

beicnet, 2018-02-11:
Open the Arduino IDE and go to:

File > Preferences > Additional Boards Manager URLs and then add this line:

`http://arduino.esp8266.com/stable/package_esp8266com_index.json`

close the IDE, then reopen it.

After that go to the Tools > Board Manager, find ESP8266 Community and install it.

mlbranchus, 2018-02-12:
Beic,

I already had it installed in prefs and updated as ver 2.3.0.

Next?

Mike

http://arduino.esp8266.com/stable/package_esp8266com_index.json

beicnet, 2018-02-12:
Ok, can you please post your current code again or update it above, and attach the screenshot of that error code?!

mlbranchus, 2018-02-12:
// Adafruit IO Temperature & Humidity Example
// Tutorial Link: https://learn.adafruit.com/adafruit-io-basics-temperature-and-humidity
//
// Adafruit invests time and resources providing this open source code.
// Please support Adafruit and open source hardware by purchasing
// products from Adafruit!
//
// Written by Todd Treece for Adafruit Industries
// Copyright (c) 2016-2017 Adafruit Industries
// Licensed under the MIT license.
//
// All text above must be included in any redistribution.

/************************** Configuration ***********************************/

// edit the config.h tab and enter your Adafruit IO credentials
// and any additional configuration needed for WiFi, cellular,
// or ethernet clients.
#include "config.h"

/************************ Example Starts Here *******************************/
#include <Adafruit_Sensor.h>
#include <DHT.h>
#include <DHT_U.h>
#include <Adafruit_SSD1306.h>
#include <ESP8266WiFi.h>

// oled display
Adafruit_SSD1306 oled = Adafruit_SSD1306();

// pin connected to DH22 data line
#define DATA_PIN 2

// create DHT22 instance
DHT_Unified dht(DATA_PIN, DHT22);

// set up the 'temperature' and 'humidity' feeds
AdafruitIO_Feed *temperature = io.feed("temperature");
AdafruitIO_Feed *humidity = io.feed("humidity");

void setup() {
  oled.begin(SSD1306_SWITCHCAPVCC, 0x3C);  // initialize with the I2C addr 0x3C (for the 128x32)
  oled.display();

  WiFi.begin("annwyn","pirate9d"); //ENTRY HERE
  
  // start the serial connection
  Serial.begin(115200);

  // wait for serial monitor to open
  while(! Serial);

  // initialize dht22
  dht.begin();

  // connect to io.adafruit.com
  Serial.print("Connecting to Adafruit IO");
  io.connect();

  // wait for a connection
  while(io.status() < AIO_CONNECTED) {
    Serial.print(".");
    delay(500);
  }

  // we are connected
  Serial.println();
  Serial.println(io.statusText());

  // text display tests
  oled.setTextSize(1);
  oled.setTextColor(WHITE);
}

void loop() {

  // io.run(); is required for all sketches.
  // it should always be present at the top of your loop
  // function. it keeps the client connected to
  // io.adafruit.com, and processes any incoming data.
  io.run();
  
  sensors_event_t event;
  dht.temperature().getEvent(&event);

  float celsius = event.temperature;
  float fahrenheit = (celsius * 1.8) + 32;

  Serial.print("celsius: ");
  Serial.print(celsius);
  Serial.println("C");

  Serial.print("fahrenheit: ");
  Serial.print(fahrenheit);
  Serial.println("F");

  // save fahrenheit (or celsius) to Adafruit IO
  temperature->save(fahrenheit);

  dht.humidity().getEvent(&event);
   
  Serial.print("humidity: ");
  Serial.print(event.relative_humidity);
  Serial.println("%");

  // save humidity to Adafruit IO
  humidity->save(event.relative_humidity);

  // print it to the OLED
  oled.clearDisplay();
  oled.setCursor(0,0);
  oled.print("SSID: "); oled.println(WIFI_SSID);
  oled.print("IP: "); oled.println(WiFi.localIP());
  oled.print("Temp: "); oled.print(fahrenheit,0); oled.print(" *F ");
  oled.print("Hum: "); oled.print(event.relative_humidity,0); oled.println(" %");
  oled.print("IO Status: ");
  aio_status_t aio_status = io.status();
  Serial.print("Status: "); Serial.println(aio_status);
  switch (aio_status) {
     case AIO_IDLE:  oled.println("IDLE"); break;
     case AIO_DISCONNECTED:
     case AIO_NET_DISCONNECTED:  oled.println("DISCONNECT"); break;
     case AIO_NET_CONNECTED:
     case AIO_CONNECTED_INSECURE:
     case AIO_CONNECTED: oled.println("CONNECTED"); break;
  }
  oled.display();

  // wait 5 seconds (5000 milliseconds == 5 seconds)
  delay(2000);
}

beicnet, 2018-02-12:
I see, you need to remove that line in your code `WiFi.begin("annwyn","pirate9d"); //ENTRY HERE` and all configurations/settings has to be done in the `config.h`.

You just need to follow the Adafruit [tutorial](https://learn.adafruit.com/adafruit-io-basics-temperature-and-humidity) included at a top of your code.

mlbranchus, 2018-02-12:
TonTon told  me to add that line of code.

tonton81, 2018-02-12:
sorry i was not aware the library had a separate file for that, another reason i hate adafruit libraries

mlbranchus, 2018-02-12:
Ok, So all of that code  belongs in config.h instead of .ino?

beicnet, 2018-02-12:
No, there is two file one is `.ino` and the second is `.h`, your settings are located in the `.h` file, there you need to setup your credentials and connection type.

mlbranchus, 2018-02-12:
Yes I know that. All of my credentials and connection type are already in the .h file.

mlbranchus, 2018-02-12:
Ok.

Everything is back to where it was before. And the code 21 fault is still there!

Mike

beicnet, 2018-02-12:
What line give you the fault code? Can you post a screenshot?

mlbranchus, 2018-02-12:
The serial monitor shows status 21...thats the fault code

beicnet, 2018-02-12:
Just that outputs? Nothing else?

mlbranchus, 2018-02-12:
The first 2 show what I get when output fails.
the third shows what serial monitor reports as it is running.
the fourth is just what the status code is i.e. status 21

That's all I have.

beicnet, 2018-02-12:
Can you post a screenshot of it?

mlbranchus, 2018-02-12:
Im going offline in 40 minutes.

I'm going offline for 3-4 hours.

Thanks
Mike

mlbranchus, 2018-02-13:
The first 3 ARE screenshots of it! 
This is all of the imformation about it that ESP is providing.
As I said , the first two screenshots are the error when the connection fails
The third screenshot shows the running serial monitor while the connection is connected.
But the third shows "status 21" every four cycles.

mlbranchus, 2018-02-13:
Hi guys.
Thanks for the work on this.

Have you come with any ideas to fix this fault code?

Mike

beicnet, 2018-02-13:
I don't see / have any screenshots from you!

mlbranchus, 2018-02-13:
We're having a serious failure to communicate here.

I've sent you all of the screen shots that I have. 

I've sent the beginning  and end of the reset comprising of the first and second screenshot(s).

I've sent a screenshot of the running serial monitor shgowing status fault 21 compriising screenshot 3.

All three were sent via email.

Did you receive them?

Mike

Is there a specific screenshot that you're looking for?

Here is the current sketch. 

// Adafruit IO Temperature & Humidity Example
// Tutorial Link: https://learn.adafruit.com/adafruit-io-basics-temperature-and-humidity
//
// Adafruit invests time and resources providing this open source code.
// Please support Adafruit and open source hardware by purchasing
// products from Adafruit!
//
// Written by Todd Treece for Adafruit Industries
// Copyright (c) 2016-2017 Adafruit Industries
// Licensed under the MIT license.
//
// All text above must be included in any redistribution.

/************************** Configuration ***********************************/

// edit the config.h tab and enter your Adafruit IO credentials
// and any additional configuration needed for WiFi, cellular,
// or ethernet clients.
#include "config.h"

/************************ Example Starts Here *******************************/
#include <Adafruit_Sensor.h>
#include <DHT.h>
#include <DHT_U.h>
#include <Adafruit_SSD1306.h>
//#include <ESP8266WiFi.h>

// oled display
Adafruit_SSD1306 oled = Adafruit_SSD1306();

// pin connected to DH22 data line
#define DATA_PIN 2

// create DHT22 instance
DHT_Unified dht(DATA_PIN, DHT22);

// set up the 'temperature' and 'humidity' feeds
AdafruitIO_Feed *temperature = io.feed("temperature");
AdafruitIO_Feed *humidity = io.feed("humidity");

void setup() {
  oled.begin(SSD1306_SWITCHCAPVCC, 0x3C);  // initialize with the I2C addr 0x3C (for the 128x32)
  oled.display();

  //WiFi.begin("annwyn","pirate9d"); //ENTRY HERE
  
  // start the serial connection
  Serial.begin(115200);

  // wait for serial monitor to open
  while(! Serial);

  // initialize dht22
  dht.begin();

  // connect to io.adafruit.com
  Serial.print("Connecting to Adafruit IO");
  io.connect();

  // wait for a connection
  while(io.status() < AIO_CONNECTED) {
    Serial.print(".");
    delay(500);
  }

  // we are connected
  Serial.println();
  Serial.println(io.statusText());

  // text display tests
  oled.setTextSize(1);
  oled.setTextColor(WHITE);
}

void loop() {

  // io.run(); is required for all sketches.
  // it should always be present at the top of your loop
  // function. it keeps the client connected to
  // io.adafruit.com, and processes any incoming data.
  io.run();
  
  sensors_event_t event;
  dht.temperature().getEvent(&event);

  float celsius = event.temperature;
  float fahrenheit = (celsius * 1.8) + 32;

  Serial.print("celsius: ");
  Serial.print(celsius);
  Serial.println("C");

  Serial.print("fahrenheit: ");
  Serial.print(fahrenheit);
  Serial.println("F");

  // save fahrenheit (or celsius) to Adafruit IO
  temperature->save(fahrenheit);

  dht.humidity().getEvent(&event);
   
  Serial.print("humidity: ");
  Serial.print(event.relative_humidity);
  Serial.println("%");

  // save humidity to Adafruit IO
  humidity->save(event.relative_humidity);

  // print it to the OLED
  oled.clearDisplay();
  oled.setCursor(0,0);
  oled.print("SSID: "); oled.println(WIFI_SSID);
  oled.print("IP: "); oled.println(WiFi.localIP());
  oled.print("Temp: "); oled.print(fahrenheit,0); oled.print(" *F ");
  oled.print("Hum: "); oled.print(event.relative_humidity,0); oled.println(" %");
  oled.print("IO Status: ");
  aio_status_t aio_status = io.status();
  Serial.print("Status: "); Serial.println(aio_status);
  switch (aio_status) {
     case AIO_IDLE:  oled.println("IDLE"); break;
     case AIO_DISCONNECTED:
     case AIO_NET_DISCONNECTED:  oled.println("DISCONNECT"); break;
     case AIO_NET_CONNECTED:
     case AIO_CONNECTED_INSECURE:
     case AIO_CONNECTED: oled.println("CONNECTED"); break;
  }
  oled.display();

  // wait 5 seconds (5000 milliseconds == 5 seconds)
  delay(2000);
}

beicnet, 2018-02-13:
Screenshot is when you press the "Print Screen" button on your keyboard and then save it in the mspaint as image and you attach that image to this post!

mlbranchus, 2018-02-13:
I know what a screenshot is!

I took 3 of them several times for you.

I just took another one for you just a few minutes ago.

mlbranchus, 2018-02-13:
what about in the email?

mlbranchus, 2018-02-13:
No it's not

beicnet, 2018-02-13:
Just drag and drop your images here!

tonton81, 2018-02-13:
jeebus, this “issue” is bigger than an average forum topic lol

beicnet, 2018-02-13:
:rofl:

mlbranchus, 2018-02-13:
I've sent MANY email messages with screenshots.

mlbranchus, 2018-02-13:
LOL. driving me insane.

beicnet, 2018-02-13:
So, did you made already account on Adafruit IO, just a simple question

mlbranchus, 2018-02-13:
I will provide whatever you want.

Do you want email? Do you want Github?

Just tell me and I'll do it.

beicnet, 2018-02-13:
Github

mlbranchus, 2018-02-13:
So...how do I open the GitHub information?

beicnet, 2018-02-13:
just upload the images to https://imgur.com/ and paste links here...

mlbranchus, 2018-02-13:
Is this a joke?

mlbranchus, 2018-02-13:
If we could define the where and what I think we can get to the how.

The issue is that the connection software fails to reset automatically. 
I have to manually reset each occurance.

beicnet, 2018-02-13:
Not a joke, if you don't know how to attach images here, then I can't help you, another thing, this is not a support forum, it's a core development branch, so, here you will not get any help regarding you sketch.

You basicaly need to ask your questions on your board manufacturer and code sample page.

And it's located [here](https://forums.adafruit.com/index.php).

Regards

vdeconinck, 2018-02-13:
Guys,
@tonton81 and @beicnet , I know how it feels, I'm always tempted to help users too, but I've been reminded in the past (and rightly so) that this is a tracker for issues in the ESP8266 core for Arduino IDE, not a forum for tracking user errors, and certainly not for teaching people how to attach an image to a comment.
Per https://github.com/esp8266/Arduino/blob/master/POLICY.md , I think the best is to move that discussion over to esp8266.com or stackoverflow and close this issue that is currently just generating spam.

beicnet, 2018-02-14:
@vdeconinck Yes, thats the reason for my last comment to him and it's not that I wont help!

devyte, 2018-02-16:
Alright, enough.
@mlbranchus please request help at a community forum, and discuss there. If you reduce your problem to an issue in the core code hosted in this repo, please open a new issue and provide the requested details, including an mcve sketch. Keep in mind that issues with 3rd party libs such as Adafruit are NOT covered here.
Closing per issue POLICY doc.

closed 2018-02-16 by devyte