esp8266/Arduino #4347: Selected flash size and CPU frequency not passed on to esptool

00pi, 2018-02-11:
Dear all,

relatively new to Arduino and esp8266, I stumbled across the following issue: compiled code of bigger projects fails to be flashed to the esp. This seems to be due to the fact, that the selected flash size and CPU frequency are not passed on to esptool.

command issued by Arduino IDE
`esptool -vv -cd nodemcu -cb 9600 -cp /dev/ttyUSB0 -ca 0x00000 /tmp/arduino_build_523731/esp8266_LED_Matrix2_2018-02-09.ino.bin`

all is working fine when the command is issued manually with the addition of -bz and -bf
`esptool -vv -cd nodemcu -cb 9600 -cp /dev/ttyUSB0 -ca 0x00000 -bz 4M -bf 80 -cf /tmp/arduino_build_523731/esp8266_LED_Matrix2_2018-02-09.ino.bin`

packages/esp8266/hardware/esp8266/2.4.0/platform.txt does not contain the parameters -bz and -bf
`tools.esptool.upload.pattern="{path}/{cmd}" {upload.verbose} -cd {upload.resetmethod} -cb {upload.speed} -cp "{serial.port}" -ca 0x00000 -cf "{build.path}/{build.project_name}.bin"
`

Hardware:			LoLin NodeMcu v3 
Core Version:      	2.4.0

Settings in IDE (1.8.5, from arduino.cc, on OpenSuse 42.3)

Board:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4M (1M SPIFFS)
CPU Frequency:		80Mhz

igrr, 2018-02-11:
It should not be necessary to pass `bz` and `bf` parameters when uploading the binary, because they are only used when generating firmware binaries, not used when uploading.

However, since you say that adding them makes a difference, I assume that there is indeed some bug or side effect in esptool-ck.

Could you please upload the problematic binary somewhere so that i could try to reproduce this? 

Additionally, could you post the verbose log output for both cases: successful upload and failed upload?

00pi, 2018-02-11:
this is the code I'm using
[https://forum.fhem.de/index.php/topic,83774.msg760320.html#msg760320](url)

an unsuccessful flash looks like this
`"/home/00pi/.arduino15/packages/esp8266/tools/esptool/0.4.12/esptool" -eo "/home/00pi/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/bootloaders/eboot/eboot.elf" -bo "/tmp/arduino_build_131154/esp8266_LED_Matrix2_2018-02-09.ino.bin" -bm dio -bf 40 -bz 4M -bs .text -bp 4096 -ec -eo "/tmp/arduino_build_131154/esp8266_LED_Matrix2_2018-02-09.ino.elf" -bs .irom0.text -bs .text -bs .data -bs .rodata -bc -ec
Bibliothek LEDMatrixDriver in Version 0.1.1 im Ordner: /home/00pi/Dokumente/Arduino/libraries/LEDMatrixDriver  wird verwendet
Bibliothek SPI in Version 1.0 im Ordner: /home/00pi/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/libraries/SPI  wird verwendet
Bibliothek ESP8266WiFi in Version 1.0 im Ordner: /home/00pi/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/libraries/ESP8266WiFi  wird verwendet
Der Sketch verwendet 269679 Bytes (25%) des Programmspeicherplatzes. Das Maximum sind 1044464 Bytes.
Globale Variablen verwenden 37184 Bytes (45%) des dynamischen Speichers, 44736 Bytes für lokale Variablen verbleiben. Das Maximum sind 81920 Bytes.
/home/00pi/.arduino15/packages/esp8266/tools/esptool/0.4.12/esptool -vv -cd nodemcu -cb 9600 -cp /dev/ttyUSB0 -ca 0x00000 -cf /tmp/arduino_build_131154/esp8266_LED_Matrix2_2018-02-09.ino.bin 
esptool v0.4.12 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
        setting board to nodemcu
        setting baudrate from 115200 to 9600
        setting port from  to /dev/ttyUSB0
        setting address from 0x00000000 to 0x00000000
        espcomm_upload_file
        espcomm_upload_mem
opening port /dev/ttyUSB0 at 9600
        tcgetattr
        tcsetattr
        serial open
opening bootloader
resetting board
trying to connect
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
trying to connect
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
Uploading 273824 bytes from /tmp/arduino_build_131154/esp8266_LED_Matrix2_2018-02-09.ino.bin to flash at 0x00000000
        erasing flash
        size: 042da0 address: 000000
        first_sector_index: 0
        total_sector_count: 67
        head_sector_count: 16
        adjusted_sector_count: 51
        erase_size: 033000
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        setting timeout 15000
        setting timeout 1000
        espcomm_send_command: receiving 2 bytes of data
        writing flash
................................................................................ [ 29% ]
................................................................................ [ 59% ]
................................................................................ [ 89% ]
....................
`

an successful flash looks like this
`/home/00pi/.arduino15/packages/esp8266/tools/esptool/0.4.12/esptool -vv -cd nodemcu -cb 9600 -cp /dev/ttyUSB0 -ca 0x00000 -bz 4M -bf 80 -cf /tmp/arduino_build_131154/esp8266_LED_Matrix2_2018-02-09.ino.bin 
esptool v0.4.12 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
        setting board to nodemcu
        setting baudrate from 115200 to 9600
        setting port from  to /dev/ttyUSB0
        setting address from 0x00000000 to 0x00000000
setting flash size from 512K to 4M
setting flash frequency from 40 to 80
        espcomm_upload_file
        espcomm_upload_mem
opening port /dev/ttyUSB0 at 9600
        tcgetattr
        tcsetattr
        serial open
opening bootloader
resetting board
trying to connect
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
trying to connect
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
Uploading 273824 bytes from /tmp/arduino_build_131154/esp8266_LED_Matrix2_2018-02-09.ino.bin to flash at 0x00000000
        erasing flash
        size: 042da0 address: 000000
        first_sector_index: 0
        total_sector_count: 67
        head_sector_count: 16
        adjusted_sector_count: 51
        erase_size: 033000
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        setting timeout 15000
        setting timeout 1000
        espcomm_send_command: receiving 2 bytes of data
        writing flash
................................................................................ [ 29% ]
................................................................................ [ 59% ]
................................................................................ [ 89% ]
............................                                                     [ 100% ]
starting app without reboot
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        espcomm_send_command: receiving 2 bytes of data
closing bootloader
`

00pi, 2018-02-11:
[esp8266_LED_Matrix2_2018-02-09.ino.bin.zip](https://github.com/esp8266/Arduino/files/1714229/esp8266_LED_Matrix2_2018-02-09.ino.bin.zip)

earlephilhower, 2019-01-24:
#5635 would sidestep this, although I'm not sure I see where there is an issue...

earlephilhower, 2019-02-25:
#5635 has merged, but after re-reading this bug I don't know what the real issue was.  Please give it a go and see if it fixes things, @00pi .

earlephilhower, 2019-07-25:
Closing as there's no feedback in months and we're not using the original esptool.exe anymore.

closed 2019-07-25 by earlephilhower