esp8266/Arduino #4350: Webserver double freeing _currentHeaders on destruction

mongozmaki, 2018-02-11:
Destructor of ESP8266WebServer double-frees _currentHeaders.
The problem is, that the destructor frees `_currentHeaders` and sets `_headerKeysCount=0` and than calls `close()`, which checks `if(!_headerKeysCount)` (=true) and calls `collectHeaders(0, 0)` which frees `_currentHeaders` again (because it is not set to `nullptr`).
A fix would be to call `close()` in the beginning of the destructor.
A more extensive fix would be to call collectHeaders(0, 0) in `begin()` and not in `close()`.

Code fixed:
```cpp
ESP8266WebServer::~ESP8266WebServer() {
  //Close here
  close();
  if (_currentHeaders) {
    delete[]_currentHeaders;
  }
  _headerKeysCount = 0;
  RequestHandler* handler = _firstHandler;
  while (handler) {
    RequestHandler* next = handler->next();
    delete handler;
    handler = next;
  }
  //Not here
  //close();
}
```

### Sketch

```cpp

#include <Arduino.h>

void setup() {
Serial.begin(115200);
  {
    Serial.println("Creating server");
    ESP8266WebServer server;
    const char * headerkeys[] = { "If-None-Match" };
    Serial.println("Setting headers");
    //Crashes also without calling collectHeaders
    server.collectHeaders(headerkeys, 1);
    Serial.println("Begin server");
    server.begin();
    Serial.println("About to delete server server");
  } //Deleted here
  Serial.println("Server deleted");
}

void loop() {
}
```

### Debug Messages

```
Creating server
Setting headers
Begin server
About to delete server server
*** CRASH ***
```

d-a-v, 2018-02-13:
That would make a good pull-request.

earlephilhower, 2018-02-13:
@mongozmaki Would you like to submit a PR for this?  

It will also fix the same problem on the SecureServer, which uses this class' destructor.

mongozmaki, 2018-02-14:
Yes, I tried creating a PR but didn't work for some reason. If I find a minor issue again, I will do that.
Thanks for the fix!

earlephilhower, 2018-02-14:
No problem, @mongozmaki .  I've referenced you in the change note, and appreciate the detailed error, repro instructions, and fix you did.  Nothing for me to do, really. :)

Next one, when you push to your private forked repo, you can go to its github website and there should be a "submit pull request" button right at the top of the page (for about 5 mins after a push) which should get you going.

closed 2018-02-14 by earlephilhower