esp8266/Arduino #4351: Missing __offsetof macro to use STAILQ_LAST macro from queue.h

crousti-bat, 2018-02-11:
### Basic Infos

#### Hardware
Hardware:			Wemos D1mini R2
Core Version:      	2.3.0

### Description

Problem description

### Settings in IDE

Module:  			Wemos D1 R2 & mini
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			-
Flash Frequency:	-
Upload Using:		OTA 
Reset Method:		-

### Sketch

```cpp

#include <Arduino.h>

typedef struct transitionQueueElem_s {
  int16_t  stepBrightness;
  uint16_t stepDelay;
  uint16_t stepCount;
  uint16_t targetBrightness;
  STAILQ_ENTRY(transitionQueueElem_s) transitionEntry;
} transitionQueueElem_t;

typedef STAILQ_HEAD(, transitionQueueElem_s) transitionQueueHead_t;

transitionQueueHead_t
void setup() {
transitionQueueHead_t transitionQueue;
}

void loop() {
/* added the next line to show the problem */
STAILQ_ENTRY(transitionQueueElem_s) tEntry;

/* this one will prevent compiling */
STAILQ_LAST(&transitionQueue),transitionQueueElem_s, tEntry);

/* this one works perfectly */
STAILQ_INSERT_TAIL(&transitionQueue, pTransition, tEntry);
}
```

### Debug Messages

```
C:\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include/queue.h:154:67: error: '__offsetof' was not declared in this scope

       ((char *)((head)->stqh_last) - __offsetof(struct type, field))))
```

Other list functions work, but it looks like only this one uses __offsetof. The compiler initially threw an error on the "tEntry" that is defined in a separate .h, so i tried declaring it right before.

Secondary: STAILQ_LAST implementation macros on other systems use 1 argument, but this one uses 3. Is the queue.h up to date ?

devyte, 2018-02-17:
Your code seems to have a lot of errors, including basic wrong parenthesis usage. This builds for me:
```
#include <Arduino.h>
#include "queue.h"

typedef struct transitionQueueElem_s {
  int16_t  stepBrightness;
  uint16_t stepDelay;
  uint16_t stepCount;
  uint16_t targetBrightness;
  STAILQ_ENTRY(transitionQueueElem_s) transitionEntry;
} transitionQueueElem_t;

typedef STAILQ_HEAD(, transitionQueueElem_s) transitionQueueHead_t;

transitionQueueHead_t transitionQueue;

void setup() {

}

void loop() {
  transitionQueueElem_t *pTransition ;
  /* added the next line to show the problem */
  STAILQ_ENTRY(transitionQueueElem_s) tEntry;

  /* this one will prevent compiling */
  STAILQ_LAST(&transitionQueue, transitionQueueElem_s, transitionEntry);

  /* this one compiles, if the STAILQ_LAST is commented out */
  STAILQ_INSERT_TAIL(&transitionQueue, pTransition, transitionEntry);
}
```

Closing as user error.

closed 2018-02-17 by devyte

crousti-bat, 2018-02-18:
The code was an extract to show the problem, it was not yet integrated in the repo. I have updated the project, which is here:
https://github.com/crousti-bat/master-dimmer

It will compile once you remove the error directive in the settings.
replace line ~852 "calling"MY_STAILQ_LAST by this and the error will show as transitionEntry not defined.
    pBaseState = STAILQ_LAST(&(gLedStrip[index].transitionQueue),transitionQueueElem_s,transitionEntry);

declare transitionEntry and the offsetof not defined will show.