esp8266/Arduino #4352: WiFi.status() still returns WL_CONNECTED when after access point is turned off

toomasz, 2018-02-11:
### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	2.4

### Description

Test scenario: 
1.Start sketch, access point is disabled
2.Enable access point
3.Sketch can connect to access point
4.Disable access point
5. WiFi.Status() still returns WL_CONNECTED

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP12-E)
Flash Size: 		4MB
CPU Frequency:		160Mhz
Upload Using:		Serial
Reset Method:		nodemcu

### Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>

void setup()
{
	Serial.begin(115200);
	Serial.println("Setup");
}
bool wifiConnect(int timeout)
{
	Serial.println("Connecting to wifi...");
	uint64_t start = millis();

	WiFi.setAutoReconnect(false);
	WiFi.setAutoConnect(false);
	WiFi.mode(WiFiMode_t::WIFI_STA);
	auto connectResult = WiFi.begin("AndroidAP", "*******");
	
	while (WiFi.status() != WL_CONNECTED)
	{
		delay(100);
		Serial.print(".");

		if (millis() - start > timeout)
		{
			WiFi.disconnect();
			return false;
		}
	}
	
	Serial.println();
	return true;
}

wl_status_t prevStatus;
void loop()
{
	
	auto wifiState = WiFi.status();
	if (WiFi.status() != prevStatus)
	{
		Serial.printf("Wifi State changed to %s\n", WlStatusToStr(WiFi.status()));
		prevStatus = WiFi.status();
	}

	if (!WiFi.isConnected())
	{		
		if (wifiConnect(5000))
		{
			Serial.println("Successfully connected");
		}
		else
		{
			Serial.println("Timeout");
		}
		delay(1000);
	}
}

const char* WlStatusToStr(wl_status_t wlStatus)
{
	switch (wlStatus)
	{
	case WL_NO_SHIELD: return "WL_NO_SHIELD";
	case WL_IDLE_STATUS: return "WL_IDLE_STATUS";
	case WL_NO_SSID_AVAIL: return "WL_NO_SSID_AVAIL";
	case WL_SCAN_COMPLETED: return "WL_SCAN_COMPLETED";
	case WL_CONNECTED: return "WL_CONNECTED";
	case WL_CONNECT_FAILED: return "WL_CONNECT_FAILED";
	case WL_CONNECTION_LOST: return "WL_CONNECTION_LOST";
	case WL_DISCONNECTED: return "WL_DISCONNECTED";
	default: return "Unknown";
	}
}
```

### Debug Messages

```
Connecting to wifi...
.....................scandone
wifi evt: 1
STA disconnect: 201
..........................Timeout
Connecting to wifi...
.....................scandone
wifi evt: 1
STA disconnect: 201
..........................Timeout
Connecting to wifi...
.......wifi evt: 2
..............scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with AndroidAP, channel 6
dhcp client start...
wifi evt: 0
.ip:192.168.43.112,mask:255.255.255.0,gw:192.168.43.1
wifi evt: 3
.
Successfully connected
Wifi State changed to WL_CONNECTED
pm open,type:2 0
state: 5 -> 2 (3c0)
rm 0
pm close 7
wifi evt: 1
STA disconnect: 3
```

LucaEstiva, 2019-01-30:
WiFi connection status tested ( On my scenario is WORKING ):
The simple scenario is one Lolin as STATION one as AP
Development board Lolin V3:

STATION CODE:

  //
  #include <ESP8266WiFi.h>

  #define WLAN_SSID "ESPap"
  #define WLAN_PASS "12345678"

    // 5 Sec interval
  #define INTERVAL_5 5000
  
  // 
  uint32_t time_1 = 0;
  //
  uint32_t StartTime = 0;
  //
  int32_t TimeLapsed = 0;

  // Expands to <HOST NAME>-<last 4 decimal chars of MAC address>
  #define WIFI_HOSTNAME "%s-%04d"
  
  // WiFi STATUS STRINGS
  const char* WiFiStat[]={"WL_IDLE_STATUS", "WL_NO_SSID_AVAIL", "WL_SCAN_COMPLETED",
  "WL_CONNECTED", "WL_CONNECT_FAILED", "WL_CONNECTION_LOST", "WL_DISCONNECTED"};
  
  // Composed Wifi hostname
  char this_hostname[33];

  // CURRENT WiFi CONNECTION STATUS
  uint8_t wifi_status;

  // WiFi FIRST TIME CONNECTION STATUS FLAG
  bool WiFi_Connected = false;
  // WiFi DISCONNECTED STATUS  FLAG
  bool WiFi_Disconnected = false;

  //
  int WiFi_Connection_Time_Out = 15;
  //
  int WiFi_Connection_Time = 0;

//------------------------------------------------------------------------------------------
// SETUP
//------------------------------------------------------------------------------------------
void setup()
{
  
  //
  Serial.begin(115200);
  //
  Serial.printf("%s%d\n", "Chip id: ", ESP.getChipId());

  // Compose the host name string
  snprintf_P( this_hostname, sizeof( this_hostname )-1, WIFI_HOSTNAME, "TEST", ESP.getChipId() & 0x1FFF );

  // At this point WiFi state is WL_DISCONNECTED
  Serial.printf("1: Connection status: %s\n", WiFiStat[WiFi.status()] );
  
  // Set up basic WiFi parameters
  WiFiStationSetup();
  // At this point WiFi state is WL_IDLE_STATUS
  Serial.printf("2: Connection status: %s\n", WiFiStat[WiFi.status()] );

}

//------------------------------------------------------------------------------------------
[... 371 lines cut ...]
  server.begin();

  //
  Serial.println("HTTP server started");
}

//------------------------------------------------------------------------------------------
// WEB SERVER HANDLE NOT FOUND PAGE - 404 NOT FOUND
//------------------------------------------------------------------------------------------
void loop()
{
  //
  server.handleClient();
}

void onNewStation(WiFiEventSoftAPModeStationConnected sta_info)
{

  char last_mac[18];
  Serial.println("New Station :");
  sprintf(last_mac,"%02X:%02X:%02X:%02X:%02X:%02X", MAC2STR(sta_info.mac));
  Serial.printf("MAC address : %s\n",last_mac);
  Serial.printf("Id : %d\n", sta_info.aid);

}

d-a-v, 2019-01-30:
@toomasz Please confirm for the fix in latest releases

devyte, 2020-02-02:
Closing in view of @LucaEstiva 's comment and due to lack of feedback by OP.

closed 2020-02-02 by devyte

HaunsTM, 2021-03-08:
Are there any news here? I'm facing the same issue on Mar 8, 2021?

JorgeVelez, 2022-04-09:
same issue april 2022