esp8266/Arduino #4354: ESP8266Webserver warning on string concat

Sparatutto, 2018-02-12:
Recently I have updated the library as you suggested me with the GIT and I have a warning on a string concatenation.
Second I see new iwIp variant names and not sure what they are.
One is for memory saving and the second for bandwidth. Not sure the meaning of this second.

I'm compiling with the warning and I did test both iwip.
No difference with the two version and no issues with the warning.

#### Hardware

[image]

### Description

I think  + ':auth:' should be with "" instead.

### Sketch

```

### Debug Messages

```
:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WebServer\src\ESP8266WebServer.cpp:198:93: warning: character constant too long for its type [enabled by default]

         md5.add(_H1 + FPSTR(colon) + _nonce + FPSTR(colon) + _nc + FPSTR(colon) + _cnonce + ':auth:' + _H2);

                                                                                             ^

```

earlephilhower, 2018-02-12:
Please try the GIT latest.  Commit #4338 fixes this and a few other problems.

Sparatutto, 2018-02-12:
@earlephilhower thanks, i will try to update tomorrow.
FYI, i did update on friday and the fix seems to be before my git.
Anyway I will tr tomorrow.
thx!

Sparatutto, 2018-02-13:
Updated now. Ok no more warning. Thanks

closed 2018-02-13 by Sparatutto