esp8266/Arduino #4358: Device "freezes" at startup when connecting a buzzer to GPIO16

bracci, 2018-02-13:
### Basic Infos

#### Hardware
Hardware:	        Wemos mini with ESP-12
Core Version:      	2.3.0

### Description

When connecting a 3.3 V Buzzer directly to GPIO16 (DO) and GND the device "freezes" at startup. I guess, it just boots in an unexpected mode. Cannot' find out why this happens. When I let the GPIO16 disconnected or when I tie it to 3.3V or even when I pull it down to GND everything works fine - just not if I connect the Buzzer. Any idea? 

### Settings in IDE

Module:  			Wemos D1 mini
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			-
Flash Frequency:	921600
Upload Using:		SERIAL
Reset Method:		-

WereCatf, 2018-02-13:
The ESP8266 outputs some sort of a short signal on GPIO16 at boot (I don't know what the purpose of that signal is), so that's most likely related to it -- perhaps the buzzer is drawing too much current from the GPIO16 during this signal? If you were curious, you could try adding a resistor in series and seeing if it changes anything, though that would obviously limit the volume of the buzzer.

bracci, 2018-02-13:
Hi, thanks for answering. I had exactly the same thought. I put a resistor in series with the buzzer. The unwanted behaviour disappeared. Problem here: the sound level is not enough anymore. To be sure it'd be really current related, i shortened the GPIO16 to GND, obviously resulting in a higher current than with the buzzer. But with this everything works as normal. Absolutely no clue... :-/

WereCatf, 2018-02-13:
Well, either use some other GPIO-pin, or drive the buzzer via a transistor or a logic-level MOSFET connected to GPIO16 instead.

tablatronix, 2018-02-14:
Gpio is special and fairly well documented.
http://www.esp8266.com/wiki/doku.php?id=esp8266_gpio_pin_allocations

bracci, 2018-02-14:
Thanks for your answers. For sure there are options to get rid of that, e.g. with a transistor or FET, but I have already the PCB manufactured (only a small batch of 10 pcs.). In the previous PCB version, the buzzer was driven directly by GPIO15. Due to more sensors needed in the new design, I had to remap the I/Os. So I changed the buzzer to GPIO16. Unfortunately, also after reading your linked documentation I don't understand, why even connecting GPIO16 directly to GND (which definitely leads to a higher current than connecting the buzzer) works, but a buzzer connected to it doesn't. It seems that no one really understands why it's behaving like that. So I will simply order a new PCB with a changed pin map (although I'd like to understand where the problem is). Thank you anyway.

RudyFiero, 2018-02-15:
You didn't indicate the type of buzzer. As far as piezo buzzers I have had to place a small value resistor in series with them to limit the disruption they cause to other signals. The resistor doesn't affect the sound level but it does control the current. One of the first products I designed had a buzzer connected to 5v and a transistor driving it. (and a higher value resistor in parallel with the buzzer) When I had problems with the prototype run one of the guys I worked with told me about the series resistor. I have included the resistor in many designs since then. (35 years)

devyte, 2018-02-16:
Very interesting, but not the right place to discuss this. Closing per issue POLICY doc.

closed 2018-02-16 by devyte