esp8266/Arduino #4364: Is there a way to hang up on an unresponsive web client?

fizgig10, 2018-02-14:
I start a web server with the basic commands:

```
WiFiClient client = server.available();
	if (!client)
		return;
```

When I get a client I wait for bytes to come in:

```
Serial.println("new client");
	while(!client.available())
		delay(1);
```

Trouble is, something on my network decides to start a connection with it causing the "new client" connection to show up on my serial log.  I guess the new client doesn't send any data though as the program hangs at the while statement waiting for data which I guess never comes.  How do I get the web server to hang up on the unresponsive client?

dragondaud, 2018-02-14:
This is an issue forum, not a general help forum. You should close this issue, and try http://www.esp8266.com/ for general help.

That being said, https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/server-class.html says client.close() is a thing.

fizgig10, 2018-02-14:
Woops.  Misuse of forum.  Closing now.

closed 2018-02-14 by fizgig10