esp8266/Arduino #4367: MQTT PSK based encryption

korish, 2018-02-14:
Hi,

According to [Mosquitto documentation](https://mosquitto.org/man/mosquitto-conf-5.html), both certificate and PSK based encryption is supported. I've managed to connect esp8266 to mosquitto broker using certificate based encryption. Now I want to try PSK based encryption.

1) Does esp8266 support this kind of encryption?
2) Is there any examples available?

Thanks!

igrr, 2018-02-14:
At the moment axTLS is used as the TLS stack. It doesn't support PSK.

earlephilhower, 2018-02-14:
BearSSL (#4273) also doesn't support true PSK, if I understand it correctly.  It only supports pre-shared keys in the sense that it ignores x509 certificates completely on a connection and will use a public key you give it (instead of the one on an x509 cert).  So it saves time(1-2 seconds) and memory(~2K) on setup.  If you're not doing massive amounts of short connections, I'm not sure it's such a win, honestly.

I see mbedTLS supports TLS-PSK (per wikipedia), so if you really need it then I think the ESP32 maybe in your future as I think it uses that library.

devyte, 2018-02-16:
Closing as won't fix.

closed 2018-02-16 by devyte