esp8266/Arduino #4375: unable to upload with Arduino IDE and NodeMCU Flasher

ts7622, 2018-02-17:
### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	2.4.0

### Description

Yesterday i uploading my sketch and it worked very well. Today i tried to upload the exact same sketch and it doesnt work. I never shutdown my computer completely so you can say i did upload (worked) -> sleep -> upload (error). I get following Output:

> warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

This problem is already documented, but i cant get it to work. I clicked through every single issues page here on Github and read every topic which could contain my problem even just a little bit. I wasted like 6 hours today and asked in two german forums for help but they couldnt resolve my problem.

I tested, i would say, nearly every possible board settings (although it worked with "NodeMCU 1.0" very well before), reinstalled Arduino three times, checked the external power supply (5V, 2A) and the USB-cable but ... yeah. I dont know. Even flashing with the NodeMCU Flasher Software wont work.

So i got [this board here from Banggood](https://www.banggood.com/Geekcreit-Doit-NodeMcu-Lua-ESP8266-ESP-12E-WIFI-Development-Board-p-985891.html?rmmds=myorder) which i ordered a couple of times before and worked very well for me. The driver is also from the article page. Im trying to control some relays and had the day before a test script running to see if my hardware works as aspected. If i try to flash it with the Arduino IDE or NodeMCU Flasher the blue LED onboard is flashing every ~2 seconds (Arduino IDE)/ ~0,5 seconds (NodeMCU Flasher). In the serial monitor at 74880 baudrate i get in normal-mode ...

>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
>
>load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

... and in flashing-mode

> ets Jan  8 2013,rst cause:2, boot mode:(1,6)

Oh yeah ... here are the port settings from the device manager. Sry for german, but the informations are at the same spot ;)

[image]

I dont know what to do by now. 

### Settings in IDE

(In general every possible combination, but here we go ...)

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp
#include <OneWire.h> 
#include <DallasTemperature.h>

#define ONE_WIRE_BUS 1 

OneWire oneWire(ONE_WIRE_BUS); 
DallasTemperature sensors(&oneWire);

void setup() {
  Serial.begin(9600);
  pinMode(16, OUTPUT);
  pinMode(5, OUTPUT);
  pinMode(4, OUTPUT);
  pinMode(0, OUTPUT);
  pinMode(2, OUTPUT);
  pinMode(14, OUTPUT);
  pinMode(12, OUTPUT);
  pinMode(13, OUTPUT);
}

void loop() {
  sensors.requestTemperatures();
  Serial.print(sensors.getTempCByIndex(0)); 
  digitalWrite(16, HIGH);
  delay(1000);
  digitalWrite(5, HIGH);
  delay(1000);
  digitalWrite(4, HIGH);
  delay(1000);
  digitalWrite(0, HIGH);
  delay(1000);
  digitalWrite(2, HIGH);
  delay(1000);
  digitalWrite(14, HIGH);
  delay(1000);
  digitalWrite(12, HIGH);
  delay(1000);
  digitalWrite(13, HIGH);
  delay(1000);
  digitalWrite(16, LOW);
  delay(1000);
  digitalWrite(5, LOW);
  delay(1000);
  digitalWrite(4, LOW);
  delay(1000);
  digitalWrite(0, LOW);
  delay(1000);
  digitalWrite(2, LOW);
  delay(1000);
  digitalWrite(14, LOW);
  delay(1000);
  digitalWrite(12, LOW);
  delay(1000);
  digitalWrite(13, LOW);
  delay(1000);
}
```

WereCatf, 2018-02-17:
Disconnect anything you have connected to the GPIO-pins, push down the Flash-button and, while keeping the Flash-button down, press the Reset-button once, then let go of the Flash-button just a moment before the Arduino IDE would begin to flash it -- this usually does the trick for me.

devyte, 2018-02-17:
Forget ESPlorer, it's aimed at lua and micropython more than us.
If even the Nodemcu flasher won't work, then it's either your OS trolling you (e.g. windows is known to swap com ports around), or your board broke.
Me, I use Linux in a VirtualBox on a Windows machine, and don't have trouble (except for bad usb cables, because chinese).
In any case, this is obviously not an issue with the repo code, so this is not the right place to dicuss.
Closing as off-topic.

closed 2018-02-17 by devyte

Slayer4477, 2020-02-18:
change tx pin to gpio ,,, i solve this problem with ota upload

void setup() {
  Serial.begin(9600);
  while (!Serial);
WiFi.hostname("xxxxx");      // DHCP Hostname (useful for finding device for static lease)
  ArduinoOTA.setHostname("xxxx");
  ArduinoOTA.setPort(8266);
  ArduinoOTA.setPassword("");
  ArduinoOTA.onStart([]() {

    /////////////////change pin to gpio////////
    //GPIO 1 (TX) swap the pin to a GPIO.
    pinMode(1, FUNCTION_3);
    //GPIO 3 (RX) swap the pin to a GPIO.
    pinMode(3, FUNCTION_3);
    ///////////////////////
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH) {
      type = "sketch";
    } else { // U_SPIFFS
      type = "filesystem";
    }

    // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
    Serial.println("Start updating " + type);
  });
  ArduinoOTA.onEnd([]() {
    ////////////////change pin to rx tx again//////////
    //GPIO 1 (TX) swap the pin to a TX.
    pinMode(1, FUNCTION_0);
    //GPIO 3 (RX) swap the pin to a RX.
    pinMode(3, FUNCTION_0);
    ///////////////////////
    Serial.println("\nEnd");
  });
  ArduinoOTA.begin();
  delay(1000);
}