esp8266/Arduino #4382: adafruit mqtt esp8266 example error

SyedShababHaider, 2018-02-17:
### Basic Infos
trying to run a mqtt esp8266 example and having this error
#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description
trying to run a mqtt esp8266 example and having this error

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
Arduino: 1.8.5 (Windows 7), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (1M SPIFFS), v2 Prebuilt (MSS=536), Disabled, None, 115200"

Build options changed, rebuilding all
C:\Program Files (x86)\Arduino\libraries\Adafruit_MQTT_Library-master\Adafruit_MQTT_Client.cpp: In member function 'virtual bool Adafruit_MQTT_Client::sendPacket(uint8_t*, uint16_t)':

C:\Program Files (x86)\Arduino\libraries\Adafruit_MQTT_Library-master\Adafruit_MQTT_Client.cpp:84:38: error: no matching function for call to 'min(uint16_t&, int)'

       uint16_t sendlen = min(len, 250);

                                      ^

C:\Program Files (x86)\Arduino\libraries\Adafruit_MQTT_Library-master\Adafruit_MQTT_Client.cpp:84:38: note: candidates are:

In file included from c:\users\shabab haider\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:62:0,

                 from C:\Users\Shabab Haider\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/Arduino.h:240,

                 from C:\Program Files (x86)\Arduino\libraries\Adafruit_MQTT_Library-master\Adafruit_MQTT.h:25,

                 from C:\Program Files (x86)\Arduino\libraries\Adafruit_MQTT_Library-master\Adafruit_MQTT_Client.h:26,

                 from C:\Program Files (x86)\Arduino\libraries\Adafruit_MQTT_Library-master\Adafruit_MQTT_Client.cpp:22:

c:\users\shabab haider\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4226:5: note: template<class _Tp, class _Compare> _Tp std::min(std::initializer_list<_Tp>, _Compare)

     min(initializer_list<_Tp> __l, _Compare __comp)

     ^

c:\users\shabab haider\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4226:5: note:   template argument deduction/substitution failed:

C:\Program Files (x86)\Arduino\libraries\Adafruit_MQTT_Library-master\Adafruit_MQTT_Client.cpp:84:38: note:   mismatched types 'std::initializer_list<_Tp>' and 'short unsigned int'

       uint16_t sendlen = min(len, 250);

                                      ^

In file included from c:\users\shabab haider\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:62:0,

                 from C:\Users\Shabab Haider\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/Arduino.h:240,

                 from C:\Program Files (x86)\Arduino\libraries\Adafruit_MQTT_Library-master\Adafruit_MQTT.h:25,

                 from C:\Program Files (x86)\Arduino\libraries\Adafruit_MQTT_Library-master\Adafruit_MQTT_Client.h:26,

                 from C:\Program Files (x86)\Arduino\libraries\Adafruit_MQTT_Library-master\Adafruit_MQTT_Client.cpp:22:

c:\users\shabab haider\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4221:5: note: template<class _Tp> _Tp std::min(std::initializer_list<_Tp>)

     min(initializer_list<_Tp> __l)

     ^

c:\users\shabab haider\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4221:5: note:   template argument deduction/substitution failed:

C:\Program Files (x86)\Arduino\libraries\Adafruit_MQTT_Library-master\Adafruit_MQTT_Client.cpp:84:38: note:   mismatched types 'std::initializer_list<_Tp>' and 'short unsigned int'

       uint16_t sendlen = min(len, 250);

                                      ^

In file included from c:\users\shabab haider\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:61:0,

                 from C:\Users\Shabab Haider\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/Arduino.h:240,

                 from C:\Program Files (x86)\Arduino\libraries\Adafruit_MQTT_Library-master\Adafruit_MQTT.h:25,

                 from C:\Program Files (x86)\Arduino\libraries\Adafruit_MQTT_Library-master\Adafruit_MQTT_Client.h:26,

                 from C:\Program Files (x86)\Arduino\libraries\Adafruit_MQTT_Library-master\Adafruit_MQTT_Client.cpp:22:

c:\users\shabab haider\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:239:5: note: template<class _Tp, class _Compare> const _Tp& std::min(const _Tp&, const _Tp&, _Compare)

     min(const _Tp& __a, const _Tp& __b, _Compare __comp)

     ^

c:\users\shabab haider\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:239:5: note:   template argument deduction/substitution failed:

C:\Program Files (x86)\Arduino\libraries\Adafruit_MQTT_Library-master\Adafruit_MQTT_Client.cpp:84:38: note:   deduced conflicting types for parameter 'const _Tp' ('short unsigned int' and 'int')

       uint16_t sendlen = min(len, 250);

                                      ^

In file included from c:\users\shabab haider\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:61:0,

                 from C:\Users\Shabab Haider\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/Arduino.h:240,

                 from C:\Program Files (x86)\Arduino\libraries\Adafruit_MQTT_Library-master\Adafruit_MQTT.h:25,

                 from C:\Program Files (x86)\Arduino\libraries\Adafruit_MQTT_Library-master\Adafruit_MQTT_Client.h:26,

                 from C:\Program Files (x86)\Arduino\libraries\Adafruit_MQTT_Library-master\Adafruit_MQTT_Client.cpp:22:

c:\users\shabab haider\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:193:5: note: template<class _Tp> const _Tp& std::min(const _Tp&, const _Tp&)

     min(const _Tp& __a, const _Tp& __b)

     ^

c:\users\shabab haider\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:193:5: note:   template argument deduction/substitution failed:

C:\Program Files (x86)\Arduino\libraries\Adafruit_MQTT_Library-master\Adafruit_MQTT_Client.cpp:84:38: note:   deduced conflicting types for parameter 'const _Tp' ('short unsigned int' and 'int')

       uint16_t sendlen = min(len, 250);

                                      ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

devyte, 2018-02-17:
@SyedShababHaider said:

> C:\Program Files (x86)\Arduino\libraries\Adafruit_MQTT_Library-master\Adafruit_MQTT_Client.cpp:84:38: error: no matching function for call to 'min(uint16_t&, int)'

The error is obviously coming from the adafruit lib, so why are you reporting it here?

I don't use their libs, but I think they fixed it in their repo, and haven't released, so try their latest git.

Closing as off topic.

closed 2018-02-17 by devyte

tonton81, 2018-02-17:
this isnt a forum for adafruit code issues. close this “issue” as it’s irrelevant here