esp8266/Arduino #4383: Timezone.h works with esp board 2.3.0 but not for 2.4.0

tofrnr, 2018-02-18:
hello
why  compiles Timezone.h 
https://github.com/JChristensen/Timezone
with esp board 2.3.0 but not for 2.4.0 ?
perhaps which different lib or board revision?

devyte, 2018-02-18:
@tofrnr yous issue is for a 3rd party lib, which is in a different repo, so why are you reporting it here?
The lib README in their repo says it requires Arduino Time lib, which after a quick search provides the missing symbols in your errors.
Closing as user error.

closed 2018-02-18 by devyte

tofrnr, 2018-02-18:
of course the Arduino Time lib is installed and included.

I am reporting it here because this Arduino standard lib works with esp8266 board driver 2.3.0, but it does NOT work with esp8266 board driver 2.4.0 any more.

tofrnr, 2018-03-05:
update:
I now also reported this  issue here:
https://github.com/JChristensen/Timezone/issues/29