esp8266/Arduino #4387: Upload process hangs/stalls constantly and won't complete

aardvarkk, 2018-02-19:
### Basic Infos

#### Hardware
Hardware: ESP8266 (Sonoff Basic Wi-Fi Smart Switch)

### Description

When uploading my sketch to the device, it constantly hangs. It has gotten to various points (sometimes as high as about 70%), but then the `...` just stops moving completely and never restarts. I have yet to successfully complete an upload after many attempts.

If I try to unplug my USB serial cable, the serial port disappears and I have to reboot to try again.

I'm using the latest FTDI driver (`  162    0 0xffffff7f837dd000 0x7000     0x7000     com.FTDI.driver.FTDIUSBSerialDriver (2.4.2) E16D9540-AF63-36AD-94C4-B1164FBD6078 <109 55 5 4 3 1>`), Arduino 1.8.5, and ESP8266 Boards v2.4.0. I'm running macOS 10.13.3 (High Sierra).

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1M (no SPIFFS)
CPU Frequency:		80Mhz
Flash Mode:			DOUT
Flash Frequency:	40Mhz
Upload Using:		FTDI
Reset Method:		ck

### Sketch

```cpp
void setup() {
  pinMode(13, OUTPUT);
}

void loop() {
  digitalWrite(13, LOW);
  delay(500);
  digitalWrite(13, HIGH);
  delay(5000);
}
```

### Debug Messages

Sketch starts uploading, but stalls at various points during the upload procedure.

aardvarkk, 2018-02-19:
I just tried rebooting and changing the upload rate to 57600 -- same issue. Here's a screenshot. The dots quickly started filling, then just stop and never restart.

[image]

aardvarkk, 2018-02-19:
This is the FTDI board I'm using: http://amzn.to/2BD29MD

Before upgrading to the latest FTDI chip, I had problems with the default Apple driver where it wouldn't even find the ESP8266 to connect to. With the new drivers the upload process starts but never finishes.

aardvarkk, 2018-02-19:
I tried uploading the file generated by Arduino IDE directly in `esptool` and got the following output:

```
		espcomm_send_command: response 0x20120707
.		espcomm_send_command: sending command header
		espcomm_send_command: sending command payload
		espcomm_send_command: receiving 2 bytes of data
		espcomm_send_command: received 2 bytes:
		0x00
		0x00
		espcomm_send_command: response 0x20120707
.		espcomm_send_command: sending command header
		espcomm_send_command: sending command payload
```

I'm unable to kill the process with`Ctrl-C`. Looks like it's just stalling waiting for a response from the serial connection and it never receives anything. No idea what could be causing it...

tonton81, 2018-02-19:
i would change the usb cable

tonton81, 2018-02-19:
and possibly the uart adaptor, btw, make sure its 3.3v

aardvarkk, 2018-02-19:
I've already tried changing the USB cable and the voltage is definitely set to 3.3V. I'm able to get a full write going with `esptool.py`, but after writing the .bin file and hard resetting the blink program doesn't seem to run.

joba-1, 2018-03-02:
Had that issue before. The current the ftdi could provide during flash was not enough. I added a separate 3.3V supply and flash worked fine.

aardvarkk, 2018-03-02:
@joba-1 I think you're exactly right. I came up with the same theory about a week ago, and I think my power supplies are arriving today. Hopefully that fixes the issue. Thanks so much for the idea! I'll close this issue if it all works out.

joba-1, 2018-03-02:
good luck 😉
-- 
iphone so i typo

aardvarkk, 2018-03-11:
Alrighty -- I'm able to program it now! The two critical things seemed to be:

1. Getting a new USB-UART programmer. Instead of using the FTDI-based one I had, I purchased a Silicon Labs CP2104-based programmer (the Adafruit Friend), and the hanging no longer occurred when trying to program the device.
2. Even after that, flashing worked properly but the device seemed to be lifeless after programming. Within the Arduino IDE, it was critical that I set Flash Mode to `DOUT` -- otherwise, the flashing would go fine but nothing would happen afterward. Now I've got a simple blink running, and serial output! All is good!

closed 2018-03-11 by aardvarkk