esp8266/Arduino #4388: HTTPClient works in loop() but not in onPulse() (interrupt)

dcolley, 2018-02-19:
### Basic Infos

#### Hardware
Hardware:			NodeMCU 1.0
Core Version:      	? where can I see this?

### Description

I am reading a pulse from the Gas meter. This works fine, and I can increment the pulseCounter in the interrupt.
Now I want to send the data to a web service:

HTTPClient works in **loop()** but not in **onPulse()** (interrupt)

I am trying to use interrupts, I read in various blogs this will help me save battery life.

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			? don't know
Flash Frequency:	? don't know
Upload Using:		USB?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
//#include <WiFiUdp.h>
#include <ESP8266WiFiType.h>
//#include <ESP8266WiFiAP.h>
#include <WiFiClient.h>
#include <WiFiClientSecure.h>
#include <WiFiServer.h>
//#include <ESP8266WiFiScan.h>
#include <ESP8266WiFiGeneric.h>
#include <ESP8266WiFiSTA.h>
#include <ESP8266HTTPClient.h>

const int led = LED_BUILTIN; //13;//the led attach to
const int pin = D2; // NodeMCU uses internal pin numbers!

char ssid[] = "MyHomeWifi";     //  your network SSID (name)
char pass[] = "s3cr3t";  // your network password

//int wifiStatus = WL_IDLE_STATUS;     // the Wifi radio's status

// Variables will change:
int ledState = HIGH;      // the current state of the output pin
int pinState;             // the current reading from the input pin
int lastPinState = LOW;   // the previous reading from the input pin

volatile long reading = 0;
volatile long pulseCount = 0;

WiFiClient client;

void setup() {
  Serial.begin(9600);
  slog("Starting...");
  pinMode(led, OUTPUT);
  pinMode(pin,INPUT_PULLUP);
  digitalWrite(led, HIGH);
  attachInterrupt(digitalPinToInterrupt(pin), onPulse, CHANGE);

  slog("Connecting WiFi ");
  WiFi.disconnect();
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, pass);
  while(WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(250);
  }

  slog("Connected WiFi");
  printWifiInfo();  
}
// ===========================================================================
void loop() {
  delay(10000);
  //getHttp(); // <=== OK here
}
// ===========================================================================
void slog(char log[]) {
  Serial.print("LOG: ");
  Serial.println(log);
}
// ===========================================================================
void printWifiInfo() {
  Serial.println("==== WIFI DETAILS ====");
  Serial.print("IP : ");
  Serial.println(WiFi.localIP());
  //WiFi.printDiag(Serial); // <== this will print password!
} 
// ===========================================================================
void onPulse() {
  slog("onPulse start");
  reading = digitalRead(pin);
  digitalWrite( led, reading );

  slog("PULSE!");
  switch(reading) {
    case LOW:  // SWITCH DOWN = ON
      pulseCount++;
      getHttp(); // <=== NOT OK HERE
      break;
    case HIGH: // SWITCH UP = OFF
      break;
  }
}

// ===========================================================================
void getHttp() {
  HTTPClient http;
  String host = "192.168.1.10";
  int port = 3000;
  String url = "/";
  http.begin(host,port,url);
  int httpCode = http.GET();
  if (httpCode) {
    if (httpCode == 200) {
      String payload = http.getString();
      Serial.println(payload);
    }
  }
  Serial.println("closing connection");
  http.end();
}
```

### Debug Messages

Sometimes I get connection refused:
```
LOG: onPulse start
LOG: PULSE!
[HTTP-Client][begin] host: 192.168.1.10 port: 3000 uri: /
[HTTP-Client] failed connect to 192.168.1.10:3000
[HTTP-Client][returnError] error(-1): connection refused
closing connection
[HTTP-Client][end] tcp is closed
Fatal exception 9(LoadLOG: onPulse start
LOG: PULSE!
StoreAlignmentCause):
epc1=0x401051b9, epc2=0x00000000, epc3=0x00000000, excvaddr=0xffffffff, depc=0x00000000

Exception (9):
epc1=0x401051b9 epc2=0x00000000 epc3=0x00000000 excvaddr=0xffffffff depc=0x00000000

ctx: sys 
sp: 3ffffd90 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff30:  4020cd5c 3fff0acc 3ffe9c04 4020cd78  
3fffff40:  4022396e 3ffe9d64 3ffef1a8 4020cdf5  
3fffff50:  40223a70 3ffe9d64 3ffef1d0 0064c228  
3fffff60:  4022b264 3ffef1a8 3ffef1d0 60000600  
3fffff70:  4022b271 3ffef1d0 3ffef1a8 4022b271  
3fffff80:  4022b2b6 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffef1f0 3fffdab0 00000000 40205083  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<
⸮⸮⸮⸮J⸮⸮
```
...and sometimes I get LoadStoreAlignmentCause
```
LOG: onPulse start
LOG: PULSE!
[HTTP-Client][begin] host: 192.168.1.10 port: 3000 uri: /
Fatal exception 9(LoadStoreAlignmentCause):
epc1=0x40105118, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000003, depc=0x00000000

Exception (9):
epc1=0x40105118 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000003 depc=0x00000000

ctx: sys 
sp: 3ffffad0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffc70:  00000001 3fff1a10 00000000 00000000  
3ffffc80:  40105344 0b766063 3ffef818 00000000  
3ffffc90:  3ffef1d0 3ffef818 3fff1a0c 4020cc14  
3ffffca0:  00000000 0b5e88df 40201364 00001388  
3ffffcb0:  3ffef818 3fff19e4 00000bb8 00000bb8  
3ffffcc0:  3fff19e4 00000000 3fff20d4 40202cba  
3ffffcd0:  0a01a8c0 00000000 00000020 0000003a  
3ffffce0:  00000024 00000008 3ffef9d8 00000000  
3ffffcf0:  3ffef9f8 00000bb8 3fff19e4 40202878  
3ffffd00:  3ffe9700 0a01a8c0 3ffe9700 0a01a8c0  
3ffffd10:  3fff1bbc 2030312e 74726f70 4020311f  
3ffffd20:  75203030 3fff1dc4 3ffffdc0 40203585  
3ffffd30:  00000000 3ffffd40 00000008 4010097c  
3ffffd40:  03ac0f00 04ac0f00 3fff19e4 00000000  
3ffffd50:  ffffffff 00000000 3ffffdc0 40203ec4  
3ffffd60:  3ffe951f 3ffffe30 3ffffd40 3ffffd70  
3ffffd70:  3ffffe3c 3ffffe30 3ffffdc0 00000bb8  
3ffffd80:  3ffe92e0 0000014d 0000014d 4010020c  
3ffffd90:  ffffffff 00000004 3fff19dc 00000020  
3ffffda0:  ffffffff 00000004 3ffef7c4 40203f72  
3ffffdb0:  ffffffff 00000004 3ffef7c4 4020221c  
3ffffdc0:  3fff1dc4 3fff19e4 3fff1bbc 0000000f  
3ffffdd0:  0000000c 00000bb8 00001388 3fff1d84  
3ffffde0:  0000000f 00000001 3fff1da4 0000000f  
3ffffdf0:  00000000 3fff1cdc 0000000f 00000000  
3ffffe00:  3fff1cfc 0000001f 00000011 3fff1d2c  
3ffffe10:  0000000f 00000000 00000000 00000000  
3ffffe20:  00000000 ffffffff 3ffef900 00000000  
3ffffe30:  00000000 00000000 00000000 00000000  
3ffffe40:  00000000 00000000 3fff19c4 0000000f  
3ffffe50:  00000001 3fff19a4 0000000f 0000000c  
3ffffe60:  7fffffff 00000004 3ffef7c4 402022d1  
3ffffe70:  40106b12 00000080 00000000 40106b6c  
3ffffe80:  c0037025 0b7aea8f 3fff10ac 00000000  
3ffffe90:  00000000 00000000 0000001f 00000022  
3ffffea0:  3fffc200 40106ad8 3fffc258 4000050c  
3ffffeb0:  40000f68 00000030 0000000c ffffffff  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  00000002 4020cdd8 00003a98 00000000  
3ffffee0:  3ffef1a8 3ffe9d64 3ffef1a8 3fffdab0  
3ffffef0:  00000000 3fffdcb0 3ffef1f8 00000030  
3fffff00:  3ffef9d8 0000000d 3ffe9c0c 4020bf74  
3fffff10:  4020cd5c 3fff0acc 3fff0fe4 40104928  
3fffff20:  40226992 00000000 3fff0fe4 3fff0ac8  
3fffff30:  4020cd5c 3fff0acc 3ffe9c0c 4020cd78  
3fffff40:  401050b6 0b5e8cf1 3ffef1a8 4020cdf5  
3fffff50:  40105172 3ffe9d64 0b5e8cf1 00000000  
3fffff60:  401050b6 0b5e8cf1 3ffef1d0 60000600  
3fffff70:  4022b290 3ffef1a8 3ffef1d0 2461d4c2  
3fffff80:  4022b2b6 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffef1f8 3fffdab0 00000000 00000000  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<
⸮⸮⸮J⸮y9⸮⸮Starting...
Connecting WiFi ................Connected WiFi
==== WIFI DETAILS ====
IP : 192.168.1.123
pm open,type:2 0
```

Pablo2048, 2018-02-19:
IMHO no WiFi related function will work in interrupt this way. You have to set some flags in interrupt and handle things in foreground (aka loop())...

dcolley, 2018-02-19:
@Pablo2048 thanks for your answer. So far I was only able to make this work in the ```loop()```.

I am using ```millis()```since last submission to determine the submission interval. 
If there is a better pattern - e.g. ```delay()``` - I would love to hear about it. 
I am interested in reducing battery drain... so any advice about [deep] sleep would also be appreciated.

devyte, 2018-02-19:
@dcolley why would you want to do http ops from inside an ISR?? ISR have very strict timing constraints for execution time, in addition to what can actually execute inside.
Like stated above: set a flag and check it in the loop, or schedule a function to be executed in the next loop run (Schedule.h).
If you need to execute something periodically and need to reduce battery drain, research deepsleep.
Closing as user error.

closed 2018-02-19 by devyte

dcolley, 2018-02-20:
agreed. this should be marked as a question

dcolley, 2020-12-16:
Just for completeness, here is the code that worked for me. I did not try any deep/sleep battery saving techniques in this script, the unit is permanently connected to a usb phone-charger.

```c
#include <SPI.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
//#include <WiFiUdp.h>
#include <ESP8266WiFiType.h>
//#include <ESP8266WiFiAP.h>
#include <WiFiClient.h>
#include <WiFiClientSecure.h>
#include <WiFiServer.h>
//#include <ESP8266WiFiScan.h>
#include <ESP8266WiFiGeneric.h>
#include <ESP8266WiFiSTA.h>
#include <ESP8266HTTPClient.h>

/*****************************************/
const int led = LED_BUILTIN; //13;//the led attach to
const int pin = D2; // NodeMCU uses internal pin numbers!

char ssid[] = "your_WIFI_SSID";     //  your network SSID (name)
char pass[] = "t0ps3cr3t";  // your network password

int wifiStatus = WL_IDLE_STATUS;     // the Wifi radio's status

// Variables will change:
int ledState = HIGH;      // the current state of the output pin
int pinState;             // the current reading from the input pin
int lastPinState = LOW;   // the previous reading from the input pin

HTTPClient http;
String host = "192.168.1.123";
int port = 3000;
String url;

const long interval = 60*1000; // 60 seconds
//const long interval = 10*1000; // 10 seconds
volatile long pulseCount = 0;
volatile long lastReport = 0;

WiFiClient client;

int LED_ON = LOW;
int LED_OFF = HIGH;

// ===========================================================================
// Note, by declaring ICACHE_RAM_ATTR, it seems this fn has to be above setup() and loop()
void ICACHE_RAM_ATTR onPulse() {
//  slog("onPulse start");
  detachInterrupt(pin);
  long reading = digitalRead(pin);
  digitalWrite( led, reading );
  switch( reading ) {
    case LOW:  // SWITCH DOWN = ON
      slog("PULSE");
      pulseCount++;
      //getHttp();
      break;
    case HIGH: // SWITCH UP = OFF
      break;
  }
  attachInterrupt(digitalPinToInterrupt(pin), onPulse, CHANGE);
}

// ===========================================================================
void setup() {
  Serial.begin(9600);
  Serial.println("Starting...");
  pinMode(led, OUTPUT);
  pinMode(pin,INPUT_PULLUP);
  digitalWrite(led, LED_ON);
  attachInterrupt(digitalPinToInterrupt(pin), onPulse, CHANGE);

  Serial.print("Connecting WiFi ");
  WiFi.disconnect();
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, pass);
  while(WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(250);
  }

  Serial.println("Connected WiFi");
  digitalWrite(led, LED_OFF);
  printWifiInfo();  
}
// ===========================================================================
void loop() {
//  delay(10000);
  if(millis()-lastReport > interval){
    getHttp(); // <=== OK here
  }
}
// ===========================================================================
void slog(String log) {
  Serial.print("LOG: ");
  Serial.println(log);
}
// ===========================================================================
void printWifiInfo() {
  Serial.println("==== WIFI DETAILS ====");
  Serial.print("IP : ");
  Serial.println(WiFi.localIP());
  //WiFi.printDiag(Serial); // <== this will print password!
} 
// ===========================================================================
void getHttp() {
  //noInterrupts();
  url = String("/device/")+getMacAddress()
    + String("/metric/count")
    + String("/value/")+pulseCount;
  slog(url);
  http.begin(host,port,url);
  int httpCode = http.GET();
  if( httpCode ) {
    if( httpCode == 200) {
      pulseCount = 0;
      String payload = http.getString();
      Serial.println( payload );
      flashOk();
    } 
    lastReport = millis();
  } else {
      String payload = http.getString();
      Serial.println( payload );
      flashError();
    }
  Serial.println("closing connection");
  http.end();
  //interrupts();
}

// ===========================================================================
String getMacAddress() {
  // print your MAC address:
  byte mac[6];
  WiFi.macAddress(mac);
  //Serial.print("MAC address: ");
  return 
     String( mac[5], HEX ) +":"
    +String( mac[4], HEX ) +":"
    +String( mac[3], HEX ) +":"
    +String( mac[2], HEX ) +":"
    +String( mac[1], HEX ) +":"
    +String( mac[0], HEX );
}
// ===========================================================================
void flashOk() {
  digitalWrite(led, LED_ON);
  delay(500);
  digitalWrite(led, LED_OFF);
}
// ===========================================================================
void flashError() {
  digitalWrite(led, LED_ON);
  delay(250);
  digitalWrite(led, LED_OFF);
  delay(250);
  digitalWrite(led, LED_ON);
  delay(250);
  digitalWrite(led, LED_OFF);
  delay(250);
  digitalWrite(led, LED_ON);
  delay(250);
  digitalWrite(led, LED_OFF);
}
// ===========================================================================

```