esp8266/Arduino #4392: Support for user provided MD5 credentials for ESP8266WebServer authentication

IMAN4K, 2018-02-19:
It would be great to have an option to provide `ESP8266WebServer` authentication's credentials as `MD5` values instead of clear texts.
This is useful in case, where `<user:pass>` is stored in `MD5` format and is modifiable.
Adding an option switch like this defaulted to `false`:
`bool authenticate(const char * username, const char * password, bool user_md5 = false);`

earlephilhower, 2018-02-19:
What threat are you trying to protect against?  Sending MD5 over an unsecured channel is still completely insecure as anyone along the path can just replay the MD5 back to authenticate.  It's just trading one plaintext password for another.  And with rainbow tables, reversing MD5 to get back to plaintext is not a problem for anyone with a computer built this decade.

For security, basic HTTP user:pass over an encrypted channel is a common recommendation.  There is now a WebServerSecure which can pretty much replace the unencrypted one w/a 1-line addition to supply its certificate and key.

IMAN4K, 2018-02-19:
Agree.
But `MD5` is better than nothing and is common in modem-router authentication mechanisms & not very easy to break!
Also `ESP8266WebServer` simply hash `user:pass` for authentication.
In my case, i have only user:pass as stored `MD5` value, so i should simply be able to provide them to `ESP8266WebServer`

Don't know if `WebServerSecure` is in buggy state or NOT!

earlephilhower, 2018-02-19:
WebServerSecure's in a pretty good state as far as we know.  I and @d-a-v have done lots of stress testing on the underlying WiFiServerSecure/ClientSecure.  Give it a try, but be aware you need 20KB+ free heap to run a SSL connection.

I'd argue MD5 is perfectly easy to decrypt today.  See https://hashkiller.co.uk/md5-decrypter.aspx for an online version....

closed 2018-03-18 by IMAN4K