esp8266/Arduino #4397: WiFi automatic reconnection not working in 2.4.0 but works in core 2.3.0

jplavoiemtl, 2018-02-20:
### Basic Infos

Hardware:			ESP-12E?
Core Version:      	2.4.0

Description
Here is a bug I found in esp core 2.4.0 that was not there in core 2.3.0.  The esp does not reconnect automatically to WiFi while using WiFiManager. When my esp is running it connect to WiFi no problem. I have a WiFi status of 3 which is connected. Serial.println(WiFi.status());

Then I deactivate WiFi in my router to simulate a WiFi problem. After I reactivate WiFi on the router, the esp does not reconnect to WiFi. I check status at 1 second intervals and I get 6 for a WL_DISCONNECT

The problem is not related with WiFiManager.  To solve it I had to revert back to esp core 2.3.0 and the WiFi reconnects fine and WiFiManager works like a charm. 

It's a really important issue basic automatic WiFi re-connection and I wonder how come it didn't surface in the reports on the new 2.4.0 core.  I hope it will be resolved in the new version.  In the mean time, if someone experience the same problem, revert to core 2.3.0 as a quick fix.

Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E)
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			

Sketch

Basic WiFiManager code to initialize WiFi.

Almost nothing in the main loop.

tonton81, 2018-02-20:
have you tried the handler instead of polling? theres an event callback when ESP is connected and gets an IP, and another event callback when its disconnected. obviously you shouldnt call reconnectMQTT from the callback, but you can set a bool in the callback and when your loop() hits itll disable the bool after running reconnectMQTT

jplavoiemtl, 2018-02-20:
@tonton81 The code here works well and going very fast through the main loop so there is no polling issue.  The purpose of this bug report issue is to report a bug in the WiFi handling in the esp core 2.4.0 that was working fine in previous version in order to fix the current core release.  But I note your disconnect handler use for a future project, thanks for the input.

devyte, 2018-02-20:
@jplavoiemtl please erase all flash, and try with latest git.

jplavoiemtl, 2018-02-20:
@devyte How can I erase all flash?

tonton81, 2018-02-20:
i presume with esptool, igor suppost to add a new feature regarding that, to erase everything, or just some parts when flashing

jplavoiemtl, 2018-02-20:
@tonton81 In the mean time of erasing flash, I'll try the latest git on a brand new Wemos D1 mini board with the latest git in Arduino IDE and report my findings shortly.

devyte, 2018-02-20:
@jplavoiemtl erase flash with esptool-ck.

jplavoiemtl, 2018-02-20:
@tonton81 Test done with a brand new Wemos board and Arduino esp8266 core 2.4.0.  Problem is there , there is no WiFi reconnection after I interrupt WiFi on my router.  I see WL_DISCONNECT value of 6 and the module is not getting back online with WiFi.
Installed core 2.3.0 and there is no re connection problem, it works fine with this older core.  So this seems to confirm the bug.  I used a brand new board instead of erasing flash because I'm not to familiar with erasing flash and lost a full day on this WiFi issue yesterday.  I'm open to try erasing flash later today or tomorrow.

drschlaumeier, 2018-02-20:
Is it same as already reported?
#4257

devyte, 2018-02-20:
@jplavoiemtl as I said, please try *latest git*, not release 2.4.0.

jplavoiemtl, 2018-02-20:
@devyte trying to install the latest git.  After restarting Arduino where do I go to see if I'm using this git?  In boards manager for esp8266 I only see 2.3.0-rc2 and previous versions.  How can I use the git?

Here are the steps I did:
Using git version

Linux build status

    Install Arduino 1.8.2 from the Arduino website.
    Go to Arduino directory
    Clone this repository into hardware/esp8266com/esp8266 directory (or clone it elsewhere and create a symlink)

cd hardware
mkdir esp8266com
cd esp8266com
git clone https://github.com/esp8266/Arduino.git esp8266

    Download binary tools (you need Python 2.7)

cd esp8266/tools
python get.py

    Restart Arduino

Now what?

devyte, 2018-02-20:
@jplavoiemtl you can ever have only one version of a package installed, and that holds for git. That means that, if you install git, you also have to uninstall whatever version you installed from the board manager. Then restart IDE, then rebuild/retest.
BTW, after this, you won't see git in the board manager.
If you want to make sure of the version you are building with, use ESP.getCoreVersion(), ESP.getSdkVersion() etc to print out to serial.

jplavoiemtl, 2018-02-20:
@devyte Using  ESP.getCoreVersion(), ESP.getSdkVersion() etc to print out to serial I get:
Core: 00000000
Sdk: 2.2.0(f28eaf2)
Does this makes sense for core? 
My boards manager shows I have esp82666 by ESP8266 Community version 2.3.0 installed.
[image]

babylonx, 2018-02-21:
I use my own connection retaining manager in my applications which basically among others disconnects and reconnects the WiFi using the disconnect() and begin() methods and then waits for the status() to return the WL_CONNECTED (3) state. In 2.4.0 if I disconnect() then begin() I always get status WL_DISCONNECTED (6) even though the WiFi is actually reconnected since I can access the module via the WiFi network. WiFi.localIP() also returns 0.0.0.0 ip.

d-a-v, 2018-02-21:
Could you please provide a simple sketch that reproduce the issue ?
I don't see reconnection problem when the AP is lost then comes back.
```WiFi.status()``` correctly returns 3(connected), then 6(disconnected), then 1(no ssid), then 3.

jplavoiemtl, 2018-02-21:
// NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS) sur COM4
// Must use ESP8266 boards core in Arduino IDE version 2.3.0 because
// WiFi do not reconnect automatically with core 2.4.0 discovered in Feb 2018

#include <Timer.h>
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
#include <PubSubClient.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager
#include <ESP8266mDNS.h>         //for OTA programming
#include <WiFiUdp.h>             //for OTA programming
#include <ArduinoOTA.h>          //for OTA programming

const int ledPin=2;    // Led on pin 2, active low

IPAddress SERVER (192, 168, 1, 130);
#define SERVERPORT  1883
#define USERNAME    "userjp"
#define KEY "delphijpmqtt"
#define CLIENT_ID "NODEMCUMOD5"

const char CMD_OUT[] = "m5cmdout";        //MQTT topic commande vers le module 
const char STA_OUT[] = "m5staout";        //MQTT topic statut du module 
const char TEMP_OUT[] = "m5temperature";  //MQTT topic statut du module
const char CMD_OTA[] = "m5otamode";       //MQTT topic commande OTA programming vers le module
const char STA_OTA[] = "m5staota";        //MQTT topic statut du module OTA
const char PAYLOAD_ON[] = "on";    
const char PAYLOAD_OFF[] = "of"; 
const int period = 600000;                //Délai en ms du timer. 5000 = 5s. 20 min = 1200000
const int periodLED = 1000;               //Délai en ms du timer de la LED

Timer t_LED;                              //Timer for LED flashing
bool otaMode;                             //OTA mode flag
bool ledActive;                           //Variable to show if led is active or not

// Create an ESP8266 client class to connect to the MQTT server
WiFiClient espClient;
PubSubClient client(espClient);

void setup() {
    Serial.begin(115200);

    //Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wifiManager;
    wifiManager.setTimeout(180);    
    
    if(!wifiManager.autoConnect("NodeMcuM5")) {
      Serial.println("failed to connect and hit timeout");
      delay(5000);            //To avoid too many quick tries
      ESP.reset();            //reset and try again
      delay(5000);
    }    
    
    Serial.println("WiFi connected...");   //if you get here you have connected to the WiFi
    Serial.println("local ip");
    Serial.println(WiFi.localIP());
    client.setServer(SERVER, SERVERPORT);
    client.setCallback(callback);
    reconnectMQTT();  
    t_LED.every(periodLED, flashLED);       //Timer for LED
    otaMode = false;
    ledActive = false;

    // OTA programming setup.  No authentication by default
    ArduinoOTA.setPassword((const char *)"123");

    ArduinoOTA.onStart([]() {
      Serial.println("Start");
    });
    ArduinoOTA.onEnd([]() {
      Serial.println("\nEnd");
    });
    ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
      Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
    });
    ArduinoOTA.onError([](ota_error_t error) {
      Serial.printf("Error[%u]: ", error);
      if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
      else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
      else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
      else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
      else if (error == OTA_END_ERROR) Serial.println("End Failed");
    });
    ArduinoOTA.begin();
    Serial.println("OTA Ready");
}

//Fonction appelée par le broker MQTT chaque fois qu'un message est publie sur un topic auquel le module est abonné
void callback(char* topic, byte* payload, unsigned int length) {
  String topicString = topic;
  String payloadString = (char*)payload;
  payloadString = payloadString.substring(0,2);
  
  if(topicString == CMD_OUT && payloadString == PAYLOAD_ON) {  //Commande pour activer la sortie
    ledActive = true;
    Serial.println("LED active");
    client.publish(STA_OUT, "on");           //Publish status ouput is ON
  }
 
}

//MQTT connection
void reconnectMQTT() {
  yield(); 
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    
    if (client.connect(CLIENT_ID, USERNAME, KEY)) {
      Serial.println(" connected");                                    
      Serial.print("Client state connected, rc=");                                                     
      Serial.println(client.state());                             
      client.subscribe(CMD_OUT, 1);   //Topic de commande
      client.subscribe(CMD_OTA, 1);   //Topic de commande OTA programming
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }  
  }
}

void loop() {

  //If MQTT client is not connected and wifi is connected
  if (!client.connected() && WiFi.status() == WL_CONNECTED) {
    reconnectMQTT();
  }

  if(otaMode==true) {                    
    ArduinoOTA.handle();
  }

  client.loop();
}

babylonx, 2018-02-21:
@d-a-v 

It is not happening if the AP disappears. It is happening if you call WiFi.disconnect() and then WiFi.begin() to reconnect. Just try this simple scetch

`#include <ESP8266WiFi.h>

unsigned long sinceWifiBegin = 0;
unsigned long sinceSerialPrint = 0;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);

  WiFi.begin("SSID", "PASS");
  sinceWifiBegin = millis();
}

void loop() {
  if (millis() - sinceSerialPrint > 5000) {
    Serial.print(F("WiFi Status: "));
    Serial.println(WiFi.status());
    sinceSerialPrint = millis();
  }

  if (millis() - sinceWifiBegin > 40000){
    Serial.println(F("WiFi reconnecting."));
    WiFi.disconnect();
    delay(1000);
    WiFi.begin("SSID", "PASS");
    sinceWifiBegin = millis();
  }
}` 

Connection in setup() is successful and gives status 3. If you disconnect then reconnect (with the same credentials) then status is always 6.

d-a-v, 2018-02-21:
Thanks, I could reproduce.

I added this at the end of ```setup()```:
```
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println(WiFi.localIP());
```
and now it works as expected.

I don't know if we can call that a workaround or if it is a normal behaviour. I understand it was working with core-2.3 but the sdk has changed since then.

You may also try to call ```WiFi.persistent(false);``` before ```.disconnect()``` instead, as the internal code has a different behaviour calling sdk functions, though I'm not sure about all implications of this call in your sketch.

d-a-v, 2018-02-21:
@jplavoiemtl you are not calling ```.disconnect()```. So when does the issue happen ?

jplavoiemtl, 2018-02-21:
@d-a-v yes I'm not calling `disconnect()` I only use WiFiManager which handles everything WiFi for me.  That's what I like about WiFiManager, it does everything behind the scene.  Like I described originally, when I disable WiFi on my router for a few seconds then re-enable it, I get 6(disconnected).  So in my main loop I only check the WiFi status to reconnect to my MQTT broker.  Since it's always 6 in core 2.4 it never reconnects.  This is why I like using core 2.3 because the WiFi status seems to world better and I get 3 like it should...

babylonx, 2018-02-21:
@d-a-v

I tried the .persistent (false) method but it didn't make any difference. The thing is that Wifi actually recconects even though status and IP are not updated. I can still reach my devices via the Wifi. So is there a workaround if someone needs to disconnect from and reconnect to the same AP?

jplavoiemtl, 2018-02-21:
@babylonx hopefully they will fix that Wifi status not updated issue.  In my case Wifi status is important because if Wifi is truly not present and I try to reconnect to my MQTT broker, there are errors and the module stops running.  So it's important to have a reliable Wifi status to prevent that.

d-a-v, 2018-02-21:
@babylonx 

Does the other "workaround" works (not calling ```disconnect``` if not connected) ?
Can you try with ```WiFi.setAutoReconnect(false);``` before ```.disconnect()``` ?

@jplavoiemtl 

I tried WiFiManager's ```AutoConnect``` example. In its ```loop()``` I print ```WiFi.status()```.
It happens that status stays at 3 (connected) even several minutes after I switch my AP off. Once it is switched back on, the ESP automatically reconnects, and status stays at 3.
This is a different (opposite?) behavior than yours which too seems to be not correct.
The SDK should report that the AP has vanished, and it does in other situations (my previous comment above).
I'll try to dig further, but this interesting question could be also asked to WiFiManager's repo.

This is the modified loop in the example:
```
void loop() {
    // put your main code here, to run repeatedly:
    static unsigned long next = 0;
    if (next < millis())
    {
        next = millis() + 1000;
        Serial.println(WiFi.status());
    }
}
```

jplavoiemtl, 2018-02-21:
@d-a-v To let you know, I opened this issue [#532](https://github.com/tzapu/WiFiManager/issues/532) to WiFiManager's repo.  They said that it was a core issue according to them this is why I posted my issue here.

d-a-v, 2018-02-22:
@igrr @tablatronix 

See the interactive sketch below.
With latest master and PR #4413, here's what happens.

Once ESP running with PC successfully pinging,
* calling disconnect() once does not change status (stays at 3=connected), but ping fails
* calling *again* disconnect() *does* change status to 0=nossid
* calling WiFi.begin(ssid,password) brings ESP back to network (ping works)

Now, with ESP running and pinging, and switching the AP off,
* WiFi.status() switches from 3 to 1=disconnected few seconds later
* and back to 3 after AP is on, but ping is badly working (between 8s to 18s not ms)
* setting to NONE_SLEEP brings good pinging back

So to summarize
* things are OK when calling **disconnect twice**, and using **only WiFi.begin(ssid,psk) to reconnect**.
* AP vanishing and coming back is OK in mode NONE_SLEEP. I never had to use NONE_SLEEP before, I don't know why me settings are LIGHT_SLEEP by default.

```
#include "ESP8266WiFi.h"
#include "user_interface.h"

const char SSID[] = "ssid";
const char PSK[] = "psk";

unsigned long sinceSerialPrint = 0;
unsigned long lastChange = 0;

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);

  WiFi.mode(WIFI_STA);
  WiFi.begin(SSID, PSK);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println(WiFi.localIP());
  lastChange = millis();
}

void loop() {
  if (millis() - sinceSerialPrint > 1000) {
    Serial.print(F("WiFi Status: "));
    Serial.print(WiFi.status());
    Serial.print(F(" - last change: "));
    Serial.println((long)(lastChange - millis()));
    sinceSerialPrint = millis();
  }

  #define DO(x...) Serial.println(F( #x )); x; lastChange=millis(); break
  
  if (Serial.available())
  {
    switch (Serial.read())
    {
      case 'd': DO(WiFi.disconnect());
      case 'b': DO(WiFi.begin());
      case 'B': DO(WiFi.begin(SSID, PSK));
      case 'r': DO(WiFi.reconnect());
      case 'c': DO(wifi_station_connect());
      case 'a': DO(WiFi.setAutoReconnect(false));
      case 'A': DO(WiFi.setAutoReconnect(true));
      case 'n': DO(WiFi.setSleepMode(WIFI_NONE_SLEEP));
      case 'l': DO(WiFi.setSleepMode(WIFI_LIGHT_SLEEP));
      case 'm': DO(WiFi.setSleepMode(WIFI_MODEM_SLEEP));
    }
  }
}
```

tablatronix, 2018-02-22:
Nice sketch

tablatronix, 2018-02-22:
I cannot reproduce this.

tablatronix, 2018-02-22:
ohh it was reverted, I need to pull in https://github.com/esp8266/Arduino/pull/4413/commits/66e2bd0c2a4afa028ae25a7c595e67e39e791ecd

tablatronix, 2018-02-22:
Still cannot reproduce, gonna erase flash

d-a-v, 2018-02-22:
It's #4412 that fixed this issue.
The sketch above shows other new issues.
I /cc you because it might help for #4372 (by adding some more ```case```).

tablatronix, 2018-02-22:
ohh ok , I see

I am not even connecting to my router now, so I gotta figure out whats going on.

tablatronix, 2018-02-22:
ok erase flash seems to have sorted it out

tablatronix, 2018-02-22:
yeah i seem to be having connection issues now with ap not found.

```

 Starting
STA+AP
Mode: STA+AP
PHY mode: N
Channel: 1
AP id: 0
Status: 1
Auto connect: 1
SSID (8): MYSSID
Passphrase (0): 
BSSID set: 0
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: wifi station disconnect
scandone
*WM: Connecting to saved AP
*WM: enableSTA PERSISTENT ON
*WM: connectTimeout not set, ESP waitForConnectResult...
wifi evt: 8
scandone
no MYSSID found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
reconnect
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
scandone
no MYSSID found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
reconnect
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
scandone
no MYSSID found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
*WM: Connection result: WL_NO_SSID_AVAIL
```

tablatronix, 2018-02-22:
Must be that other bug corrupting some memory, Ill test one thing at a time and come back to this.

tablatronix, 2018-02-22:
Yeah even after erasing flash I am getting weird issues, like this on boot. It refuses to connect to my saved SSID.

tablatronix, 2018-02-22:
ok nm, I think I was using 2.4.0 ( so that might be a known issue )

romansavrulin, 2018-03-03:
@tablatronix Good point with memory corruption.

We've noticed the following: being built with Arduino IDE, and 2.4.0 the issue with wrong connection status pops-up. But the same code being built with Platformio reconnects perfectly.

The only difference I see is the link order and map file. So very likely to be memory corruption issue.

pieman64, 2018-03-03:
@devyte I always use the portable setup for Arduino IDE because the regular setup is quite nasty. Users can run dozens of these portable versions (2.30, 2.40-rc2, 2.4.0, master etc) all at the same time whilst they are waiting for the next release.

tablatronix, 2018-03-03:
I use platformio, I can change it on the fly.

romansavrulin, 2018-03-03:
@tablatronix, a bit off-topic, but can you point me how to use git version with platformio?

tablatronix, 2018-03-03:
http://docs.platformio.org/en/latest/platforms/espressif8266.html#stable-and-upstream-versions
and for some reason they removed this from their docs

`platform = https://github.com/platformio/platform-espressif8266.git#feature/stage`

So maybe its no longer needed, in latest version of pio, I have not tested that yet.

https://github.com/platformio/platform-espressif8266

d-a-v, 2018-03-08:
@jplavoiemtl Is this still relevant with 2.4.1 ?

tablatronix, 2018-03-08:
I think my issue turned out to be NVS , when I erased NVS, I have not encountered this again.

jplavoiemtl, 2018-03-09:
@d-a-v I tested with the new 2.4.1 core and it is now working fine.  My issue looks to be fixed here.  Thank you.

closed 2018-03-09 by jplavoiemtl

Nakul93, 2018-04-18:
I am using 2.4.1 core and the latest git commit and also enabled the WiFi debug. So on Wifi disconnection (note: I don't turn off the router I just turn off the internet connection in my router) and on reconnection it is still not connecting. In debug I am getting 'wifi evt: 7', what does this mean?

pfabri, 2019-04-09:
@Nakul93 
Could you please let me know how one switches between different version of esp8266/Arduino using PlatformIO? Many thanks!

Pablo2048, 2019-04-10:
@pfabri  just search/read the documentation... https://docs.platformio.org/en/latest/projectconf/section_env_general.html

pfabri, 2019-04-10:
@Pablo2048 
Thanks for your response. I have read the documentation numerous times and spent hours searching for an answer. But the info you provided a link to only tells you how to use a specific PlatformIO version of arduinoespressif, which is NOT the same as selecting an Arduino Core version. E.g. you can't select 2.4.1rc2 from PlatformIO. IvanKravets explicitly points out that PlatformIO does not support beta version software, but people seem to somehow still manage to switch to these branches. I presume they don't do so from ArduinoIDE, as that isn't really suitable for any serious development. Many thanks if you can help me out.

tablatronix, 2019-04-10:
Most people fork the platformio repo and manually pull from espressif, pio is working on a way to pull versioning from espressif instead if their useless versioning releases. I have had some success with pulling staging and manual git checkouts in platform

pfabri, 2019-04-12:
@tablatronix Many thanks!